joshuagrisham/samsung-galaxybook-extras #2: GB3 Ultra LED turn off slowly after turned on

archdsp, 2024-01-26:
Hi! joshua!
I always thanks to you to make audio, kbd backlight and other things enabled for galaxy book

## Issue
I cloned your repo and install driver 3~4 days2 ago and there are no issue at all. 

But after i sync latest commit(today Jan 26/2024), Led only turns on through manual command or script and not working through gnome GUI.
```sh
./set-keyboard-brightness.sh 3
```
Also, LED turns on physically but soon it turns off despite of driver showing 
```sh
% cat /sys/class/leds/samsung-galaxybook::kbd_backlight/brightness
3
```
Also no dmesg error appear.

## Question
I want to contribute your work to add functionality like 
* FN+F9 key works via ACPI or WMI driver
* My camera id OV02c10. intel has IPU6 driver repo and offer OVO02c10. but not work my system.
* Make fingerprint works. I showed your PR on libfprint. Mine has different egistec module number. I

Where can i find a CONTRIBUTION guide or Project ToDo list or Code guide, wiki or else?

joshuagrisham, 2024-01-27:
Hi @archdsp thanks for testing and giving it a try!

I don't have access to a GB3 Ultra to test with unfortunately, but I did just push a big update to the driver and other stuff today which I am crossing my fingers will work well for you?

I have also noticed this behavior where the backlight starts to turn off and on a bit sporadically during my testing and debugging; in my experience this seems to happen when something was not executed quite right against the Samsung ACPI device and it seems to kind of "bug out".

But now I have added a lot more of the proper "init" logic to the driver so hopefully this will help -- it has helped on my GB2 Pro and I am very interested to hear feedback from you if you manage to try it out!

joshuagrisham, 2024-01-27:
Regarding making the hotkeys work via driver instead, from what I understand when trying to research this, what is typically done is more that an ACPI event on a custom device is generated for hotkeys instead of an event going to the standard keyboard. In these cases, the device/platform drivers can listen to those ACPI events and then either directly act based on them, or "remap" them to normal keyboard events (or both, actually). However, that is not really the case here, I think -- these events are already coming in as standard keyboard events.

Do you know of a pattern that will work as you stated? If so then I would be very eager to hear about it!

Regarding camera - yeah in this repo it is just a dummy script to unbind/rebind the driver with intention to "fake" the Camera Privacy toggle hotkey like it works in Windows. Not really a full-blown solution but nice that the key works I guess :) It should be relatively easy to change the script so that it matches based on a list instead of just one? Or that it searches all camera devices instead?

For fingerprint please feel free to add a comment at libfprint and/or create a new issue with your device ID and I can try to see if I can add it and/or test it, we can take that one over there :)

And lastly .... contribution guide? Hmm so far this is kind of a "one man show" over here, it is just me hacking, debugging, testing, and writing code, but I welcome the help and collaboration if you are willing and able!! :sunglasses:

joshuagrisham, 2024-04-20:
Hi both

I have since this issue implemented a keyboard event filter to catch Fn+F9 on at least my specific model (suspect it will work for most other variants which have the `SAM0429` device as well, and maybe even `SAM0427` but that might need some tweaks) -- hopefully this is working better now?

@NailGit I assume that what you mentioned here is out-of-date compared to your comments in #10 ?

closed 2024-04-20 by joshuagrisham