joshuagrisham/samsung-galaxybook-extras #4: Integration with the desktop power managers

jcarrano, 2024-02-09:
First of all, great job. I tested this on a 950XEE and it works just as advertised (after adding it to the HWDB pattern)

# using `kbdillumtoggle`

The keyboard backlight is picked up automatically by UPower. Setting the kbd brightness by sending dbus messages as [described in the arch wiki](https://wiki.archlinux.org/title/keyboard_backlight) and has the advantage of not requiring root privileges.

Now the issue is how to get the power manager to recognize the key. I don't think `prog2` is the best choice. Browsing the HWDB I see three values related to kbd backlight, which correspond to xf86 keysim values:

- kbdillumup: XF86XK_KbdBrightnessUp
- kbdillumdown: XF86XK_KbdBrightnessDown
- kbdillumtoggle: XF86XK_KbdLightOnOff

I'm using xfce4-power-manager and it recognizes "up" and "down" keys and I just made a patch for it to recognize the "toggle" key. I don't know whether other desktop environments handle the keys as well.

# Performance mode 

I took a look at power-profiles-daemon and TLP and they both use [platform_profile](https://www.kernel.org/doc/html/latest/userspace-api/sysfs-platform_profile.html) to set the performance mode. If this driver were to use that interface it would work out of the box. I have no experience with kernel hacking, but I could give it a try.

joshuagrisham, 2024-02-11:
Hi again @jcarrano regarding the keyboard hotkey for backlight I tested with all of these and it was not exactly what we want IMO

- kbdillumup: On repeated presses, the backlight goes from 0 to max (3) with each press and then just stays at 3 with every subsequent press (there is no way to lower or turn off the backlight)
- kbdillumdown: Similar to the above, on repeated presses, the backlight goes from max (3) to 0 with each press and then just stays at 0 with every subsequent press (there is no way to raise or turn on the backlight)
- kbdillumtoggle: The backlight just toggles back and forth between 0 and max (3) and there is no way to get to the intermediate levels (1, 2, etc)

So because of this to me it seems we need a "custom" implementation which can support the same kind of "round robin" -like functionality that Samsung has implemented in Windows (each key press cycles from 0 to 3 and then starts over at 0 again).

Now what exists on the main branch is a very "early" implementation of trying to implement this within the driver itself using an i8042 filter to catch the hotkey and then implement this "round robin" logic in the hotkey handler itself. I hope that this will be a much better solution but of course open to input on this!

Regarding performance mode, this you already fixed for us ( :tada: :heart: ) and I have merged and tweaked it a bit to fit in with stuff I had already locally for fan speed and this hotkey stuff etc, some naming and code flow stuff, etc

jcarrano, 2024-02-11:
> kbdillumtoggle: The backlight just toggles back and forth between 0 and max (3) and there is no way to get to the intermediate levels (1, 2, etc)

Yes, my point is that this should be fixed in the desktop environment, e.g. xfce4-power-manager or gnome. Of course, we don't know if upstream will accept it, but IMO cycling through different levels make much more sense than toggling on/off. I'm waiting for the XFCE dev's to give me access to gitlab to see what they think. Considering backlight support is still [fairly new](https://www.omglinux.com/gnome-adds-keyboard-backlight-control/) and that there will be more laptops with only one button, I'd believe there's a chance the key handling will improve in the future.

There's no "right" answer, but looking forward to the day this is included in the official kernel, using `kbdillumtoggle` will make it "just work" even if not optimally, and this repo can still exist to provide an alternative HWDB entry and scripts to be able to use all levels.

>  trying to implement this within the driver itself....

My concern is whether the kernel devs will want to merge in a patch that integrates the key logic instead of delegating it to userspace.

joshuagrisham, 2024-04-20:
Thanks for the feedback @cptMustard ! Very good to hear.

@jcarrano thanks again for your PR and assume you have already seen that I went for this pattern which exists in some of the other platform drivers (i8042) but yes it is a bit as you say, a bit of a "fuzzy area" in that it can in some ways be interpreted as "user space logic", underscored by the fact that even Samsung implemented this logic as a user space feature (a Windows Service).

IMHO and a bit as you say, the best case would be if the various major Desktop Environments (GNOME, KDE, Xfce, etc) could unify on how a single brightness toggle key should work in cases where there are multiple fixed levels (for example maybe the "round-robin" cycle is the one that makes the most sense) and then incorporate some kind of standardized logic to handle this?

closed 2024-10-16 by joshuagrisham