joshuagrisham/samsung-galaxybook-extras #6: Include the sound fix in this scripts

Rodsevich, 2024-02-12:
There's a fix for the non-working sound in this forum and so on:
https://forum.manjaro.org/t/howto-set-up-the-audio-card-in-samsung-galaxy-book/37090/47

Does this repo add support for the verbs yet? How did you managed to fix that?

joshuagrisham, 2024-04-20:
Hi @Rodsevich regarding the link that you shared, the "script" that was given by the other user was actually also created by me as well (see https://github.com/joshuagrisham/galaxy-book2-pro-linux/tree/main?tab=readme-ov-file#sound and https://github.com/joshuagrisham/galaxy-book2-pro-linux/blob/main/sound/necessary-verbs.sh), but what the script is, is actually quite "hacky", as it is basically a "replay" of the entire section from the log of all of the HDA verbs that were sent to the device during a very early capture session from me, and I don't know what everything in it is actually doing.

There is evidence that repeated usages of the script actually cause degradation in volume levels (see: https://github.com/thesofproject/linux/issues/4055#issuecomment-1366387548) and even one user who indicated that it might have caused physical damage to his speakers (see: https://github.com/thesofproject/linux/issues/4055#issuecomment-1704340009)

I have made quite a bit of my own effort trying to narrow down exactly what is required and even taken something like 10-20x different captures in a row and tried to compare them all to see if I can filter out only the exact similarities between them that seemed to "turn on" the speakers, but was never really able to come to anything 100% solid unfortunately. To top it off I don't have a super strong understanding of Intel's HDA and DMA in general so I struggle with things like knowing how to "reset" the buffer so that I can run clean tests from the start each time -- this makes testing slow and inconsistent as it is often that the "buffer" still has stuff in it from prior tests so I get a lot of false positives and false negatives.

Having said all of that, my ultimate goal was to either add a feature in this platform driver and/or a new Realtek Patch and/or enhancement to the existing [patch_realtek.c](https://github.com/torvalds/linux/blob/master/sound/pci/hda/patch_realtek.c) so that the speakers will just "work" as they should, and then the script approach would not be necessary.

mhagnumdw, 2024-10-15:
If there is still no definitive solution, what is the latest workaround to get the sound working? My kernel version: `6.10.12-200.fc40.x86_64`

I currently use this: `curl -s -L https://raw.githubusercontent.com/joshuagrisham/galaxy-book2-pro-linux/main/sound/necessary-verbs.sh | sudo bash`

If I can help with anything testing, let me know.

Galaxy Book3 Ultra, Product Name NP960XFH-XA1BR

mhagnumdw, 2024-10-15:
I just updated to kernel `6.11.3-200.fc40.x86_64` and the sound started working after a restart.

Now I can't be sure if the new kernel already has the driver or if it's some cache from the command in the previous post.

jcarrano, 2024-10-15:
@mhagnumdw try a full power-off/power-on. It does **not** work for me with a 6.11.3.

mhagnumdw, 2024-10-15:
> @mhagnumdw try a full power-off/power-on. It does **not** work for me with a 6.11.3.

@jcarrano , for me the sound continued to work after turning the laptop off and on.

joshuagrisham, 2024-10-15:
> I just updated to kernel `6.11.3-200.fc40.x86_64` and the sound started working after a restart.
> 
> Now I can't be sure if the new kernel already has the driver or if it's some cache from the command in the previous post.

Yes, my update to mainline will release in 6.12 but was also added to 6.11.3 (stable backport patch) so I think it should actually be there.

Please take a look here: https://github.com/joshuagrisham/galaxy-book2-pro-linux/blob/main/sound/necessary-verbs.sh and here: https://github.com/thesofproject/linux/issues/4055#issuecomment-2323411911

@jcarrano if your speakers are still not working then I can probably help you to get them working (and we can get them added in a patch to mainline :sunglasses: )

jcarrano, 2024-10-15:
My speakers work with the necessary-verbs.sh. I'm using Artix (based on Arch) which has very few patches on top of mainline: https://github.com/archlinux/linux/commits/v6.11.3-arch1/ . Fedora seems to have more patches but I cannot find anything related to  SOF or the Galaxy in https://src.fedoraproject.org/rpms/kernel/c/0f493006a3c93ce0847a3120257709ea031aefce?branch=f40

joshuagrisham, 2024-10-15:
The Github link you referred to does have my updates and the payloads that exist there should be exactly the same as what comes from my necessary-verbs.sh script, but my guess is that your device is not added to the quirk table?

https://github.com/archlinux/linux/blob/d64f49940d40607903d9d81eaa36363cf6418414/sound/pci/hda/patch_realtek.c#L10695-L10698

What is the value you see if you run `cat /sys/class/sound/hwC0D0/subsystem_id` ? Also do you know if your device has 4 speakers or just 2?

jcarrano, 2024-10-17:
Thanks. My device is not there: the id is `0x144d c872`. I believe I have only two speakers. As far as I'm aware the XEE variant is the same as the XED but with an ARC gpu. Is there a way to try out the quirk without patching and compiling the kernel again?

joshuagrisham, 2024-10-18:
Yes @jcarrano I think so, can you try something like this:

```
sudo tee /etc/modprobe.d/audio-fix.conf <<< 'options snd-hda-intel model=alc298-samsung-amp-v2-2-amps'
```

And then reboot or even power off/power on?

(keeping in mind this will only work with 6.11.3 or higher)

And then if/when your device is added you would be able to remove the file `/etc/modprobe.d/audio-fix.conf`

Regarding if you should use the 2-amps vs 4-amps version, well I usually actually try to find Youtube videos or similar where someone has removed the backplate and you can actually see the speaker drivers; if there are 4, it is probably 4.. if there are 2, then it is probably 2 :laughing: 
And absolute worst-case (or best-case?? depending on your perspective :wink:  ) scenario, you can always do a trace using the Windows driver and see what exact payloads are being sent to your device and then it will be 100% clear

mhagnumdw, 2024-10-23:
It's happened to me twice recently that the sound stopped working, input and output. Has anyone else had the problem below?

`journalctl -f`:

```
out 23 12:06:59 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x60050000 (msg/reply size: 12/0)
out 23 12:06:59 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
out 23 12:06:59 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x80000000 intlctl 0x40000081 rirb 00
out 23 12:06:59 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x80000000 adspis 0x00000001
out 23 12:06:59 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x80000000 dsp status 0x00000000 mask 0x00000003
out 23 12:06:59 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
out 23 12:06:59 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
out 23 12:06:59 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
out 23 12:06:59 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
out 23 12:06:59 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x00000005: module: ROM, state: FW_ENTERED, running
out 23 12:06:59 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: unexpected fault 0x00000000 trace 0x00004000
out 23 12:06:59 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
out 23 12:06:59 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
out 23 12:06:59 dwnote2 kernel:  HDA Analog: ASoC: trigger FE cmd: 0 failed: -110
out 23 12:06:59 dwnote2 menulibre-discord-tar.desktop[287830]: 12:06:59.936 › [RTCControlSocket(default)] Sending heartbeat with last received sequence number: 57
out 23 12:07:00 dwnote2 menulibre-discord-tar.desktop[287830]: 12:07:00.036 › [RTCControlSocket(default)] Heartbeat ACK received
out 23 12:07:00 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx error for 0x60040000 (msg/reply size: 12/0): -16
out 23 12:07:00 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -16
out 23 12:07:00 dwnote2 kernel:  HDA Analog: ASoC: trigger FE cmd: 1 failed: -16
out 23 12:07:00 dwnote2 pipewire[2912]: spa.alsa: hw:sofhdadspp: snd_pcm_start: Device or resource busy
out 23 12:07:00 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x60050000 (msg/reply size: 12/0)
out 23 12:07:00 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump start ]------------
out 23 12:07:00 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: hda irq intsts 0x00000000 intlctl 0xc0000081 rirb 00
out 23 12:07:00 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: dsp irq ppsts 0x00000000 adspis 0x00000000
out 23 12:07:00 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: error: host status 0x00000000 dsp status 0x00000000 mask 0x00000003
out 23 12:07:00 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ IPC dump end ]------------
out 23 12:07:00 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump start ]------------
out 23 12:07:00 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
out 23 12:07:00 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: fw_state: SOF_FW_BOOT_COMPLETE (7)
out 23 12:07:00 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: 0x00000005: module: ROM, state: FW_ENTERED, running
out 23 12:07:00 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: unexpected fault 0x00000000 trace 0x00004000
out 23 12:07:00 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ------------[ DSP dump end ]------------
out 23 12:07:00 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
out 23 12:07:00 dwnote2 kernel:  HDA Analog: ASoC: trigger FE cmd: 0 failed: -110
out 23 12:07:01 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x60050000 (msg/reply size: 12/0)
out 23 12:07:01 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
out 23 12:07:01 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ASoC: error at soc_component_trigger on 0000:00:1f.3: -110
out 23 12:07:01 dwnote2 kernel:  DMIC: ASoC: trigger FE cmd: 0 failed: -110
out 23 12:07:01 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: ipc tx timed out for 0x60040000 (msg/reply size: 12/0)
out 23 12:07:01 dwnote2 kernel: sof-audio-pci-intel-tgl 0000:00:1f.3: IPC timeout
```

I need to reboot to get back to normal.

Does it make sense to open an issue for this?

- Fedora 40
- Kernel: 6.11.3-200.fc40.x86_64
- Galaxy Book3 Ultra, Product Name: NP960XFH-XA1BR

joshuagrisham, 2024-10-24:
Hi @mhagnumdw just curious is this with using the driver in 6.11.3+ or is this with running the script one time after starting up?

mhagnumdw, 2024-10-24:
Hi @joshuagrisham, I no longer need to run the verb script. The driver is running.

raelgc, 2024-12-15:
I just got the Ultra Book4 (`960xgl-xg1`). The `necessary-verbs.sh` script didn't worked for me. Can I help providing any additional info?

pyxfluff, 2024-12-26:
If any of your are on Arch, I got my sound working by simply running `sudo pacman -S sol-firmware` while using Pipewire.

raelgc, 2024-12-29:
@pyxfluff Which kernel? And I imagine you're talking about `sof-firmware`.

pyxfluff, 2024-12-29:
> @pyxfluff Which kernel? And I imagine you're talking about `sof-firmware`.

Zen kernel

joshuagrisham, 2025-01-24:
@raelgc  if it did not work for you to install `sof-firmware` then my suggestion would be to capture some traces using a Windows VM with QEMU.

You will have to forward the device to the VM and also download and install the drivers for the sound card within the Windows VM before you can start to take the traces. Ideally take traces on a fresh boot after everything is set up..

Here is a link to more info that I recommended on doing this: https://github.com/thesofproject/linux/issues/4055#issuecomment-2349301491

raelgc, 2025-01-24:
@joshuagrisham By coincidence, I was exactly trying that yesterday (read a bunch about it, thanks for all your notes). I've tried with a Win10, but the main issue is: I could not get the Windows sound to work (even with all drivers installed).