joshuagrisham/samsung-galaxybook-extras #7: Issue with load modprobe

chm10, 2024-03-30:
I tried to load the driver but have this issue.
```
sudo modprobe samsung-galaxybook
libkmod: ERROR ../libkmod/libkmod-config.c:712 kmod_config_parse: /etc/modprobe.d/blacklist.conf line 54: ignoring bad line starting with 'snd-hda-intel'
modprobe: ERROR: could not insert 'samsung_galaxybook': Key was rejected by service

```

6.5.0-27-generic
VERSION="23.10 (Mantic Minotaur)"
Samsung Book 360

closed 2024-03-30 by chm10

jcarrano, 2024-03-30:
Does it have to do with secure boot or with having kernel module signing required?

simon-ament, 2024-03-30:
I was facing the same issue and managed to resolve it without disabling SecureBoot by just signing the module. You first have to enroll a Machine Owner Key (MOK) on your machine and then sign the module using this key and some suitable scripts (either preinstalled on your device or available online).
For the exact procedure see articles like this one:
https://ubuntu.com/blog/how-to-sign-things-for-secure-boot

chm10, 2024-03-31:
> Does it have to do with secure boot or kernel module signing required?

With kernel module. I installed the driver, but the backlights and other features not take effect. The compiled kernel has not changed anything on my Samsung 360 book.

chm10, 2024-03-31:
> I was facing the same issue and managed to resolve it without disabling SecureBoot by just signing the module. You first have to enroll a Machine Owner Key (MOK) on your machine and then sign the [...]

Thank you. I installed the drivers but wondered if I need to sign the module. It looks like the compiled kernel does not take effect.

jcarrano, 2024-04-01:
@chm10 I doubt you will be able to sign modules without rebuilding the kernel. Distros will generate a one-time key to sign modules for each kernel build and delete the private key afterwards.