joshuagrisham/samsung-galaxybook-extras #9: Installation troubles

AntoninMayet, 2024-04-16:
I've tried to install all the drivers. I have followed the procedure in the README.md in the `driver` folder but it never worked. It has always returned:
> FATAL: Module samsung-galaxybook not found in directory /lib/modules/6.5.0-27-generic

If someone could help me I would really appreciate it

[image]

jcarrano, 2024-04-17:
does it work if you use `insmod`?

joshuagrisham, 2024-04-20:
Hi @AntoninMayet I have not really intended to cover how one builds and loads an out-of-tree module in my instructions as the topic can vary and depend a lot on each person's distribution, version, kernel etc etc etc (it can be a big topic! with lots of info here: https://docs.kernel.org/kbuild/modules.html) but the quick steps given here: https://github.com/joshuagrisham/samsung-galaxybook-extras/blob/main/driver/README.md should probably work for Ubuntu assuming that you have all of the right dependencies

One thing from your screenshot was that it does not show if/where you ran make by itself to actually compile the module first? Per the readme:

```sh
make -C /lib/modules/`uname -r`/build M=$PWD
```
This should be the first step and then you should run `make modules_install` after.

In case you have upgraded kernels then I have found that it helps to first clean out previous compiled versions of the driver .. Today I added a crude `clean` target to the Makefile that you can run as the first step to help with this if you wanted pull a new copy of the repo and try that too? (`make clean`)

klausagnoletti, 2024-04-22:
I have more or less the same problems on PopOS:
```
klaus@slambook:~/Documents/git/samsung-galaxybook-extras/driver$ make -C /lib/modules/`uname -r`/build M=$PWD
make: Entering directory '/usr/src/linux-headers-6.8.0-76060800daily20240311-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
make: Leaving directory '/usr/src/linux-headers-6.8.0-76060800daily20240311-generic'

```
I interpret this as if the module has been built, however upon running `make modules_install`, the fun stops:
klaus@slambook:~/Documents/git/samsung-galaxybook-extras/driver$ sudo make modules_install
```
make -C  M=/home/klaus/Documents/git/samsung-galaxybook-extras/driver modules_install
make[1]: *** M=/home/klaus/Documents/git/samsung-galaxybook-extras/driver: No such file or directory.  Stop.
make: *** [Makefile:8: modules_install] Error 2
```
How do I know if all deps are fulfilled?

Thanks for all the great work you've done on this @joshuagrisham :-)

axcosta, 2024-05-19:
I've got Galaxy Book Pro (1st version) [950XDB/951XDB/950XDY, Version: P11RFW] running Zorin 17 (which is based on Ubuntu Jammy 22.04 LTS). 
Had gone successfully through the driver compilation/installation but, unfortunately, keyboard backlight Fn key is not working and neither are present the folders `\sys\lclass\eds\samsung...` nor `\sys\devices\platform\samsung`, for instance. Had not seen any comment about incompatibility with this Galaxy Pro version; is there any known restrictions?

PremiumUsername, 2024-05-19:
I can build and load the the module ok, but it doesn't seem to do anyhting. Nothin new in `/sys/class/leds/`.  I have a Galaxy book ion 15.6 (950XCJ), running Garuda Linux.  I'm not sure how to trouble shoot this. If there is anything I can do to help the effort, let me know.

joshuagrisham, 2024-07-15:
Hi @axcosta and @PremiumUsername , sorry for my delay in getting back to you!

I built this driver by reverse engineering + debugging how it works in Windows on my Galaxy Book2 Pro 15.6 so can never be 100% sure it is going to work exactly the same on other devices.

Having said that, the driver works by executing methods on a Samsung ACPI device (the one they called `SCAI`) which seems "relatively" standardized across several models.

On my particular device, the ACPI Device ID for the `SCAI` device is `SAM0429` but when I was looking into this in Windows I saw that the same driver on my device was matching both `SAM0427` and `SAM0429`. So far I have only put `SAM0429` in the code for this platform driver, as it is the only device I have access to test and would not want to make it even easier for anyone to accidentally harm their device :wink: 

But, as said, it probably works relatively out-of-the-box with `SAM0427` and possibly other device IDs as well, and because it is using ACPI then the most likely scenario is that either the device method names are different or require different payloads and will just reject/not respond to anything that is sent that would not work with other devices. However, if Samsung made other devices that happen to use exactly the same `SCAI` device, with exactly the same device methods (`CSFI` and `CSXI`), that can accept the exact same kind of input parameters (so that it is not rejected by the ACPI subsystem and actually executes the code in the BIOS), but behave completely differently and/or would "harm" the device somehow? Hmm I guess it is entirely possible but it seems pretty unlikely (I certainly would hope, anyway! :smile: )

If you run something like `ls /sys/devices/platform/ | grep -i SAM` do you see a match? If so, what is the device ID that shows up there? (for me it is `SAM0429:00`) Perhaps we can try adding it to the code and you can test to see if the driver will work on your device?

If there is no match, and/or you don't see anything under `/sys/devices/platform/` that look promising, how does it look in Windows on your device?  Specifically, is there a device driver installed called `Samsung System Event Controller`, and if you look at its properties / INI file can you tell what it has for device ID / vendor ID / product ID / etc ?

axcosta, 2024-07-15:
Hi @joshuagrisham, running `ls /sys/devices/platform/ | grep -i SAM`, it returns `SAM0428:00`. It is a matter of changing these lines (below) to test it?

```
static const struct acpi_device_id galaxybook_device_ids[] = {
	{ "SAM0429", 0 },
	{ "", 0 },
};
```

joshuagrisham, 2024-07-15:
Yes @axcosta actually better to add it as another entry to the array like this:

```c
static const struct acpi_device_id galaxybook_device_ids[] = {
	{ "SAM0428", 0 },
	{ "SAM0429", 0 },
	{ "", 0 },
};
```

PremiumUsername, 2024-07-16:
Thank you @joshuagrisham ! Adding `{ "SAM0427", 0 }` to the array seems to be working. I can now control my kb backlight. I haven't had time to try anything else yet, but so far so good.

PremiumUsername, 2024-07-16:
Sadly on further experimentation I now see that with this module I can no longer put suspend or hibernate my laptop:

```
jul 16 11:22:45 david-ion15 systemd-sleep[1696]: Failed to put system to sleep. System resumed again: Device or resource busy
jul 16 11:22:45 david-ion15 systemd-udevd[486]: SAM0427:00: Worker [502] processing SEQNUM=3718 is taking a long time
```

It also hangs when I try to restart or shutdown the laptop. Still I'm very happy the intitial success for the keyboard BL.  I still can't read fan levels.  Could that be the problem perhaps?

axcosta, 2024-07-17:
> Yes @axcosta actually better to add it as another entry to the array like this:
> 
> ```c
> static const struct acpi_device_id galaxybook_device_ids[] = {
> 	{ "SAM0428", 0 },
> 	{ "SAM0429", 0 },
> 	{ "", 0 },
> };
> ```

@joshuagrisham, I have tested on mine after this change, and it's working alright with no side effects so far.

joshuagrisham, 2024-07-18:
> Sadly on further experimentation I now see that with this module I can no longer suspend or hibernate my laptop:
> 
> ```
> jul 16 11:22:45 david-ion15 systemd-sleep[1696]: Failed to put system to sleep. System resumed again: Device or resource busy
> jul 16 11:22:45 david-ion15 systemd-udevd[486]: SAM0427:00: Worker [502] processing SEQNUM=3718 is taking a long time
> ```
> 
> It also hangs when I try to restart or do a shutdown. Still I'm very happy the intitial success for the keyboard BL. I still can't read the fan levels. Could that be the problem perhaps?

Hi @PremiumUsername I have been labbing quite a few different things over a few months, but just now pushed a relatively big change to the code (the biggest "change" was refactoring and then adding module parameters to control each "feature" so you can disable anything you don't want to use (or does not seem to work right)).

I thought this was especially important if people start to run this on other devices! Please feel free to enable/disable various features and let me know which ones work vs do not work. You can also enable the debug parameter to get a lot more info in the kernel log (i.e. `dmesg`).

Example:

```sh
sudo rmmod samsung-galaxybook
sudo modprobe samsung-galaxybook debug=true kbd_backlight=on performance_mode=off fan_speed=off i8042_filter=off acpi_hotkeys=off wmi_hotkeys=off
```

I am thinking longer term it would be good to maybe set up some flags or similar per device id to maybe enable/disable certain things depending on which device is present, so it would be super if you could report back which things seem to work well vs not on your device!

PremiumUsername, 2024-07-18:
I think the modular approach is a great idea @joshuagrisham.  For my device, which uses 'SAM0427', The `i8042_filter=on` makes the system crash for fn+F9 and `fan_speed=on` breaks suspend, reboot and shutdown with:

```
[  553.400512] BUG: kernel NULL pointer dereference, address: 0000000000000000
[  553.400514] #PF: supervisor read access in kernel mode
[  553.400516] #PF: error_code(0x0000) - not-present page
```

With your latest changes I can now use this module when I disable the troublesome parts, so I thank you for your efforts. 

I have question about how the screen brightness is regulated.  My laptop has an ultrabright mode (fn+F10) in windows, it seems to me that in linux regular brightness can go further then regular brighness in windows but I don't know if the max is the same thing as the ultrabright settings that exist in windows. I'm genuinly unsure  whether it is or not. Do you know if the max brightness in linux is the same as in windows even without the an ultrabright settings explicitly being enabled?

joshuagrisham, 2024-07-19:
That is really good feedback @PremiumUsername , we can for example now maybe add SAM0427 to the id table with some flags so that those two are automatically disabled without having to set anything for anyone else who has SAM0427

I am curious what the problem is however, can it be that something is totally different with your device?

If you have the patience and tolerance to run again knowing that it will fail, it would be interesting if you could set `debug=on` and then run these breaking things again, and send a full copy of the dmesg or journalctl ? I am mostly interested in the `samsung_galaxybook` messages but maybe there could be other messages during startup and/or at the same time of the failures with other devices that would give some clues?

Regarding fan_speed it would actually also be helpful to get a copy of your DSDT ... my suspicion is that maybe the ACPI devices or fields on your device have slightly different names or something like that?

Regarding the laptop screen brightness I cannot say.. I know on my device (with OLED screen, not sure if yours has one or LCD?) there is an HDR+ kind of mode you can set in Samsung Settings in Windows but I am not sure exactly how it works or what it changes compared to the base brightness of the screen backlight itself.. I guess it would need to be debugged and best-case is if someone had some kind of device to measure the screen brightness to actually be able to compare the differences :sunglasses:

PremiumUsername, 2024-07-19:
DSDT:
[dsdt.dsl.txt](https://github.com/user-attachments/files/16309936/dsdt.dsl.txt)

I don't know if I'm abe to get a log from the fn+F9 system freeze. I didn't see anything from the journal for the last boot after that happened, but I can try again at a later date.

Looking at the fan problem I can see:
```
jul 18 09:57:33 david-ion15 kernel: samsung_galaxybook: initializing fan speed
jul 18 09:57:33 david-ion15 kernel: RIP: 0010:galaxybook_acpi_add.cold+0x311/0x82a [samsung_galaxybook]
jul 18 09:57:33 david-ion15 kernel:  ? galaxybook_acpi_add.cold+0x311/0x82a [samsung_galaxybook 8c51852e42d15def19f9b09e82211ba1f677f7c9]
jul 18 09:57:33 david-ion15 kernel:  ? galaxybook_acpi_add.cold+0x300/0x82a [samsung_galaxybook 8c51852e42d15def19f9b09e82211ba1f677f7c9]
jul 18 09:57:33 david-ion15 kernel:  ? __pfx_samsung_galaxybook_init+0x10/0x10 [samsung_galaxybook 8c51852e42d15def19f9b09e82211ba1f677f7c9]
jul 18 09:57:33 david-ion15 kernel:  samsung_galaxybook_init+0x39/0xff0 [samsung_galaxybook 8c51852e42d15def19f9b09e82211ba1f677f7c9]
jul 18 09:57:33 david-ion15 kernel:  ? __pfx_samsung_galaxybook_init+0x10/0x10 [samsung_galaxybook 8c51852e42d15def19f9b09e82211ba1f677f7c9]
```

If need be I can give you a more thorough log later, but I'm a bit pressed for time just now.

Concerning the screen brightness, it really is a brightness setting (I think it's called outdoor mode in Samsung settings in Windows) and not a HDR+ setting (which also exists).  I think it might be different for your device, mine is a lcd screen which is supposed to go up to 600 Nits when you activate the outdoor mode. I did actually test it by measuring the brighness for max brightness for a white background for both windows and linux, and although my method to measure it was pretty crude, I'm confident that it is a real difference. My measurments for max brightness without outdoor mode activated was the same for windows and linux so that also suggests that my conlusion is sound.

Zippy-boy, 2024-07-20:
Hello. So I added the correct device ID (galaxy book 1 pro 360) (Fedora workstation) but when I try and use the make command I get the error `make: *** /lib/modules/6.9.8-200.fc40.x86_64/build: No such file or directory.  Stop.`

PremiumUsername, 2024-08-22:
Was the DSDT file the right file @joshuagrisham? I had to rename it for github to accept the upload.

joshuagrisham, 2024-09-15:
@axcosta I have now pushed an update which includes support for `SAM0428` with no other changes -- would you mind to give it a try without any manual edits and see how it works for you?

joshuagrisham, 2024-09-15:
> Was the DSDT file the right file @joshuagrisham? I had to rename it for github to accept the upload.

Hi yes, thank you @PremiumUsername ! Sorry for my delay in getting back to you -- I was focused on trying to help get the speaker amps working (now there is a patch that should be included in kernel 6.11 or 6.12+ I think?? with support for several but not all devices  :partying_face:  )

I did take a look at your DSDT and it is exactly the right format. I unfortunately could not find any device or methods that seem to be related to fan speeds in your DSDT -- do you already get fan speeds reported in Windows and/or Linux somehow?  Does your device have any fans at all or is it completely passively cooled/silent?  Otherwise it seems like Samsung have not followed the ACPI standard at all for Fan devices on your device (https://uefi.org/specs/ACPI/6.5/11_Thermal_Management.html#fan-device) 

On my device they have "some" of the right devices and methods but there are bugs (exceptions thrown by the BIOS when you try to execute some of the methods etc) and it is not 100% according to the standard even, but at least I was able to fix together "something" that kind of resembled fan speed support :laughing: 

What I have done for now in the latest push is added your ACPI device ID `SAM0427` but also added a framework for setting quirks, and for your device ID I have added a quirk to always disable the fan speed and the i8042 keyboard filter. Can you please give this a try when you have a chance (without changing anything and without setting any module parameters) and see if it works for your device?

Regarding the keyboard filter it seems a little strange -- what key(s) does your device have for adjusting the keyboard backlight?  Is it just a single key toggle as Fn+F9 or are there different key(s) on your device? Do they work without this i8042 filter or are you controlling the kbd backlight in a different way? Out of curiosity do you have the scan code and event code for the key(s) on your device? :smile_cat:

PremiumUsername, 2024-09-17:
I don't know about Windows but I don't have any fans sensors being reported in linux by default and it isn't fanless. 

The new drivers works as you intended. I no longer need to specify things to be excluded. 

Regarding the keyboard filter, `evtest` gives me the value `ac` for the keyboard backlight fn+F9 combination. Other fn keys are strange (fn+F3 is `left ctrl` I think) or just plain missing so I don't use them.
I use the window manager Sway so making my own keyboard controls to control the backlight is not complicated, that's what I'm using now. 

Did you see anything about the "Outdoor mode" in the DSDT?

joshuagrisham, 2024-09-18:
Hi @PremiumUsername thank you for getting back!

Regarding your DSDT unfortunately I did not see anything that immediately jumped out, but when I was looking into another issue I noticed that your "performance mode" stuff seems totally different.

Actually from what I can tell your ACPI will just ignore the payloads that are being sent by this driver as of right now.

First question: "normally" with your device, how many "performance modes" do you have? I am actually wondering if it is only 2 (e.g. "on" vs "off" or similar).

And in addition to the answer to that question, I think it would be interesting if you were willing to run some test commands using the `acpi_call` module and paste back what the return values are?

Here is a "quick" little set of commands I cooked up just based on quickly glancing through the DSDT.

```sh
# test SAM0427 performance mode ACPI methods

# 1. load acpi_call module
sudo modprobe acpi_call

# 2. enable ACPI device
echo '\_SB.SCAI.SDLS' 1 | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call

# 3. enable "power management" feature
echo '\_SB.SCAI.CSFI' b43587a0000bbaa0000000000000000000000000000 | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call

# 4. get supported modes ??
echo '\_SB.SCAI.CSXI' b43589100008d024682ca8b554aba0f6f1e6b921b8f5000 | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call

# 5. Not sure what this does ?? is it "value" per supported mode or does it toggle some value ??
echo '\_SB.SCAI.CSXI' b43589100008d024682ca8b554aba0f6f1e6b921b8f5001 | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call

# 6. Not sure what this does ?? is it "set current mode" or does it set a mode ??
echo '\_SB.SCAI.CSXI' b43589100008d024682ca8b554aba0f6f1e6b921b8f5002 | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call

# 7. Not sure what this does ?? looks like it is just reporting back either a 0 or 1 to IOB0 depending on some condition value ??
echo '\_SB.SCAI.CSXI' b43589100008d024682ca8b554aba0f6f1e6b921b8f5003 | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call

# note based on the UUID then I think the only method that gets immediately called from above call to CSXI is HIPF; ELBC and EPWS match on different UUIDs

# it KIND OF seems like 01 and 02 just toggle something back and forth (01 for one mode, 02 for the other) but it is a bit hard to tell just from glacning through ... ?
```

Tests 1 through 4 you definitely only need to run once at the beginning, but the rest are kind of interesting. I wonder if you get a list of values as the final values in the return payload from the one ending in "5001" that you might be able to even append to the one in "5002" (e.g. if you get values like 0x0a, 0x0b, 0x02 etc to try with "50020a" etc) but maybe it is good enough to just run then maybe in a "loop" (e.g. test 5, 6, 7) and then run a few multiple times (e.g. tests 5, 5, 5, 6, 6, 6, 7, 7, 7) and just see what all of the results were from them (and if any specific values change??)

Even though as of right now the driver does not really "complain" I think it is just because your ACPI is just completely ignoring its attempts to execute certain payloads... (so nothing is actually happening with your fans etc)

joshuagrisham, 2024-09-19:
@axcosta or anyone else with `SAM0428` would you mind to also share a copy of your DSDT file? I am curious about how it looks for the above stuff as well plus if there is maybe anything else that might not be exactly the same :)

axcosta, 2024-09-19:
Hi @joshuagrisham , sorry, have been quite busy these days. Will respond accordingly soon.

PremiumUsername, 2024-09-21:
I don't have a specific keyboard button to switch between performance modes @joshuagrisham. In windows I can use "Samsung Settings" to set the laptop to "Silent Mode". That is the only samsung option for performance I'm aware of. I tried to perform the commands you suggested but since I'm not really sure what you are looking for there is a good posibility that I performed them incorrectly. You tell me:

[acpi_call.txt](https://github.com/user-attachments/files/17085531/acpi_call.txt)

I can't detect any difference when performing these commands, maybe I need some other kernel parameters. 

One question. What is it that the samsung way to alter the performance does that can't already be done by modifying Intel performance and energy bias hints directly (or the AMD eqv)?

joshuagrisham, 2024-09-22:
Hi @PremiumUsername this was perfectly executed :) The result was semi-disappointing in that each method call just returned the same buffer you sent to it but at least it is "some" info we can glean from this.

Regarding "what do you get from this" I would say is that Samsung has packaged together what they call "Silent Mode" on your device as something that they think is good from a user experience perspective. This "Silent Mode" probably includes a list of things that toggle off/on or adjust including power sent to the CPU, potentially lowering max clock or other similar things, lowering if not aiming to completely turn off the cooling fan(s), etc, but possibly even more than this? Yes, you can probably achieve some of the same result going directly to the processor but it will be in more steps, and you may need to trial-and-error a bit to figure out what works well vs what does not. You also might not be able to affect the fans in the same way as Samsung might have done in the firmware, since there does not seem to be an ACPI device for your fan controller from what I can tell? So in the end it is a more "pre-tested and packaged" solution you get from Samsung for what their view is of a "silent mode" vs non-"silent mode". Is it perfect? Probably not.. but useful? Maybe... depends on what you want from it :)

But as of right now there is nothing in Linux on your device that is using this method in the right way.. but maybe we can change that?

One thing I found interesting was that you said it is called "Silent Mode" in the Samsung Settings app, but the method in your ACPI that would actually eventually be doing stuff on your device with this is called `HIPF`, which I am clumsily assuming stands for something like "High Performance". So "interesting" that the naming is a bit opposite, so I wonder which value actually does what?

If it actually works, you should be able to tell a difference when stress-testing your CPU.. both in the performance results but also hopefully how loud the fan is.

I am curious if you can try to run steps 1 through 4 of my previous commands (as a pre-setup basically):

```sh
# 1. load acpi_call module
sudo modprobe acpi_call

# 2. enable ACPI device
echo '\_SB.SCAI.SDLS' 1 | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call

# 3. enable "power management" feature
echo '\_SB.SCAI.CSFI' b43587a0000bbaa0000000000000000000000000000 | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call

# 4. get supported modes ??
echo '\_SB.SCAI.CSXI' b43589100008d024682ca8b554aba0f6f1e6b921b8f5000 | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call
```

And then run a "stress test" with passing 0x01 vs 0x02, maybe sort of like this:

```sh
# call method with SUBN=0x01
echo '\_SB.SCAI.CSXI' b43589100008d024682ca8b554aba0f6f1e6b921b8f5001 | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call

# wait for the dust to settle a bit
sleep 60

# run a stress test with SUBN=0x01
sudo stress-ng --cpu 0 --cpu-load 100 --metrics-brief --perf -t 60
```

Then wait a while for your CPU to settle and cool down a bit, maybe like 5 minutes.. then do it again with SUBN = 0x02 like this:

```sh
# call method with SUBN=0x02
echo '\_SB.SCAI.CSXI' b43589100008d024682ca8b554aba0f6f1e6b921b8f5002 | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call

# wait
sleep 60

# run a stress test with SUBN=0x02
sudo stress-ng --cpu 0 --cpu-load 100 --metrics-brief --perf -t 60
```

It would be interesting to see if there is a noticeable difference in the two tests and also if you observe any difference in how loud the fan seems to be in one mode vs the other :)

PremiumUsername, 2024-09-24:
> It would be interesting to see if there is a noticeable difference in the two tests and also if you observe any difference in how loud the fan seems to be in one mode vs the other :)

There is no discernable difference regarding fan noise and  CPU Cycles and  Total cycles performed are also almost the same. 

When you perform this performance mode change for your working setup, does it change anything that you can see directly, e.g. something you can see in `/sys/devices/system/cpu/cpu1/cpufreq`?

axcosta, 2024-10-15:
> @axcosta I have now pushed an update which includes support for `SAM0428` with no other changes -- would you mind to give it a try without any manual edits and see how it works for you?

Had just done that and it's working perfectly.

axcosta, 2024-10-15:
> @axcosta or anyone else with `SAM0428` would you mind to also share a copy of your DSDT file? I am curious about how it looks for the above stuff as well plus if there is maybe anything else that might not be exactly the same :)

How to do that DSDT file?