joshuagrisham/samsung-galaxybook-extras #12: Driver functionality fails after resuming from hibernate

bhargavg1, 2024-05-25:
Hello,

My device is a galaxy book 2 360 running debian 12, and overall this driver has been working, but I noticed that it completely stops working after I resume from hibernation. None of the features work until i manually ```rmmod``` the module and ```modprobe``` it to reload it again, and then it starts working.

I thought this was strange and have been looking at the source code for a while and running a few tests (just throwing a bunch of println statements everywhere), I noticed that you had some TODO comments about reading the acpi function's return buffer after every call. I implemented this and found a few strange things.
1. The return does match perfectly when the driver is working, but doesnt match when it stops working (i.e. when I resume from hibernation) -> The return buffer has ```0x43, 0x58, 0x78, 0x00, 0xaa, 0xff, 0x00, etc etc```. The ```0xff ```there is weird, and also the 7th position is ```0x00``` when it should have ```brightness + 0x80```. The long line of 0x00's at the end are all there as normal.
2. The return buffer is actually _48_ bytes for some reason, the first 24 bytes are mostly 0x00's though, but there are some changing numbers there that change without any noticable pattern (at least one I cant figure out). The second set of 24 bytes is where the expected reply from the TODO lives.
Those first 24 bytes generally conform to ```0x03, 0x00, 0x00, 0x00, 0x15, 0x00, 0x00, 0x00, X1, X2, Y1, Y2, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00```, where ```X1``` and ```X2``` can change after every backlight change, and ```Y1``` and ```Y2``` only seem to change after a restart and stay the same no matter how many times the module is reloaded.

In my experimenting with the driver code, I only fiddled with the keyboard backlight part, as I am not that experienced with C nor am I experienced with writing kernal modules. This whole problem could just also be a problem with my setup though. I have not entirely ruled out that I am just configuring something wrong. I guess I could just continue unloading and reloading the module everytime I turn on my computer after a hibernate if thats the case.

joshuagrisham, 2024-07-15:
Thanks @bhargavg1 I am super stoked to have other folks like you getting into the nitty-gritty on this and trying to iron out issues :slightly_smiling_face: This is exactly the kind of case I was thinking when I wrote the TODO in the comments, e.g. maybe it is worth to check the response payload and is there anything that can be done as a mitigation in the case of certain responses (e.g. if it is a "failure" response do we need to run some init logic again ? etc)

I have also seem some "weirdness" when the laptop starts to go to sleep / wakes up, but mostly that the fan starts to go crazy :laughing:  however, in my local version I have added several other things to try and figure out how to capture Fn+F11 to drive the performance mode via the hotkey so I think some of my extra issues are actually related to that...

I am glad you brought this up because I think it is worth looking into further. I will put it on my internal TODO-list to try and take a look at as well; my first guess is to try and maybe do something like I just mentioned: check response of payload; if bad, send (some or all of) init, and then send original command again ?

joshuagrisham, 2024-10-16:
Hi @bhargavg1 , there have been a lot of updates to the driver since you reported this issue.. I wonder if you try to use the latest version, do you still have these problems?

Viverun, 2026-11-12:
I hit this on a Galaxy Book4 (750XGK, SAM0430, BIOS P11CFP.025): after restore from hibernation every CSFI call fails with 0xff (battery threshold, kbd backlight, firmware-attributes return -ENXIO) until the driver is re-bound. I've sent a patch to platform-driver-x86 that adds a resume callback repeating the probe-time enable sequence; tested with a hibernate/restore cycle on this machine.
https://lore.kernel.org/all/20261112165433.64461-1-khanjamilahmed202@gmail.com/