joshuagrisham/samsung-galaxybook-extras #14: Driver feedback (Book3 Ultra + NixOS)

ItzDerock, 2024-06-01:
Hello, first of all, thanks so much for the work you've put into this. I'm writing to report that the driver is working great with my system:
- NixOS (unstable branch, 24.05.20240509.f1010e0)
- Linux 6.8.6-zen1
- Samsung Galaxy Book3 Ultra

Things ive noticed:
- Fan speed readings seem accurate (ie they reflect what I hear, can't attest to whether the values truly represent the fan rpm though since I've got no way to measure that)
- Keyboard backlight works great except there's sometimes a bit of a delay when turning the backlight off, not a big issue though and I'd imagine this is probably out of your control.
- Setting the platform profile works excellently. This was the primary reason I tried this out. Originally, the fans would not ramp up with CPU load, and so I would have fans running at 50% while the CPU was at 90C, at which point the CPU throttled down to 2.6ghz
  
  Setting the platform profile to `performance` allows the fans to ramp up a lot more than usual, maxing out at 5740rpm and helps to squeeze some more performance out of the machine
- Battery saver works as expected
overall, everything seems to be working smoothly!

For any other NixOS users I've written a derivation that you can copy/paste: https://github.com/ItzDerock/dotfiles/blob/main/pkgs/samsung-galaxybook-extras.nix
```nix
  boot.extraModulePackages = 
    let 
      sgbextras = config.boot.kernelPackages.callPackage ../../pkgs/samsung-galaxybook-extras.nix { };
      # change path to match where you save the package to
    in
    [ sgbextras ];
```

One last thing I wanted to mention was that, during my prior usage of this laptop, I've noticed the trackpad eats up a lot of battery in the form of interrupts. On powertop I've noticed idma64 was always at the top with thousands of events/sec. But now that I'm checking it again, I only see idma64 sending 200 events/sec, which is way less than before. I'm not sure if this is the result of this driver or a different change I've made with my system, and would like to hear if anyone else has had a similar experience. 

Thanks again to all contributors for their work on this, and especially to joshuagrisham for putting this together! Just wanted to leave some feedback and confirm that it's working on the book3 ultra, so feel free to close this whenever fit.

xhos, 2024-06-28:
oh my god I love you, thank you so much for doing this for nixos, it works perfectly on my book 2 pro 360

joshuagrisham, 2024-07-15:
Awesome feedback, thank you @ItzDerock !

Regarding the trackpad, there is currently nothing within this platform driver that touches that device so my guess would be that it should be 100% unrelated. The only wild idea I have is that maybe Samsung has done something "funny" in the BIOS code where if you change certain settings e.g. performance mode, that something also changes with the trackpad ?  Though that would seem quite strange to be honest... otherwise it should be a completely separate issue is my guess.

Regarding the fan speeds, it is a bit how I wrote in the README: basically the ACPI device has a "speed" value stored of 0 through 5, where 0 is totally off, and then there are hard-coded numbers that it reports what I think are RPM for levels 1 through 4; level 5 it reports the same number as level 4 but IMO this is not right, as the volume you can hear with your ear seems WAY louder at 5 than at 4... so I just threw a wild guess there and added 1000 to the number for 5 :laughing: But it would be interesting if someone who had one of these devices also had some kind of device to measure rotation of the fan and see how close the numbers actually seem to be! (because I just, as said, trusted the numbers from Samsung and then threw an extra 1000 on the "really fast-sounding" speed)