joshuagrisham/samsung-galaxybook-extras #15: Key was rejected by service (Zorin OS 17.1) - Trying to install driver

bryan977, 2024-06-23:
I am trying to figure out why I receive the error below when trying to load the module. I am running Zorin OS 17.1 (an Ubuntu derivative).

```
$ uname -r
6.5.0-41-generic
```

Error when running modprobe:
```
$ sudo modprobe samsung-galaxybook
modprobe: ERROR: could not insert 'samsung_galaxybook': Key was rejected by service
```

My build and install process:

```
$ ls -la
total 46
drwxrwxr-x 2 bryan bryan     6 Jun 23 10:25 .
drwxrwxr-x 6 bryan bryan     9 Jun 21 15:21 ..
-rw-rw-r-- 1 bryan bryan   388 Jun 21 15:21 Makefile
-rw-rw-r-- 1 bryan bryan     0 Jun 23 08:57 md
-rw-rw-r-- 1 bryan bryan  1255 Jun 21 15:21 README.md
-rw-rw-r-- 1 bryan bryan 40999 Jun 21 15:21 samsung-galaxybook.c

$ make -C /lib/modules/`uname -r`/build M=$PWD
make: Entering directory '/usr/src/linux-headers-6.5.0-41-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  You are using:           gcc-12 (Ubuntu 12.3.0-1ubuntu1~22.04) 12.3.0
  CC [M]  /home/bryan/Documents/git/samsung-galaxybook-extras/driver/samsung-galaxybook.o
  MODPOST /home/bryan/Documents/git/samsung-galaxybook-extras/driver/Module.symvers
  CC [M]  /home/bryan/Documents/git/samsung-galaxybook-extras/driver/samsung-galaxybook.mod.o
  LD [M]  /home/bryan/Documents/git/samsung-galaxybook-extras/driver/samsung-galaxybook.ko
  BTF [M] /home/bryan/Documents/git/samsung-galaxybook-extras/driver/samsung-galaxybook.ko
Skipping BTF generation for /home/bryan/Documents/git/samsung-galaxybook-extras/driver/samsung-galaxybook.ko due to unavailability of vmlinux
make: Leaving directory '/usr/src/linux-headers-6.5.0-41-generic'

$ sudo /usr/src/linux-source-6.5.0/debian/scripts/sign-module sha512 /usr/src/linux-source-6.5.0/debian/certs/signing_key.pem /usr/src/linux-source-6.5.0/debian/certs/signing_key.x509 /lib/modules/6.5.0-41-generic/updates/samsung-galaxybook.ko

$ sudo make -C /lib/modules/`uname -r`/build M=$PWD modules_install
make: Entering directory '/usr/src/linux-headers-6.5.0-41-generic'
  INSTALL /lib/modules/6.5.0-41-generic/updates/samsung-galaxybook.ko
  SIGN    /lib/modules/6.5.0-41-generic/updates/samsung-galaxybook.ko
  DEPMOD  /lib/modules/6.5.0-41-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make: Leaving directory '/usr/src/linux-headers-6.5.0-41-generic'

$ sudo cp samsung-galaxybook.ko /lib/modules/6.5.0-41-generic/updates

$ sudo depmod

$ sudo sudo modprobe samsung-galaxybook
modprobe: ERROR: could not insert 'samsung_galaxybook': Key was rejected by service
```

Any help is MUCH appreciated!

mhagnumdw, 2024-06-24:
I think you are signing the driver and placing it directly in the final location (`sign-module` line) and later you copy overwriting what I think is the unsigned driver.

I also think that using `uname -r` reduces the chance of errors.

I'll give you my exact sequence of commands. Here I use Fedora.

```shell
git clone git@github.com:joshuagrisham/samsung-galaxybook-extras.git

cd samsung-galaxybook-extras

cd driver

# it makes sense if you just did a git pull and want to start from scratch
git clean -fdx

make -C /lib/modules/`uname -r`/build M=$PWD

# If UEFI is enabled with Secure Boot, you need to sign the driver
# HERE YOU NEED TO ADJUST FOR YOUR DISTRO
sudo /usr/src/kernels/$(uname -r)/scripts/sign-file \
  sha256 \
  /etc/pki/akmods/private/private_key.priv \
  /etc/pki/akmods/certs/public_key.der \
  samsung-galaxybook.ko

ls -lrtha

sudo make -C /lib/modules/`uname -r`/build M=$PWD modules_install

sudo depmod

# enable the driver
sudo modprobe -v samsung-galaxybook

# check if the driver has been loaded
lsmod | grep samsung_galaxybook
```

bryan977, 2024-06-24:
@mhagnumdw Thank you. I did get the driver to load last night; however, just haven't had time to post the steps. That said, the keyboard backlight still isn't working so I still have work to do. This is a Galaxy Book Pro 360 (now v2 or v3) so maybe this driver doesn't support it.

```
$ lsmod | grep samsung_galaxybook
samsung_galaxybook     24576  0
platform_profile       12288  1 samsung_galaxybook
wmi                    40960  3 video,samsung_galaxybook,wmi_bmof
```

bryan977, 2024-06-24:
Problems I am still having:
1. Module does not stay loaded after reboot.
2. The keyboard backlight still does not light nor does Fn+F9 have an affect.

Any help or suggestions is appreciated. Thank you!

bryan977, 2024-06-25:
Any idea how to fix the missing System.map file error?

```
$ sudo make -C /lib/modules/`uname -r`/build M=$PWD modules_install
make: Entering directory '/usr/src/linux-headers-6.5.0-41-generic'
  INSTALL /lib/modules/6.5.0-41-generic/updates/samsung-galaxybook.ko
  SIGN    /lib/modules/6.5.0-41-generic/updates/samsung-galaxybook.ko
  DEPMOD  /lib/modules/6.5.0-41-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make: Leaving directory '/usr/src/linux-headers-6.5.0-41-generic'
```

bryan977, 2024-06-25:
It does appear the driver is loaded though, correct:
```
$ lsmod | grep samsung_galaxybook
samsung_galaxybook     24576  0
platform_profile       12288  1 samsung_galaxybook
wmi                    40960  3 video,samsung_galaxybook,wmi_bmof
```

I do not see the samsung-galaxybook class here. I assume I should?
```
$ ls /sys/class/leds
input2::capslock  input2::numlock  input2::scrolllock  mmc0::  phy0-led
```

The keyboard backlight still does not work, and, the driver does not reload upon reboot.

bryan977, 2024-06-26:
Also, I am running the Samsung Galaxy Book Pro 360. Here is the evemu-describe output. Can we add this to the 61-keyboard-samsung-galaxybook.hwdb file?

```
$ sudo evemu-describe
Available devices:
/dev/input/event0:	Lid Switch
/dev/input/event1:	Power Button
/dev/input/event2:	AT Translated Set 2 keyboard
/dev/input/event3:	ZNT0001:00 14E5:650E Mouse
/dev/input/event4:	ZNT0001:00 14E5:650E Touchpad
/dev/input/event5:	gpio-keys
/dev/input/event6:	ELAN902C:00 04F3:2DCE
/dev/input/event7:	ELAN902C:00 04F3:2DCE UNKNOWN
/dev/input/event8:	WCOM0134:00 2D1F:0135 Stylus
/dev/input/event9:	WCOM0134:00 2D1F:0135
/dev/input/event10:	Video Bus
/dev/input/event11:	Intel HID events
/dev/input/event12:	Intel HID 5 button array
/dev/input/event13:	ELAN902C:00 04F3:2DCE UNKNOWN
/dev/input/event14:	sof-hda-dsp Mic
/dev/input/event15:	sof-hda-dsp Headphone
/dev/input/event16:	sof-hda-dsp HDMI/DP,pcm=3
/dev/input/event17:	sof-hda-dsp HDMI/DP,pcm=4
/dev/input/event18:	sof-hda-dsp HDMI/DP,pcm=5
Select the device event number [0-18]: 2                  
# EVEMU 1.3
# Kernel: 6.5.0-41-generic
# DMI: dmi:bvnAmericanMegatrendsInternational,LLC.:bvrP14AKH.029.231116.KS:bd11/16/2023:br5.19:svnSAMSUNGELECTRONICSCO.,LTD.:pn930QDB:pvrP14AKH:rvnSAMSUNGELECTRONICSCO.,LTD.:rnNP930QDB-KD1US:rvrSGLA867A1X-C01-G001-S0002+10.0.19042:cvnSAMSUNGELECTRONICSCO.,LTD.:ct31:cvrN/A:skuSCAI-A5A5-A5A5-TGL3-PAKH:
# Input device name: "AT Translated Set 2 keyboard"
# Input device ID: bus 0x11 vendor 0x01 product 0x01 version 0xab83
# Supported events:
#   Event type 0 (EV_SYN)
#     Event code 0 (SYN_REPORT)
#     Event code 1 (SYN_CONFIG)
#     Event code 2 (SYN_MT_REPORT)
#     Event code 3 (SYN_DROPPED)
#     Event code 4 ((null))
#     Event code 5 ((null))
#     Event code 6 ((null))
#     Event code 7 ((null))
#     Event code 8 ((null))
#     Event code 9 ((null))
#     Event code 10 ((null))
#     Event code 11 ((null))
#     Event code 12 ((null))
#     Event code 13 ((null))
#     Event code 14 ((null))
#     Event code 15 (SYN_MAX)
#   Event type 1 (EV_KEY)
#     Event code 1 (KEY_ESC)
#     Event code 2 (KEY_1)
#     Event code 3 (KEY_2)
#     Event code 4 (KEY_3)
#     Event code 5 (KEY_4)
#     Event code 6 (KEY_5)
#     Event code 7 (KEY_6)
#     Event code 8 (KEY_7)
#     Event code 9 (KEY_8)
#     Event code 10 (KEY_9)
#     Event code 11 (KEY_0)
#     Event code 12 (KEY_MINUS)
#     Event code 13 (KEY_EQUAL)
#     Event code 14 (KEY_BACKSPACE)
#     Event code 15 (KEY_TAB)
#     Event code 16 (KEY_Q)
#     Event code 17 (KEY_W)
#     Event code 18 (KEY_E)
#     Event code 19 (KEY_R)
#     Event code 20 (KEY_T)
#     Event code 21 (KEY_Y)
[... 147 lines cut ...]
I: 0011 0001 0001 ab83
P: 00 00 00 00 00 00 00 00
B: 00 0b 00 00 00 00 00 00 00
B: 01 fe ff ff ff ff ff ff ff
B: 01 ff ff ef fb df ff ff fe
B: 01 01 d0 70 f8 78 30 80 03
B: 01 03 04 00 02 0f 50 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 02 00 00 00 00 00 00 00 00
B: 03 00 00 00 00 00 00 00 00
B: 04 10 00 00 00 00 00 00 00
B: 05 00 00 00 00 00 00 00 00
B: 11 07 00 00 00 00 00 00 00
B: 12 00 00 00 00 00 00 00 00
B: 14 03 00 00 00 00 00 00 00
B: 15 00 00 00 00 00 00 00 00
B: 15 00 00 00 00 00 00 00 00
```

bryan977, 2024-06-26:
I added the following to the  /etc/udev/hwdb.d/61-keyboard-samsung-galaxybook.hwdb file based on the above and still no luck:
```
# xxxQDB: Galaxy Book 360 Pro series
evdev:atkbd:dmi:bvn*:bvr*:bd*:svn[sS][aA][mM][sS][uU][nN][gG]*:pn[0-9][0-9][0-9]QDB:*
```

Gugu7264, 2024-07-12:
Same issue here on Samsung Galaxy Book 4. Module installed fine but keyboard isn't backlit. (By the way, for some reason on my laptop F9 turns the screen off?? (even before module install though)).

If needed, I can provide any logs that can be useful for fixing this ;)

bryan977, 2024-07-12:
I ended up giving up. Too many issues trying to get  Linux to be fully functional on Samsung hardware. I purchased a Dell XPS 13 9340 with Ubuntu pre-installed. Everything works ;-)

I still have the Samsung laptop, so if we ever get full drivers, maybe I can put it back into service.

Zippy-boy, 2024-07-14:
Ima alos trying to get my galaxy book pro 360 working with this driver, but whenever I try and make it gives me the error:
`make: *** /lib/modules/6.9.8-200.fc40.x86_64/build: No such file or directory.  Stop.`
Any idea why this is?

joshuagrisham, 2024-07-15:
Hi everyone here; regarding support for Galaxy Book(1) Pro 360 and Galaxy Book 4, I have no idea which devices are present on those models and/or if they would be compatible to how this platform driver is intended to work. It is the same story as I posted here:

> Hi @axcosta and @PremiumUsername , sorry for my delay in getting back to you!
> 
> I built this driver by reverse engineering + debugging how it works in Windows on my Galaxy Book2 Pro 15.6 so can never be 100% sure it is going to work exactly the same on other devices.
> 
> Having said that, the driver works by executing methods on a Samsung ACPI device (the one they called `SCAI`) which seems "relatively" standardized across several models.
> 
> On my particular device, the ACPI Device ID for the `SCAI` device is `SAM0429` but when I was looking into this in Windows I saw that the same driver on my device was matching both `SAM0427` and `SAM0429`. So far I have only put `SAM0429` in the code for this platform driver, as it is the only device I have access to test and would not want to make it even easier for anyone to accidentally harm their device :wink: 
> 
> But, as said, it probably works relatively out-of-the-box with `SAM0427` and possibly other device IDs as well, and because it is using ACPI then the most likely scenario is that either the device method names are different or require different payloads and will just reject/not respond to anything that is sent that would not work with other devices. However, if Samsung made other devices that happen to use exactly the same `SCAI` device, with exactly the same device methods (`CSFI` and `CSXI`), that can accept the exact same kind of input parameters (so that it is not rejected by the ACPI subsystem and actually executes the code in the BIOS), but behave completely differently and/or would "harm" the device somehow? Hmm I guess it is entirely possible but it seems pretty unlikely (I certainly would hope, anyway! :smile: )
> 
> If you run something like `ls /sys/devices/platform/ | grep -i SAM` do you see a match? If so, what is the device ID that shows up there? (for me it is `SAM0429:00`) Perhaps we can try adding it to the code and you can test to see if the driver will work on your device?
> 
> If there is no match, and/or you don't see anything under `/sys/devices/platform/` that look promising, how does it look in Windows on your device?  Specifically, is there a device driver installed called `Samsung System Event Controller`, and if you look at its properties / INI file can you tell what it has for device ID / vendor ID / product ID / etc ?
> 
_Originally posted by @joshuagrisham in https://github.com/joshuagrisham/samsung-galaxybook-extras/issues/9#issuecomment-2228058826_

Please feel free to take a look at the same on your devices and report back; it might be possible to just add your device's device IDs to the table and then it "just starts working" -- will just need to try it and see!

Zippy-boy, 2024-07-20:
Hi