joshuagrisham/samsung-galaxybook-extras #19: Kernel module fails to compile on Kernel 6.10.1 on Fedora 40

achapleau, 2024-08-04:
Running Fedora 40 with kernel Linux 6.10.1-200.fc40.x86_64, when I use the command make -C /lib/modules/`uname -r`/build M=$PWD the following output is presented and the kernel module fails to build:

andrew@galaxy:~/samsung-galaxybook-extras/driver$ make -C /lib/modules/`uname -r`/build M=$PWD
make: Entering directory '/usr/src/kernels/6.10.1-200.fc40.x86_64'
  CC [M]  /home/andrew/samsung-galaxybook-extras/driver/samsung-galaxybook.o
/home/andrew/samsung-galaxybook-extras/driver/samsung-galaxybook.c:805:6: warning: no previous prototype for ‘galaxybook_fan_speed_exit’ [-Wmissing-prototypes]
  805 | void galaxybook_fan_speed_exit(struct samsung_galaxybook *galaxybook)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/andrew/samsung-galaxybook-extras/driver/samsung-galaxybook.c:1649:10: error: ‘struct acpi_driver’ has no member named ‘owner’
 1649 |         .owner = THIS_MODULE,
      |          ^~~~~
In file included from ./include/linux/printk.h:6,
                 from ./include/linux/kernel.h:31,
                 from /home/andrew/samsung-galaxybook-extras/driver/samsung-galaxybook.c:13:
./include/linux/init.h:180:21: error: initialization of ‘const char *’ from incompatible pointer type ‘struct module *’ [-Wincompatible-pointer-types]
  180 | #define THIS_MODULE (&__this_module)
      |                     ^
/home/andrew/samsung-galaxybook-extras/driver/samsung-galaxybook.c:1649:18: note: in expansion of macro ‘THIS_MODULE’
 1649 |         .owner = THIS_MODULE,
      |                  ^~~~~~~~~~~
./include/linux/init.h:180:21: note: (near initialization for ‘galaxybook_acpi_driver.drv.name’)
  180 | #define THIS_MODULE (&__this_module)
      |                     ^
/home/andrew/samsung-galaxybook-extras/driver/samsung-galaxybook.c:1649:18: note: in expansion of macro ‘THIS_MODULE’
 1649 |         .owner = THIS_MODULE,
      |                  ^~~~~~~~~~~
/home/andrew/samsung-galaxybook-extras/driver/samsung-galaxybook.c:1639:52: warning: missing braces around initializer [-Wmissing-braces]
 1639 | static struct acpi_driver galaxybook_acpi_driver = {
      |                                                    ^
make[2]: *** [scripts/Makefile.build:244: /home/andrew/samsung-galaxybook-extras/driver/samsung-galaxybook.o] Error 1
make[1]: *** [/usr/src/kernels/6.10.1-200.fc40.x86_64/Makefile:1946: /home/andrew/samsung-galaxybook-extras/driver] Error 2
make: *** [Makefile:252: __sub-make] Error 2
make: Leaving directory '/usr/src/kernels/6.10.1-200.fc40.x86_64'

achapleau, 2024-08-04:
Downgraded to 6.9.12 and the module complies successfully.

achapleau, 2024-08-10:
Kernel 6.9 series is deprecated. Same issue with Kernel 6.10.3-200.fc40.x86_64:

andrew@galaxy:~/samsung-galaxybook-extras/driver$ make -C /lib/modules/`uname -r`/build M=$PWD
make: Entering directory '/usr/src/kernels/6.10.3-200.fc40.x86_64'
  CC [M]  /home/andrew/samsung-galaxybook-extras/driver/samsung-galaxybook.o
/home/andrew/samsung-galaxybook-extras/driver/samsung-galaxybook.c:805:6: warning: no previous prototype for ‘galaxybook_fan_speed_exit’ [-Wmissing-prototypes]
  805 | void galaxybook_fan_speed_exit(struct samsung_galaxybook *galaxybook)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/andrew/samsung-galaxybook-extras/driver/samsung-galaxybook.c:1649:10: error: ‘struct acpi_driver’ has no member named ‘owner’
 1649 |         .owner = THIS_MODULE,
      |          ^~~~~
In file included from ./include/linux/printk.h:6,
                 from ./include/linux/kernel.h:31,
                 from /home/andrew/samsung-galaxybook-extras/driver/samsung-galaxybook.c:13:
./include/linux/init.h:180:21: error: initialization of ‘const char *’ from incompatible pointer type ‘struct module *’ [-Wincompatible-pointer-types]
  180 | #define THIS_MODULE (&__this_module)
      |                     ^
/home/andrew/samsung-galaxybook-extras/driver/samsung-galaxybook.c:1649:18: note: in expansion of macro ‘THIS_MODULE’
 1649 |         .owner = THIS_MODULE,
      |                  ^~~~~~~~~~~
./include/linux/init.h:180:21: note: (near initialization for ‘galaxybook_acpi_driver.drv.name’)
  180 | #define THIS_MODULE (&__this_module)
      |                     ^
/home/andrew/samsung-galaxybook-extras/driver/samsung-galaxybook.c:1649:18: note: in expansion of macro ‘THIS_MODULE’
 1649 |         .owner = THIS_MODULE,
      |                  ^~~~~~~~~~~
/home/andrew/samsung-galaxybook-extras/driver/samsung-galaxybook.c:1639:52: warning: missing braces around initializer [-Wmissing-braces]
 1639 | static struct acpi_driver galaxybook_acpi_driver = {
      |                                                    ^
make[2]: *** [scripts/Makefile.build:244: /home/andrew/samsung-galaxybook-extras/driver/samsung-galaxybook.o] Error 1
make[1]: *** [/usr/src/kernels/6.10.3-200.fc40.x86_64/Makefile:1946: /home/andrew/samsung-galaxybook-extras/driver] Error 2
make: *** [Makefile:252: __sub-make] Error 2
make: Leaving directory '/usr/src/kernels/6.10.3-200.fc40.x86_64'

mhagnumdw, 2024-08-12:
Same issue here:

```
make -C /lib/modules/`uname -r`/build M=$PWD
make: Entering directory '/usr/src/kernels/6.10.3-200.fc40.x86_64'
  CC [M]  /home/mhagnumdw/projetos/samsung-galaxybook-extras/driver/samsung-galaxybook.o
/home/mhagnumdw/projetos/samsung-galaxybook-extras/driver/samsung-galaxybook.c:805:6: warning: no previous prototype for ‘galaxybook_fan_speed_exit’ [-Wmissing-prototypes]
  805 | void galaxybook_fan_speed_exit(struct samsung_galaxybook *galaxybook)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/mhagnumdw/projetos/samsung-galaxybook-extras/driver/samsung-galaxybook.c:1649:10: error: ‘struct acpi_driver’ has no member named ‘owner’
 1649 |         .owner = THIS_MODULE,
      |          ^~~~~
In file included from ./include/linux/printk.h:6,
                 from ./include/linux/kernel.h:31,
                 from /home/mhagnumdw/projetos/samsung-galaxybook-extras/driver/samsung-galaxybook.c:13:
./include/linux/init.h:180:21: error: initialization of ‘const char *’ from incompatible pointer type ‘struct module *’ [-Wincompatible-pointer-types]
  180 | #define THIS_MODULE (&__this_module)
      |                     ^
/home/mhagnumdw/projetos/samsung-galaxybook-extras/driver/samsung-galaxybook.c:1649:18: note: in expansion of macro ‘THIS_MODULE’
 1649 |         .owner = THIS_MODULE,
      |                  ^~~~~~~~~~~
./include/linux/init.h:180:21: note: (near initialization for ‘galaxybook_acpi_driver.drv.name’)
  180 | #define THIS_MODULE (&__this_module)
      |                     ^
/home/mhagnumdw/projetos/samsung-galaxybook-extras/driver/samsung-galaxybook.c:1649:18: note: in expansion of macro ‘THIS_MODULE’
 1649 |         .owner = THIS_MODULE,
      |                  ^~~~~~~~~~~
/home/mhagnumdw/projetos/samsung-galaxybook-extras/driver/samsung-galaxybook.c:1639:52: warning: missing braces around initializer [-Wmissing-braces]
 1639 | static struct acpi_driver galaxybook_acpi_driver = {
      |                                                    ^
make[2]: *** [scripts/Makefile.build:244: /home/mhagnumdw/projetos/samsung-galaxybook-extras/driver/samsung-galaxybook.o] Error 1
make[1]: *** [/usr/src/kernels/6.10.3-200.fc40.x86_64/Makefile:1946: /home/mhagnumdw/projetos/samsung-galaxybook-extras/driver] Error 2
make: *** [Makefile:252: __sub-make] Error 2
make: Leaving directory '/usr/src/kernels/6.10.3-200.fc40.x86_64'
```

PR #22 (2024-08-16): fix: remove acpi_driver owner field for kernel >= 6.10
Fixes #19

The ACPI `.owner` field was removed with 6.10 in https://github.com/torvalds/linux/commit/cc85f9c05bba23eb525497b42ac5b74801ccbd87. This PR adds in a guard to prevent setting the owner field on kernel versions after 6.10. 

I have tested building this with `6.6.44`, `6.9.12`, `6.10.1`, and `6.10.3`. Functionality tested with only `6.10.1`.
closed 2024-08-16 by joshuagrisham