joshuagrisham/samsung-galaxybook-extras #24: Problem installing

Kokkey, 2024-08-23:
I'm trying to install but facing problems:

sudo make -C /lib/modules/`uname -r`/build M=$PWD 
make: Map '/usr/src/linux-headers-6.8.0-40-generic' wordt binnengegaan
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  CC [M]  /home/niels/Downloads/samsung-galaxybook-extras-main/driver/samsung-galaxybook.o
/home/niels/Downloads/samsung-galaxybook-extras-main/driver/samsung-galaxybook.c:806:6: warning: no previous prototype for ‘galaxybook_fan_speed_exit’ [-Wmissing-prototypes]
  806 | void galaxybook_fan_speed_exit(struct samsung_galaxybook *galaxybook)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~
  MODPOST /home/niels/Downloads/samsung-galaxybook-extras-main/driver/Module.symvers
  CC [M]  /home/niels/Downloads/samsung-galaxybook-extras-main/driver/samsung-galaxybook.mod.o
  LD [M]  /home/niels/Downloads/samsung-galaxybook-extras-main/driver/samsung-galaxybook.ko
make: Map '/usr/src/linux-headers-6.8.0-40-generic' wordt verlaten

joshuagrisham, 2024-09-28:
Hi @Kokkey  do  you have any update on this? Based on your message it looks like there was a warning but the module was built anyway.. did you try to `insmod` or install it somehow and see if it worked?

Kokkey, 2024-09-30:
Hi @joshuagrisham 

 When I try "make -C /lib/modules/`uname -r`/build M=$PWD modules_install" I get the following error:

make: Map '/usr/src/linux-headers-6.8.0-41-generic' wordt binnengegaan
  INSTALL /lib/modules/6.8.0-41-generic/updates/samsung-galaxybook.ko
  SIGN    /lib/modules/6.8.0-41-generic/updates/samsung-galaxybook.ko
  DEPMOD  /lib/modules/6.8.0-41-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make: Map '/usr/src/linux-headers-6.8.0-41-generic' wordt verlaten

joshuagrisham, 2024-10-16:
Hi @Kokkey 
My assumption is the error is due to Secure Boot, so you will need to sign the module before trying to load it.
You can get a copy of the certs+keys used in Debian/Ubuntu if you install the `linux-source` package using `apt` (default or to match whatever linux-image version you are running..)
From there I think you can follow my instructions here: https://github.com/joshuagrisham/samsung-galaxybook-extras?tab=readme-ov-file#how-to-avoid-signature-andor-required-key-missing

Kokkey, 2024-11-12:
@joshuagrisham I think it worked! thanks! 
I used the mokutil tools and a lot of google.

Thank you for your help!

Kokkey, 2024-11-12:
this issue can be closed now, it was my mistake

closed 2024-11-12 by Kokkey