joshuagrisham/samsung-galaxybook-extras #27: Key mapping for disabling touchpad (Fn+F5)

bortan1, 2024-09-24:
Hey, I will start with cheering you for this incredible work you've done, I really appreciate it.
First thing first, I am running Fedora 40, KDE spin and after a fresh install the Fn+F5 key was already doing what it's meant to be doing (like on windows side) - enabling and disabling the touchpad. After installing your driver, this key apparently does nothing anymore and I don't know how to revert back this specific key, while keeping the rest of the modifications you've made.
My device is Samsung Galaxy Book2 Pro 360
This is my evemu-describe for the AT Transated Set 2 keyvoard:
`2
# EVEMU 1.3
# Kernel: 6.10.10-200.fc40.x86_64
# DMI: dmi:bvnAmericanMegatrendsInternational,LLC.:bvrP11AKK.750.240404.SH:bd04/04/2024:br5.25:svnSAMSUNGELECTRONICSCO.,LTD.:pn950QED:pvrP11AKK:rvnSAMSUNGELECTRONICSCO.,LTD.:rnNP950QED-KA2DE:rvrSGLB182A1H-C01-G002-S0001+10.0.22000:cvnSAMSUNGELECTRONICSCO.,LTD.:ct31:cvrN/A:skuSCAI-ICPS-A5A5-ADLP-PAKK:
# Input device name: "AT Translated Set 2 keyboard"
# Input device ID: bus 0x11 vendor 0x01 product 0x01 version 0xab83
# Supported events:
#   Event type 0 (EV_SYN)
#     Event code 0 (SYN_REPORT)
#     Event code 1 (SYN_CONFIG)
#     Event code 2 (SYN_MT_REPORT)
#     Event code 3 (SYN_DROPPED)
#     Event code 4 ((null))
#     Event code 5 ((null))
#     Event code 6 ((null))
#     Event code 7 ((null))
#     Event code 8 ((null))
#     Event code 9 ((null))
#     Event code 10 ((null))
#     Event code 11 ((null))
#     Event code 12 ((null))
#     Event code 13 ((null))
#     Event code 14 ((null))
#     Event code 15 (SYN_MAX)
#   Event type 1 (EV_KEY)
#     Event code 1 (KEY_ESC)
#     Event code 2 (KEY_1)
#     Event code 3 (KEY_2)
#     Event code 4 (KEY_3)
#     Event code 5 (KEY_4)
#     Event code 6 (KEY_5)
#     Event code 7 (KEY_6)
#     Event code 8 (KEY_7)
#     Event code 9 (KEY_8)
#     Event code 10 (KEY_9)
#     Event code 11 (KEY_0)
#     Event code 12 (KEY_MINUS)
#     Event code 13 (KEY_EQUAL)
#     Event code 14 (KEY_BACKSPACE)
#     Event code 15 (KEY_TAB)
#     Event code 16 (KEY_Q)
#     Event code 17 (KEY_W)
#     Event code 18 (KEY_E)
#     Event code 19 (KEY_R)
#     Event code 20 (KEY_T)
#     Event code 21 (KEY_Y)
#     Event code 22 (KEY_U)
#     Event code 23 (KEY_I)
#     Event code 24 (KEY_O)
#     Event code 25 (KEY_P)
#     Event code 26 (KEY_LEFTBRACE)
#     Event code 27 (KEY_RIGHTBRACE)
#     Event code 28 (KEY_ENTER)
#     Event code 29 (KEY_LEFTCTRL)
#     Event code 30 (KEY_A)
#     Event code 31 (KEY_S)
#     Event code 32 (KEY_D)
#     Event code 33 (KEY_F)
#     Event code 34 (KEY_G)
#     Event code 35 (KEY_H)
#     Event code 36 (KEY_J)
#     Event code 37 (KEY_K)
#     Event code 38 (KEY_L)
#     Event code 39 (KEY_SEMICOLON)
#     Event code 40 (KEY_APOSTROPHE)
#     Event code 41 (KEY_GRAVE)
#     Event code 42 (KEY_LEFTSHIFT)
#     Event code 43 (KEY_BACKSLASH)
#     Event code 44 (KEY_Z)
#     Event code 45 (KEY_X)
#     Event code 46 (KEY_C)
#     Event code 47 (KEY_V)
#     Event code 48 (KEY_B)
#     Event code 49 (KEY_N)
#     Event code 50 (KEY_M)
#     Event code 51 (KEY_COMMA)
#     Event code 52 (KEY_DOT)
#     Event code 53 (KEY_SLASH)
#     Event code 54 (KEY_RIGHTSHIFT)
#     Event code 55 (KEY_KPASTERISK)
#     Event code 56 (KEY_LEFTALT)
#     Event code 57 (KEY_SPACE)
#     Event code 58 (KEY_CAPSLOCK)
#     Event code 59 (KEY_F1)
#     Event code 60 (KEY_F2)
#     Event code 61 (KEY_F3)
#     Event code 62 (KEY_F4)
#     Event code 63 (KEY_F5)
#     Event code 64 (KEY_F6)
#     Event code 65 (KEY_F7)
#     Event code 66 (KEY_F8)
#     Event code 67 (KEY_F9)
#     Event code 68 (KEY_F10)
#     Event code 69 (KEY_NUMLOCK)
#     Event code 70 (KEY_SCROLLLOCK)
#     Event code 71 (KEY_KP7)
#     Event code 72 (KEY_KP8)
#     Event code 73 (KEY_KP9)
#     Event code 74 (KEY_KPMINUS)
#     Event code 75 (KEY_KP4)
#     Event code 76 (KEY_KP5)
#     Event code 77 (KEY_KP6)
#     Event code 78 (KEY_KPPLUS)
#     Event code 79 (KEY_KP1)
#     Event code 80 (KEY_KP2)
#     Event code 81 (KEY_KP3)
#     Event code 82 (KEY_KP0)
#     Event code 83 (KEY_KPDOT)
#     Event code 86 (KEY_102ND)
#     Event code 87 (KEY_F11)
#     Event code 88 (KEY_F12)
#     Event code 89 (KEY_RO)
#     Event code 91 (KEY_HIRAGANA)
#     Event code 92 (KEY_HENKAN)
#     Event code 93 (KEY_KATAKANAHIRAGANA)
#     Event code 94 (KEY_MUHENKAN)
#     Event code 95 (KEY_KPJPCOMMA)
#     Event code 96 (KEY_KPENTER)
#     Event code 97 (KEY_RIGHTCTRL)
#     Event code 98 (KEY_KPSLASH)
#     Event code 99 (KEY_SYSRQ)
#     Event code 100 (KEY_RIGHTALT)
#     Event code 102 (KEY_HOME)
#     Event code 103 (KEY_UP)
#     Event code 104 (KEY_PAGEUP)
#     Event code 105 (KEY_LEFT)
#     Event code 106 (KEY_RIGHT)
#     Event code 107 (KEY_END)
#     Event code 108 (KEY_DOWN)
#     Event code 109 (KEY_PAGEDOWN)
#     Event code 110 (KEY_INSERT)
#     Event code 111 (KEY_DELETE)
#     Event code 112 (KEY_MACRO)
#     Event code 113 (KEY_MUTE)
#     Event code 114 (KEY_VOLUMEDOWN)
#     Event code 115 (KEY_VOLUMEUP)
#     Event code 116 (KEY_POWER)
#     Event code 117 (KEY_KPEQUAL)
#     Event code 119 (KEY_PAUSE)
#     Event code 121 (KEY_KPCOMMA)
#     Event code 122 (KEY_HANGEUL)
#     Event code 123 (KEY_HANJA)
#     Event code 124 (KEY_YEN)
#     Event code 125 (KEY_LEFTMETA)
#     Event code 126 (KEY_RIGHTMETA)
#     Event code 127 (KEY_COMPOSE)
#     Event code 128 (KEY_STOP)
#     Event code 140 (KEY_CALC)
#     Event code 142 (KEY_SLEEP)
#     Event code 143 (KEY_WAKEUP)
#     Event code 148 (KEY_PROG1)
#     Event code 149 (KEY_PROG2)
#     Event code 150 (KEY_WWW)
#     Event code 155 (KEY_MAIL)
#     Event code 156 (KEY_BOOKMARKS)
#     Event code 157 (KEY_COMPUTER)
#     Event code 158 (KEY_BACK)
#     Event code 159 (KEY_FORWARD)
#     Event code 163 (KEY_NEXTSONG)
#     Event code 164 (KEY_PLAYPAUSE)
#     Event code 165 (KEY_PREVIOUSSONG)
#     Event code 166 (KEY_STOPCD)
#     Event code 171 (KEY_CONFIG)
#     Event code 172 (KEY_HOMEPAGE)
#     Event code 173 (KEY_REFRESH)
#     Event code 183 (KEY_F13)
#     Event code 184 (KEY_F14)
#     Event code 185 (KEY_F15)
#     Event code 191 (KEY_F21)
#     Event code 192 (KEY_F22)
#     Event code 193 (KEY_F23)
#     Event code 202 (KEY_PROG3)
#     Event code 212 (KEY_CAMERA)
#     Event code 217 (KEY_SEARCH)
#     Event code 224 (KEY_BRIGHTNESSDOWN)
#     Event code 225 (KEY_BRIGHTNESSUP)
#     Event code 226 (KEY_MEDIA)
#     Event code 227 (KEY_SWITCHVIDEOMODE)
#     Event code 236 (KEY_BATTERY)
#     Event code 238 (KEY_WLAN)
#     Event code 240 (KEY_UNKNOWN)
#   Event type 4 (EV_MSC)
#     Event code 4 (MSC_SCAN)
#   Event type 17 (EV_LED)
#     Event code 0 (LED_NUML)
#        State 1
#     Event code 1 (LED_CAPSL)
#        State 0
#     Event code 2 (LED_SCROLLL)
#        State 0
#   Event type 20 (EV_REP)
#     Event code 0 (REP_DELAY)
#     Event code 1 (REP_PERIOD)
# Properties:
N: AT Translated Set 2 keyboard
I: 0011 0001 0001 ab83
P: 00 00 00 00 00 00 00 00
B: 00 0b 00 00 00 00 00 00 00
B: 01 fe ff ff ff ff ff ff ff
B: 01 ff ff cf fb df ff bf fe
B: 01 01 d0 70 f8 78 38 80 83
B: 01 03 04 10 02 0f 50 01 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 01 00 00 00 00 00 00 00 00
B: 02 00 00 00 00 00 00 00 00
B: 03 00 00 00 00 00 00 00 00
B: 04 10 00 00 00 00 00 00 00
B: 05 00 00 00 00 00 00 00 00
B: 11 07 00 00 00 00 00 00 00
B: 12 00 00 00 00 00 00 00 00
B: 14 03 00 00 00 00 00 00 00
B: 15 00 00 00 00 00 00 00 00
B: 15 00 00 00 00 00 00 00 00
L: 00 1
`

joshuagrisham, 2024-09-28:
Hi @bortan1 can you try removing the hwdb script from this driver, rebooting, and seeing what these keys report back when using `evtest` ?

If you have installed the hwdb from my repo then you can remove them like this:

```
sudo rm /etc/udev/hwdb.d/61-keyboard-samsung-galaxybook.hwdb
sudo systemd-hwdb update
sudo udevadm trigger
```

Then reboot just to be sure. After rebooting you can run `sudo evtest`, select the keyboard, and then press and release the key you are having an issue with a few times and report back with the output?

Also after you have removed and rebooted, does the key work again to toggle the touchpad?

bortan1, 2024-10-15:
Sorry for a late reply. I have reinstalled fedora on my laptop in the meantime and haven't reinstalled your driver yet. Glad to say that fingerprint seems to work out of the box on fedora 40 kde. Idk if this has to do anything with me keeping the old /home partition, but it works. As for the key issue I was telling you before, here is the command output, with the first keypress disabling the touchpad and the second one re-enabling it.

`$ sudo evtest /dev/input/event2
Input driver version is 1.0.1
Input device ID: bus 0x11 vendor 0x1 product 0x1 version 0xab83
Input device name: "AT Translated Set 2 keyboard"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 2 (KEY_1)
    Event code 3 (KEY_2)
    Event code 4 (KEY_3)
    Event code 5 (KEY_4)
    Event code 6 (KEY_5)
    Event code 7 (KEY_6)
    Event code 8 (KEY_7)
    Event code 9 (KEY_8)
    Event code 10 (KEY_9)
    Event code 11 (KEY_0)
    Event code 12 (KEY_MINUS)
    Event code 13 (KEY_EQUAL)
    Event code 14 (KEY_BACKSPACE)
    Event code 15 (KEY_TAB)
    Event code 16 (KEY_Q)
    Event code 17 (KEY_W)
    Event code 18 (KEY_E)
    Event code 19 (KEY_R)
    Event code 20 (KEY_T)
    Event code 21 (KEY_Y)
    Event code 22 (KEY_U)
    Event code 23 (KEY_I)
    Event code 24 (KEY_O)
    Event code 25 (KEY_P)
    Event code 26 (KEY_LEFTBRACE)
    Event code 27 (KEY_RIGHTBRACE)
    Event code 28 (KEY_ENTER)
    Event code 29 (KEY_LEFTCTRL)
    Event code 30 (KEY_A)
    Event code 31 (KEY_S)
    Event code 32 (KEY_D)
    Event code 33 (KEY_F)
    Event code 34 (KEY_G)
    Event code 35 (KEY_H)
    Event code 36 (KEY_J)
    Event code 37 (KEY_K)
    Event code 38 (KEY_L)
    Event code 39 (KEY_SEMICOLON)
    Event code 40 (KEY_APOSTROPHE)
    Event code 41 (KEY_GRAVE)
    Event code 42 (KEY_LEFTSHIFT)
    Event code 43 (KEY_BACKSLASH)
    Event code 44 (KEY_Z)
    Event code 45 (KEY_X)
    Event code 46 (KEY_C)
    Event code 47 (KEY_V)
    Event code 48 (KEY_B)
    Event code 49 (KEY_N)
    Event code 50 (KEY_M)
    Event code 51 (KEY_COMMA)
    Event code 52 (KEY_DOT)
    Event code 53 (KEY_SLASH)
    Event code 54 (KEY_RIGHTSHIFT)
    Event code 55 (KEY_KPASTERISK)
    Event code 56 (KEY_LEFTALT)
    Event code 57 (KEY_SPACE)
    Event code 58 (KEY_CAPSLOCK)
    Event code 59 (KEY_F1)
    Event code 60 (KEY_F2)
    Event code 61 (KEY_F3)
[... 121 lines cut ...]
Event: time 1729008717.321300, -------------- SYN_REPORT ------------
Event: time 1729008717.324568, type 4 (EV_MSC), code 4 (MSC_SCAN), value db
Event: time 1729008717.324568, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 0
Event: time 1729008717.324568, -------------- SYN_REPORT ------------
Event: time 1729008717.330074, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1d
Event: time 1729008717.330074, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 0
Event: time 1729008717.330074, -------------- SYN_REPORT ------------
Event: time 1729008718.444731, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1d
Event: time 1729008718.444731, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
Event: time 1729008718.444731, -------------- SYN_REPORT ------------
Event: time 1729008718.447336, type 4 (EV_MSC), code 4 (MSC_SCAN), value db
Event: time 1729008718.447336, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 1
Event: time 1729008718.447336, -------------- SYN_REPORT ------------
Event: time 1729008718.448636, type 4 (EV_MSC), code 4 (MSC_SCAN), value 76
Event: time 1729008718.448636, type 1 (EV_KEY), code 85 (KEY_ZENKAKUHANKAKU), value 1
Event: time 1729008718.448636, -------------- SYN_REPORT ------------
Event: time 1729008718.579361, type 4 (EV_MSC), code 4 (MSC_SCAN), value 76
Event: time 1729008718.579361, type 1 (EV_KEY), code 85 (KEY_ZENKAKUHANKAKU), value 0
Event: time 1729008718.579361, -------------- SYN_REPORT ------------
Event: time 1729008718.582495, type 4 (EV_MSC), code 4 (MSC_SCAN), value db
Event: time 1729008718.582495, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 0
Event: time 1729008718.582495, -------------- SYN_REPORT ------------
Event: time 1729008718.584546, type 4 (EV_MSC), code 4 (MSC_SCAN), value 1d
Event: time 1729008718.584546, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 0
`

joshuagrisham, 2024-10-15:
Hi @bortan1 I will try to take a further look at this.

I think it would be good to try my latest file I have put here now: https://github.com/joshuagrisham/samsung-galaxybook-extras/blob/main/61-keyboard-samsung-galaxybook.hwdb

And make sure it is working for you.. otherwise I need to take a much more serious look at this!
Because once systemd releases version 256.8, then the file I linked to there will basically be what is included out-of-the-box with systemd itself :wink:

joshuagrisham, 2024-10-16:
Now I looked a bit.. it is very interesting that KEY_ZENKAKUHANKAKU is actually disabling your touchpad in Fedora 40 + KDE ?? That seems very strange to me... if you try to live boot to another distro e.g. Ubuntu does it still work there or behave differently?

BTW I recognize the key mapping and believe my device had the same before I remapped it using the hwdb file.. now it works fine for me in Ubuntu+GNOME and it also matches pretty much the same mapping across many different brands and even multiple Samsung models, so I am a bit confused that it does not work for you?

Curious to know if you try again with the link above, how it goes. :smiley_cat: Maybe try to reboot after applying the hwdb file and see how it works after that? And if it does not work, maybe post another evtest log ?