joshuagrisham/samsung-galaxybook-extras #31: Failed to set performance to low-power (Silent)

mhagnumdw, 2024-10-10:
Galaxy Book3 Ultra
- Product Name: NP960XFH-XA1BR
- Serial Number: (preciso informar?)
- Kernel: 6.10.12-200.fc40.x86_64
- OS: Fedora Linux 40 (Workstation Edition) x86_64
- DE: GNOME 46.5
- Driver version commit hash: f1aba1a7a48ec32b0a32d9eba150be998cae1b07

On the command line, error:
```bash
echo low-power | sudo tee /sys/firmware/acpi/platform_profile
low-power
tee: /sys/firmware/acpi/platform_profile: Input/output error
```

In Gnome, `Power Mode > Power Saver` does not work, error:
```
out 10 08:20:43 dwnote2 kernel: samsung_galaxybook: failed setting performance_mode with ACPI method CSXI; device did not respond with success code 0xaa
out 10 08:20:43 dwnote2 gnome-shell[2949]: Could not set property ActiveProfile on remote object /net/hadess/PowerProfiles: GDBus.Error:org.gtk.GDBus.UnmappedGError.Quark._g_2dio_2derror_2dquark.Code0: Failed to activate platform driver 'platform_profile': Failed to write to acpi_platform_profile: Error writing '/sys/firmware/acpi/platform_profile': Input/output error
```

More information:

```bash
cat /sys/firmware/acpi/platform_profile_choices
low-power quiet balanced performance
```

The only mode I can't set on the command line is `low-power`.

On the command line, when I set `quiet`, Gnome says it's in `Power Saver` mode.

```bash
sudo dmidecode --type 3
# dmidecode 3.6
Getting SMBIOS data from sysfs.
SMBIOS 3.5.0 present.

Handle 0x0003, DMI type 3, 22 bytes
Chassis Information
	Manufacturer: SAMSUNG ELECTRONICS CO., LTD.
	Type: Notebook
	Lock: Not Present
	Version: N/A
	Serial Number: omitted - do I need to inform?
	Asset Tag: No Asset Tag
	Boot-up State: Safe
	Power Supply State: Safe
	Thermal State: Safe
	Security Status: None
	OEM Information: 0x00000000
	Height: Unspecified
	Number Of Power Cords: 1
	Contained Elements: 0
	SKU Number: Samsung Chassis
```

From docs:
> PLUS that the ACPI device SCAI is present with ACPI Device IDs

How do I find out the ID?

Tks!

joshuagrisham, 2024-10-10:
Thanks @seaasses 
The part I am also interested to see is where it should say `[DEBUG] get supported performance modes` and `[DEBUG] get performance mode values` (this will happen only one time after the module is loaded)
Also if you set to other profile names , does it work ? (e.g. it is only the `low-power` profile that complains?)

mhagnumdw, 2024-10-10:
Hi @joshuagrisham 

> Hi @mhagnumdw can you please enable the module parameter `debug` and then perform this again and post the kernel log (e.g. from `dmesg`) ?

### `sudo rmmod samsung-galaxybook`
dmesg:
```
[10947.872837] samsung_galaxybook: removing driver
[10947.901211] ACPI: battery: extension unregistered: Samsung Galaxy Book Battery Extension
[10947.902815] samsung_galaxybook: set kbd_backlight brightness to 0
[10947.935243] samsung_galaxybook: driver successfully removed
```

### `sudo modprobe samsung-galaxybook debug=true`
dmesg:
```
[10963.724159] samsung_galaxybook: loading driver
[10963.724392] samsung_galaxybook: [DEBUG] received following device quirks:
[10963.724397] samsung_galaxybook: [DEBUG]   disable_kbd_backlight       = false
[10963.724400] samsung_galaxybook: [DEBUG]   disable_battery_threshold   = false
[10963.724402] samsung_galaxybook: [DEBUG]   disable_performance_mode    = false
[10963.724403] samsung_galaxybook: [DEBUG]   disable_fan_speed           = false
[10963.724405] samsung_galaxybook: [DEBUG]   disable_i8042_filter        = false
[10963.724406] samsung_galaxybook: [DEBUG]   disable_acpi_hotkeys        = false
[10963.724408] samsung_galaxybook: [DEBUG]   disable_wmi_hotkeys         = false
[10963.724409] samsung_galaxybook: [DEBUG] performance_modes override exist in quirks
[10963.724606] samsung_galaxybook: initializing ACPI device
[10963.724664] samsung_galaxybook: [DEBUG] enabling ACPI feature
[10963.724668] samsung_galaxybook: [DEBUG]     43 58 7a 00 00 bb aa 00 00 00 00 00 00 00 00 00
[10963.724671] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[10963.726750] samsung_galaxybook: [DEBUG] response was:
[10963.726771] samsung_galaxybook: [DEBUG]     43 58 7a 00 aa dd cc 00 00 00 00 00 00 00 00 00
[10963.726776] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[10963.726781] samsung_galaxybook: [DEBUG] enabling ACPI feature
[10963.726783] samsung_galaxybook: [DEBUG]     43 58 8a 00 00 bb aa 00 00 00 00 00 00 00 00 00
[10963.726784] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[10963.727812] samsung_galaxybook: [DEBUG] response was:
[10963.727816] samsung_galaxybook: [DEBUG]     43 58 8a 00 aa dd cc 00 00 00 00 00 00 00 00 00
[10963.727818] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[10963.727819] samsung_galaxybook: initializing platform device
[10963.728045] samsung_galaxybook: initializing performance_mode
[10963.728062] samsung_galaxybook: [DEBUG] get supported performance modes
[10963.728067] samsung_galaxybook: [DEBUG]     43 58 91 00 00 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[10963.728070] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 00 00 00 00 00 00 00 00 00 00
[10963.728072] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.728074] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.728075] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.728077] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.728078] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.728079] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.728081] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.728082] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.728084] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.728085] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.728087] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.728088] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.728090] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.728091] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.730220] samsung_galaxybook: [DEBUG] response was:
[10963.730223] samsung_galaxybook: [DEBUG]     43 58 91 00 aa 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[10963.730225] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 00 01 01 01 01 00 00 00 00 00
[10963.730227] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.730228] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.730230] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.730232] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.730233] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.730235] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.730236] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.730238] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10963.730239] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... 189 lines cut ...]
[10989.446484] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10989.448969] samsung_galaxybook: [DEBUG] response was:
[10989.448973] samsung_galaxybook: [DEBUG]     43 58 91 00 aa 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[10989.448975] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 02 02 00 00 00 00 00 00 00 00
[... cut ...]
[10989.448999] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[10989.449001] samsung_galaxybook: [DEBUG] performance_mode is currently 'optimized' (0x02)
[10992.398448] samsung_galaxybook: [DEBUG] reporting fan_speed of 0 (level 0)
[10995.394698] samsung_galaxybook: [DEBUG] reporting fan_speed of 0 (level 0)
[10998.402326] samsung_galaxybook: [DEBUG] reporting fan_speed of 0 (level 0)
[11001.403772] samsung_galaxybook: [DEBUG] reporting fan_speed of 0 (level 0)
[11004.400565] samsung_galaxybook: [DEBUG] reporting fan_speed of 0 (level 0)
```

joshuagrisham, 2024-10-10:
Thank you both!

Interesting... so your device (which looks like also has the `SAM0429` device based on the quirk that it got) seems to support these modes:

- 00
- 01
- 02
- 0a
- 14
- 15
- 16

The "fun" part is that what we have seen is that the Windows driver/service is picking only 4 of these (even though your device supports all 7) when you press the hotkey or change the settings in Windows (assuming you both also have the same 4 modes in Windows? Silent, Quiet, Optmized, and Performance ?) and we have not yet figured out exactly why it picks what it picks for each level...

Maybe it would be interesting to see a copy of your DSDT if you can upload one? It can hopefully provide a clue...

joshuagrisham, 2024-10-10:
On the plus side: this new logic "works" in that this is actually a failure .. your device's ACPI method is basically rejecting the payload and the setting is not being applied, so the fact that it is reporting as a failure is exactly what we want :) If you had used the driver prior to the version merged in today, it would have been still behaving the same way but the failure was not recognized properly (so you think you are setting "low-power" but nothing was actually happening...)

seaasses, 2024-10-10:
That's interesting! In the previous version, when I set it to low-power mode, the fans would slow down compared to the performance mode, so it actually seemed to be working in that regard

seaasses, 2024-10-10:
> On the plus side: this new logic "works" in that this is actually a failure .. your device's ACPI method is basically rejecting the payload and the setting is not being applied, so the fact that it [...]

That's interesting! In the previous version, when I set it to low-power mode, the fans would slow down compared to the performance mode, so it actually seemed to be working in that regard

seaasses, 2024-10-10:
And here is my driver loading logs as my model is different than @mhagnumdw's model, maybe we have different things
```
[ 3866.477959] samsung_galaxybook: loading driver
[ 3866.478174] samsung_galaxybook: [DEBUG] received following device quirks:
[ 3866.478177] samsung_galaxybook: [DEBUG]   disable_kbd_backlight       = false
[ 3866.478179] samsung_galaxybook: [DEBUG]   disable_battery_threshold   = false
[ 3866.478181] samsung_galaxybook: [DEBUG]   disable_performance_mode    = false
[ 3866.478182] samsung_galaxybook: [DEBUG]   disable_fan_speed           = false
[ 3866.478184] samsung_galaxybook: [DEBUG]   disable_i8042_filter        = false
[ 3866.478185] samsung_galaxybook: [DEBUG]   disable_acpi_hotkeys        = false
[ 3866.478187] samsung_galaxybook: [DEBUG]   disable_wmi_hotkeys         = false
[ 3866.478189] samsung_galaxybook: [DEBUG] performance_modes override exist in quirks
[ 3866.478362] samsung_galaxybook: initializing ACPI device
[ 3866.478427] samsung_galaxybook: [DEBUG] enabling ACPI feature
[ 3866.478430] samsung_galaxybook: [DEBUG]     43 58 7a 00 00 bb aa 00 00 00 00 00 00 00 00 00
[ 3866.478433] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[ 3866.479540] samsung_galaxybook: [DEBUG] response was:
[ 3866.479543] samsung_galaxybook: [DEBUG]     43 58 7a 00 aa dd cc 00 00 00 00 00 00 00 00 00
[ 3866.479546] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[ 3866.479548] samsung_galaxybook: [DEBUG] enabling ACPI feature
[ 3866.479550] samsung_galaxybook: [DEBUG]     43 58 8a 00 00 bb aa 00 00 00 00 00 00 00 00 00
[ 3866.479552] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[ 3866.480479] samsung_galaxybook: [DEBUG] response was:
[ 3866.480482] samsung_galaxybook: [DEBUG]     43 58 8a 00 aa dd cc 00 00 00 00 00 00 00 00 00
[ 3866.480484] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[ 3866.480486] samsung_galaxybook: initializing platform device
[ 3866.480574] samsung_galaxybook: initializing performance_mode
[ 3866.480576] samsung_galaxybook: [DEBUG] get supported performance modes
[ 3866.480578] samsung_galaxybook: [DEBUG]     43 58 91 00 00 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[ 3866.480580] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 00 00 00 00 00 00 00 00 00 00
[ 3866.480582] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480583] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480585] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480587] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480588] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480590] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480591] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480593] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480595] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480596] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480598] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480599] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480601] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480603] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482816] samsung_galaxybook: [DEBUG] response was:
[ 3866.482827] samsung_galaxybook: [DEBUG]     43 58 91 00 aa 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[ 3866.482832] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 00 01 01 01 01 00 00 00 00 00
[ 3866.482834] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482836] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482838] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482840] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482842] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482844] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482846] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482847] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482849] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482851] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482853] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482854] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482856] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482858] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482861] samsung_galaxybook: [DEBUG] get performance mode values
[ 3866.482863] samsung_galaxybook: [DEBUG]     43 58 91 00 00 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[ 3866.482865] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 01 00 00 00 00 00 00 00 00 00
[ 3866.482866] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
[ 3866.493219] samsung_galaxybook: [DEBUG]     43 58 86 00 aa 02 00 00 00 00 00 00 00 00 00 00
[ 3866.493221] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[ 3866.493222] samsung_galaxybook: initializing hotkey input device
[ 3866.493280] input: Samsung Galaxy Book extra buttons as /devices/platform/samsung-galaxybook/input/input40
[ 3866.493348] samsung_galaxybook: enabling WMI notifications
[ 3866.493354] samsung_galaxybook: installing WMI notify handler
[ 3866.493386] samsung_galaxybook: driver successfully loaded
```

joshuagrisham, 2024-10-10:
> That's interesting! In the previous version, when I set it to low-power mode, the fans would slow down compared to the performance mode, so it actually seemed to be working in that regard

Very interesting -- if you want, you can try to use the old version but with debug enabled and see what is actually reported by the device.

If it is 

```
[ 1118.672827] samsung_galaxybook: [DEBUG] setting performance_mode
[ 1118.672836] samsung_galaxybook: [DEBUG]     43 58 91 00 00 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[ 1118.672838] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 03 0b 00 00 00 00 00 00 00 00
...
[ 1118.674851] samsung_galaxybook: [DEBUG] response was:
[ 1118.674859] samsung_galaxybook: [DEBUG]     43 58 91 00 ff 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[ 1118.674861] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 03 00 00 00 00 00 00 00 00 00
...
```
(e.g. when trying to set performance mode to `0b`, the response value `ff` at position 4 instead of `aa` and then the value at the end after `51 03` is `00` instead of `0b`) 

...then it confirms what I said before (the device is rejecting the payload and possibly setting it to `00` (or maybe just ignoring it?) instead?

If it is "setting to `00` instead" / as a default when the payload is rejected, then it could be that `00` does actually give you a lower mode so that would also match what you said (it sounds like the fans slow down).. but it is still a failure anyway :)

mhagnumdw, 2024-10-10:
> Maybe it would be interesting to see a copy of your DSDT if you can upload one? It can hopefully provide a clue...

```bash
sudo acpidump -n DSDT -b -o dsdt.dat
iasl -d dsdt.dat
```

[dsdt.dsl.txt](https://github.com/user-attachments/files/17331494/dsdt.dsl.txt)

rapgenic, 2024-10-10:
Hi! I had some time so I gave a look at the DSDT. At line 118470 is the code that should be setting the performance mode.
It seems that only `0x02`, `0x0A`, `0x15` and `0x16` actually do something. It appears they write something into the SNVS memory area which is near a DSDT region described as "Embedded Controller".

I have no clue however as to what is the association between values and modes...

joshuagrisham, 2024-10-13:
Hi @rapgenic thanks for looking into this. I think it is actually a little misleading what seems to swap around things in that specific method, because what I have seen the "real stuff" seems to happen in the methods called later (e.g. CHOV / CMDD / etc...) which will always run after (at least that is how it works on my device :)

mhagnumdw, 2024-10-13:
Let me know if you want me to test something.

joshuagrisham, 2024-10-13:
An interesting development, though.. I am pretty sure that I have finally figured out basically where the different values are "picked" that should be used on the device. It is not very good news, either (IMO..) because it actually seems to be hard-coded in the userspace software in Windows. (This is not the answer I would have hoped for...)

The device driver in Windows seems to be basically just providing a dumb/thin "endpoint" to be able to invoke the ACPI methods but does not have any logic or any payload values at all "baked" in to it.

For both using the "Samsung Settings" application as well as the application listening to the hotkey events, they identify the 4 different performance modes as an integer value (0 through 3). Then, there is a background application "SamsungSystemSupportEngine.exe" which is handling the "translation" of these 0,1,2,3 values into the payloads that will actually be sent to the ACPI method "endpoint" of the device driver (including setting the GUID and everything for the payload in this background application actually, not in the driver..).

My assumption is that each different "variant" has a different version of the Samsung background application in Windows (e.g. "SamsungSystemSupportEngine.exe" or whatever it might be named on different devices), and that each different "variant" of the background application presumably has different values hard-coded into it for what to send for each of the different supported modes.

On my device, plus a Galaxy Book3 Pro 360 that I was able to test on, the payloads are exactly the same even though they are running slightly different versions of the application that handles this on our devices. In this case it looks like actually exactly the same app, just a slightly newer version seems to have been shipped to the GB2 Pro (it was from September 2024 instead of May 2024 like was running on the GB3 Pro 360 but otherwise the application looked to be the same "variant", if that makes sense).

Let's just say I was able to "take a peek" and found where this was happening in the app's code, and can confirm for both my device and the GB3 Pro 360 I checked, it is hard-coded as:
- 0 (Silent) = 0xb
- 1 (Quiet) = 0xa
- 2 (Optimized) = 0x2
- 3 (High Performance) = 0x15

> Interestingly enough, the background app has support for 6 different levels, 0 through 5, called "FanOff", "LowNoise", "Optimize", "HighPerf", "OffTable", and "Ultra", but only the first 4 are actually used in the Samsung Settings app and when you toggle the modes with the hotkey  (with slightly different labels given in these apps, per my above list).. then, on my device, "OffTable" and "Quiet" are mapped to the same value 0xa while "Ultra" is mapped to 0x16 which is reported to be supported by the ACPI method, but the Windows apps just don't expose or use either of these :wink: .. all of that beside the point, the only ones that matter if we give the same "experience" with the Linux driver are the first 4!)
>
> The other weird thing was that trying to use Samsung Update to download a copy of this app for various devices always gave me the same result -- a much older version that was 5+MB in size and did not at all match what was really running on the devices (much newer versions and around 1MB in size..) so I have not really figured out how to reliably find all of the different variations of this background application, unless people can take a copy of the one that is actually running on their devices.

So then my ask to @mhagnumdw  and @seaasses  (since your devices are slightly different), if you would like for me to dig into this further for you in the same way, then I would ask for the following:

1. Use WinDbg in Windows and attach to the running SamsungSystemSupportEngine.exe process on your deivce, run some specific tests to get some traces, and provide the traces here (or in email) (more info below, if you need!)
2. Find your own running "SamsungSystemSupportEngine.exe" in the Windows Processes list, Right click on it and click "Open file location", Pack both "SamsungSystemSupportEngine.exe" and the associated .inf file (mine is called "SFourSWComp15.inf", for example) into a zip or similar archive and email that to me

The WinDbg traces will help find what I need to look for in the 2nd point :wink:

Here is slightly more info on using WinDbg for this below, if you need it:

> Download and install WinDbg (e.g. from Microsoft Store)
> Open it, and then go to File > Start Debugging > Attach to process
> Check the box "Show processes from all users"
> Select "SamsungSystemSupportEngine.exe" and click "Attach"
>
> By default it will break once you attach; just click "Go" right after that or type the command g so the program keeps working (we just care about seeing the debug messages in the console)
>
> It can be occasionally helpful to use Command > Clear Window (in the menu ribbon) to clean away too much other stuff from the log.
>
> Take some "pointed tests" and then copy and paste the text from the console log into a text file.
>
> By "pointed tests", I mean:
> One test could be to cycle through all of the different performance modes at least 1-1.5x times using either the hotkey, Samsung Settings, and/or both
> Then, clear the command window and do a simple test to enable and disable "start on lid open", and save that text to a separate file
>etc etc for however many different events we might need or want to look into...

joshuagrisham, 2024-10-13:
In case it helps, an example of what I am looking for with the WinDbg traces will look like this (setting performance mode to "Quiet" a.k.a. "LowNoise"):

```
[24-10-13 12:49:14][T][3]EventHandler<class Registry *>::Invoke > Invoke start: SettingsDBEventListener
[24-10-13 12:49:14][I][8]AbstractPerfMode::SetFanMode > >>>>> PerformanceModeV2 -> FanMode=LowNoise
[24-10-13 12:49:14][T][3]EventHandler<struct DBChangeData>::Invoke > Invoke start: SettingsEventHandler
...
[24-10-13 12:49:14][T][3]EventHandler<class DatabaseEventData>::Invoke > Invoke start: ModulePerformance
[24-10-13 12:49:14][T][3]AbstractModule::DatabaseChanged > DatabaseChanged: ModulePerformance,Data=Name=ModulePerformance,Key=Value,Val=1
[24-10-13 12:49:14][T][3]EventHandler<class DatabaseEventData>::Invoke > Invoke end: ModulePerformance
...
```

(in the above example I have filtered out everything else that does not really matter.. there will be a lot of lines that don't matter but you can just give them all :smile: )

joshuagrisham, 2024-10-13:
@rapgenic based on above info that we can get on this for other devices, we will need to think about how it should be implemented in the Linux driver. 

My "hope" would be that maybe we can create some kind of prioritized mapping model of values per "mode" and then compare that to what is returned in the ACPI method of "supported modes" (basically similar to what you already mentioned here: https://github.com/joshuagrisham/samsung-galaxybook-extras/issues/21#issuecomment-2381358955)

e.g. "Optimized" can be either 0x2 or 0x0, if 0x2 is supported then use 0x2 otherwise use 0x0

The "problem" with this kind of model that I can already see is that if we say "High Performance" is 0x15 (alternative 1) or 0x1 (alternative 2), then on your device it will use 0x15 instead of 0x1 -- does that seem ok? (as you had mentioned that even though your device reports support for 0x15, you always saw 0x1 in the traces??)

Another option is to do DMI-based matching with quirks per different model (or similar) where we hard-code (but we should always probably check against what is returned from `SUBN=0x01` and never try to send anything that is not "supported" no matter what.. ?? is my opinion, anyway!) and/or some kind of mix of the two (e.g. "default" logic based on what is returned vs the priority-based matching model, but also have the ability to hard-code specific values for specific devices?? would rather avoid this if it is possible, though.. getting too complicated I think... :laughing: )

mhagnumdw, 2024-10-13:
Hi @joshuagrisham, here's what you requested. Tks!

> Pack both "SamsungSystemSupportEngine.exe" and the associated .inf file

[SamsungSystemSupportEngine.zip](https://github.com/user-attachments/files/17356208/SamsungSystemSupportEngine.zip)

---

> One test could be to cycle through all of the different performance modes

[performance-test.zip](https://github.com/user-attachments/files/17356213/performance-test.zip)

Via shortcut `Fn + F11`, in this sequence: high performance > quiet > optimized > high performance > quiet > optimized > high performance > quiet > optimized

Via the Samsung settings GUI, follow this order: high performance > quiet > optimized > high performance > quiet > optimized

---

> Then, clear the command window and do a simple test to enable and disable "start on lid open"

[start-up-when-lid-opended-test.zip](https://github.com/user-attachments/files/17356215/start-up-when-lid-opended-test.zip)

It was activated, so I followed this order: deactivated > activated > deactivated > activated

joshuagrisham, 2024-10-13:
Thank you @mhagnumdw , it seems like your device does not actually have a "Silent" mode ?
Also interesting that in the trace it looks the device is using the "Ultra" mode for what the GUI/hotkey notification calls "High Performance"

It seems the first Zip file has a password; is it possible for you to share the password either here or via email?
Thank you!

mhagnumdw, 2024-10-13:
Sorry, @joshuagrisham , I updated the previous comment with the zip file without password.

> it seems like your device does not actually have a "Silent" mode

Yes, there is no `Silent`, just the ones in the image below:

[image]

Just remembering that my model is a `Galaxy Book3 Ultra` with product name NP960XFH-XA1BR: https://www.samsung.com/br/computers/samsung-book/galaxy-book3-ultra-16-inch-i7-32gb-1tb-np960xfh-xa1br/buy/

joshuagrisham, 2024-10-13:
Thanks @mhagnumdw this makes sense!

Now that I have dug a little bit more into this I actually found that there is some mapping logic in the code and it seems like they actually do very similar to what we had thought -- the same ACPI function payload to get "supported modes" is then actually read in by this backend application in Windows and creates an internal "SupportedModeList" table. The values are this:

0x1 = "OffTable" (I think this is similar to / in absence of 0xa "LowNoise" ? )
0x2 = "Optmized"
0xa = "LowNoise"
0xb = "FanOff"
0x15 = "High"
0x16 = "Ultra"

Then for others that we have seen reported as supported by the ACPI (e.g. 0xc, 0x14), the Windows app does not seem to care about (it just ignores them).

What I am thinking/wondering is if we can just map each of these 6 (or maybe 5? e.g. combine 0x1 and 0xa as Samsung has done? )  and map them each in to their own discrete Platform Profile (see https://www.kernel.org/doc/Documentation/ABI/testing/sysfs-platform_profile) , and then read the "supported modes" ACPI result during load of the module in order to dynamically read in the values and enable/disable supported profiles for the platform profile and just give them directly as the ACPI method replies with them.

This can mean that sometimes the device in Linux will be different than how it works in Windows (e.g. in the case of GB2 Pro you will be able to use the "Ultra" mode in Linux but it is not really available in Windows, and hopefully it does not cause problems with the actual device...??)

I can try to work something up for this in the coming days; it seems like the "cleanest" approach that should support anything without needing additional hard-coded lists per device matching string (which could quickly get cumbersome to maintain I think..), and we can test it and see how it works?  The only worry I would have is that it would mean that we are potentially allowing usage of certain "performance modes" that are never used in Windows so they are a bit "untested" on these devices.. however, as we have seen, since we are mostly hitting thermal throttling at 100C in all modes except silent, maybe it does not really matter that much :laughing:

joshuagrisham, 2024-10-15:
@mhagnumdw @seaasses @jcarrano @rapgenic can you please pull a copy of the latest commit on main now, build/install and see if this works any better for this problem for you?

I wrote and re-wrote this a few times, trying to keep it as simple and "straight-forward" as possible, but that will hopefully work across all devices. While I was trying different options, I also noticed that I had some problems with GNOME and other things (e.g. "nothing worked" except via CLI..), and after some snooping around, I realized that there are 3 profiles which are sort of the "standard" ones that a lot of these tools seem to rely on (low-power, balanced, and performance), so I re-thought how the mapping should work based on that, and tried to create a mapping in the code which is biased to always try to make sure to map these 3. Any additional modes supported by the device (based on ACPI method response) I tried to just fit "in between" these three, using information from all of the various traces and logs provided by everyone here.

My hope is that what I landed on here is simple enough that it is understandable, will be quite robust for supporting other and newly-released devices without needing to have custom quirks etc, and will just "work" in a good way :sunglasses: 

Here is the relevant part with the "mapping" logic: https://github.com/joshuagrisham/samsung-galaxybook-extras/blob/49952900736f4c10cea6e06ca810b1192b601b99/samsung-galaxybook.c#L1234-L1295

Please try it and let me know if it works better/worse/your laptop caught on fire/etc ? :innocent:

joshuagrisham, 2024-10-15:
@mhagnumdw for your device specifically I think your experience with the above mapping will be slightly different compared to Windows. Now you should have 4 profiles to choose from with this driver:

- "low-power" is what you see in Windows as "Quiet"
- "balanced" maps to what all other devices seem to call "Optimized" but your device does not allow you to pick this mode in Windows (though your ACPI reports that your device supports it); hopefully it works well on your device but please report back if you experience issues!
- "balanced-performance" is actually the same as what your device in Windows calls "Optmized"; the background program in Windows calls it "HighPerf" and all other devices I have seen that support this mode call it "High performance"
- "performance" is what your device in Windows calls "High performance"; the background program in Windows calls it "Ultra"...

For everyone else in this thread, I think it will be basically exactly the same as you probably see in Windows (same number of available profiles and that they will use the same values as in Windows).

(unless I am thinking wrong or there is a bug in the code ;) )

joshuagrisham, 2024-10-15:
And it case it helps, here is an example on my device (GB2 Pro) of sort of the "expected" result e.g. in the kernel log with the debug module parameter set:

```c
[    3.433333] samsung_galaxybook: initializing performance mode and platform profile
[    3.433337] samsung_galaxybook: [DEBUG] get supported performance modes
[    3.433339] samsung_galaxybook: [DEBUG]     43 58 91 00 00 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[    3.433340] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 01 00 00 00 00 00 00 00 00 00
[    3.433341] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.433342] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.433343] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.433344] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.433344] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.433345] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.433346] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.433347] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.433348] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.433348] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.433349] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.433350] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.433351] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.433352] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.434667] samsung_galaxybook: [DEBUG] response was:
[    3.434670] samsung_galaxybook: [DEBUG]     43 58 91 00 aa 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[    3.434671] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 01 07 00 01 02 0a 0b 14 15 00
[    3.434672] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.434673] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.434673] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.434674] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.434675] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.434676] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.434677] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.434677] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.434678] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.434679] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.434680] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.434681] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.434681] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.434682] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[    3.434684] samsung_galaxybook: will support profile 'performance' with performance mode value 0x15
[    3.434685] samsung_galaxybook: unmapped performance mode value 0x14 will be ignored
[    3.434686] samsung_galaxybook: will support profile 'low-power' with performance mode value 0xb
[    3.434687] samsung_galaxybook: will support profile 'quiet' with performance mode value 0xa
[    3.434688] samsung_galaxybook: will support profile 'balanced' with performance mode value 0x2
[    3.434689] samsung_galaxybook: unmapped performance mode value 0x1 will be ignored
[    3.434690] samsung_galaxybook: unmapped performance mode value 0x0 will be ignored
```

jcarrano, 2024-10-15:
@joshuagrisham your commit works on my device. Here are the logs:

```
# cat /sys/firmware/acpi/platform_profile_choices
low-power balanced performance
```

`Fn-F11` works to cycle though all profiles.

<details>
 <summary>Some logs</summary>

```
 1204.875993] samsung_galaxybook: loading driver
[ 1204.876177] samsung_galaxybook: initializing ACPI device
[ 1204.876214] samsung_galaxybook: [DEBUG] enabling ACPI feature
[ 1204.876216] samsung_galaxybook: [DEBUG]     43 58 7a 00 00 bb aa 00 00 00 00 00 00 00 00 00
[ 1204.876218] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[ 1204.877355] samsung_galaxybook: [DEBUG] response was:
[ 1204.877356] samsung_galaxybook: [DEBUG]     43 58 7a 00 aa dd cc 00 00 00 00 00 00 00 00 00
[ 1204.877358] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[ 1204.877359] samsung_galaxybook: [DEBUG] enabling ACPI feature
[ 1204.877360] samsung_galaxybook: [DEBUG]     43 58 8a 00 00 bb aa 00 00 00 00 00 00 00 00 00
[ 1204.877361] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[ 1204.878361] samsung_galaxybook: [DEBUG] response was:
[ 1204.878362] samsung_galaxybook: [DEBUG]     43 58 8a 00 aa dd cc 00 00 00 00 00 00 00 00 00
[ 1204.878363] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[ 1204.878364] samsung_galaxybook: initializing platform device
[ 1204.878415] samsung_galaxybook: initializing performance mode and platform profile
[ 1204.878416] samsung_galaxybook: [DEBUG] get supported performance modes
[ 1204.878418] samsung_galaxybook: [DEBUG]     43 58 91 00 00 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[ 1204.878419] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 01 00 00 00 00 00 00 00 00 00
[ 1204.878420] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.878421] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.878422] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.878423] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.878424] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.878425] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.878426] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.878427] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.878428] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.878429] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.878430] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.878431] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.878432] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.878433] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.879650] input: AT Translated Set 2 keyboard as /devices/platform/i8042/serio4/input/input24
[ 1204.880283] samsung_galaxybook: [DEBUG] response was:
[ 1204.880285] samsung_galaxybook: [DEBUG]     43 58 91 00 aa 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[ 1204.880287] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 01 06 00 01 02 0a 14 15 00 00
[ 1204.880288] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.880289] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.880290] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.880291] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.880292] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.880293] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.880294] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.880296] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.880297] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.880298] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.880299] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.880300] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.880301] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.880302] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1204.880303] samsung_galaxybook: will support profile 'performance' with performance mode value 0x15
[ 1204.880305] samsung_galaxybook: unmapped performance mode value 0x14 will be ignored
[ 1204.880307] samsung_galaxybook: will support profile 'low-power' with performance mode value 0xa
[ 1204.880308] samsung_galaxybook: will support profile 'balanced' with performance mode value 0x2
[ 1204.880310] samsung_galaxybook: unmapped performance mode value 0x1 will be ignored
[ 1204.880311] samsung_galaxybook: unmapped performance mode value 0x0 will be ignored
[... 95 lines cut ...]
[ 1241.117106] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1241.117107] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1241.118774] samsung_galaxybook: [DEBUG] response was:
[... cut ...]
[ 1241.118799] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1241.118800] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1241.118801] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 1241.118804] samsung_galaxybook: [DEBUG] platform profile is currently 'balanced' (performance mode 0x02)
```

</details>

As a sidenote, when unloading the driver, the keyboard stops working. I will open an issue and upload logs.

jcarrano, 2024-10-15:
One more thing, why is ` balanced    = "optimized", otherwise "performance" when "ultra" is supported` is the proper choice? Seems like "balanced" should use the default for the device. My device does not support ultra so I don't know what that default is.

mhagnumdw, 2024-10-15:
@joshuagrisham, everything looks fine, thank you very much! I can set the profiles in three ways: through Gnome, Fn + F11 and on the command line.

joshuagrisham, 2024-10-15:
> One more thing, why is ` balanced = "optimized", otherwise "performance" when "ultra" is supported` is the proper choice? Seems like "balanced" should use the default for the device. My device does not support ultra so I don't know what that default is.

This is actually more to cater for the GB3 Ultra (e.g. NP960XFH-XA1BR) case like shown here: https://github.com/joshuagrisham/samsung-galaxybook-extras/issues/31#issuecomment-2409090478

The problem/"strangeness" with the setup for this device is that in Windows they are using the SamsungSystemSupportEngine.exe mode "High" 0x15 as what they call "Optimized" in the actual user GUIs (Samsung Settings app and the toast notification when you press hotkeys etc). So the "mapping" for that device is like this:

|Samsung Settings name|SamsungSystemSupportEngine.exe Mode|samsung-galaxybook profile|
|---|---|---|
|"High performance" | "Ultra" 0x16 | "performance" |
|"Optimized" | "High" 0x15 | "balanced" |
|"Quiet" | "LowNoise" 0xa | "low-power" |

But, as mentioned, I think you will get the best experience if the driver can map to the profiles `performance`, `balanced`, and `low-power` otherwise there are things that don't work in power-profiles-daemon, GNOME, etc.  So my goal was that even for this device, it should match to those 3 profiles, but luckily/incidentally it also aligns pretty well to the same names that are present in Windows with Samsung Settings.

For other devices which do not have the "Ultra" mode (my guess is it is only for actual "Ultra" devices? :laughing: ) then I guess it will be more "normal" looking.. e.g. for my device (Galaxy Book2 Pro 15") the table looks more like this:

|Samsung Settings display name|SamsungSystemSupportEngine.exe Mode|samsung-galaxybook profile|
|---|---|---|
|"High performance" | "High" 0x15 | "performance" |
|"Optimized" | "Optmized" 0x2 | "balanced" |
|"Quiet" | "LowNoise" 0xa | "quiet" |
|"Silent" | "FanOff" 0xb | "low-power" |

So that was kind of the goal of what I tried to achieve with the logic that I wrote ... that each different variation should ideally get `performance`, `balanced`, and `low-power` profiles so that you get nice out-of-the-box experience with userspace tools and hopefully it also matches relatively well with how it works in Windows?

(hopefully I achieved this without making the code super-complicated or inefficient; this was my goal anyway and what I landed on after at least 4 or 5 different versions :laughing: )

joshuagrisham, 2024-10-15:
I think it is also good to add, the table in the above comment was the "goal" and how I wanted to match what this device driver will do vs what will work well in Linux vs how it works in Windows. However, even more specifically, for the GB3 Ultra, what I said in https://github.com/joshuagrisham/samsung-galaxybook-extras/issues/31#issuecomment-2414426903 actually gives a slight twist to that specific device. :wink: 

Actually, the "real" result of the mapping table (at least how it sits in the code right now) for the GB3 Ultra given by @mhagnumdw  should be like this:

|Samsung Settings name|SamsungSystemSupportEngine.exe Mode|samsung-galaxybook profile|
|---|---|---|
|"High performance" | "Ultra" 0x16 | "performance" |
|"Optimized" | "High" 0x15 | "balanced-performance" |
|N/A | "Optimized" 0x2 | "balanced" |
|"Quiet" | "LowNoise" 0xa | "low-power" |

(i.e. it is using "Optimized" from the ACPI method a bit how it seems to work for other devices, even though it is not even possible to select that mode in Windows, and the "High" 0x15 mode is actually mapped to "balanced-performance" in between the two :sunglasses: )

Just in case anyone else has a GB3 Ultra and sees this, hopefully it makes it more clear.

My point from https://github.com/joshuagrisham/samsung-galaxybook-extras/issues/31#issuecomment-2414836551 still stands, though.. the goal with the logic here was to map to the three primary profiles commonly used in Linux and then try to fit any other possible "supported" modes in between these 3 standard profiles wherever they seem to best fit.

rapgenic, 2024-10-15:
Hi @joshuagrisham, I want to add some information! I have performed your windbg trick to understand better what's happening (btw you have no idea how much reverse engineering I did to print those same logs, if only I knew it were so easily available in windbg...).

It appears that my pc uses a different version of the "performance module" of the samsung driver: your example log (and also my let's say explorations of the binary) suggested that there are two modules that handle the performance mode, called _PerformanceModeV1_ and _PerformanceModeV2_. My PC supports the V1. See the attached file.

[samsungsystemsupportengine.logs.txt](https://github.com/user-attachments/files/17384603/samsungsystemsupportengine.logs.txt)

Should we take into consideration this when deciding what our driver does? Also I am trying to catch up with all the information you have collected and try your proposed driver, I must say that I am very impressed and inspiered of your dedication!

rapgenic, 2024-10-15:
Going through what I did when I analyzed my pc, I can see that a similar function to get the performance modes exists also for the _PerformanceMode_ module, which checks if the response from the CSXI method contains one of the following values:

* 0x00: optimized
* 0x01: performance
* 0x0a: quiet
* 0x0b: silent
* 0x15: _unknown_

Unfortunately there is no string that actually says what they refer to. However i could map them from analyzing the traces to the 4 performance modes. The 0x15 mode is actually unknown.

I believe we should look for a method to distinguish which module is supported by each laptop, I don't know if something exists in the driver itself or if it is hardcoded with the laptop model. However it seems that the two versions are quite different, so we should do the same in my opinion

joshuagrisham, 2024-10-15:
Hi @rapgenic , I also saw the V1 version and all of its values, and based on your prior info I actually tried to account for the case of 0x0 "Normal" = "Optimized" and 0x1 "OffTable" = "Performance" already in the logic of the code. However, "performance"=0x1 will only "kick in" if the device does not have 0x16 or 0x15 in its supported list, and "balanced" = 0x0 will only "kick in" if the device does not have 0x2 in its supported list.

From what I can tell in your device's DSDT it will actually pick up 0x15 as "performance" but then there is no 0x2 so it will use 0x0 for "Optimized"/"balanced", but I would be curious to hear if 0x15 seems to work ok and otherwise I guess/hope everything already checks out as it is in the latest version on main?

seaasses, 2024-10-16:
Hey guys, I installed Windows today and here I am again. It seems that my laptop uses the V2 Performance Mode as you can see in the windbg logs ([Optimized] -> Performance -> Silent -> Optimized -> Performance -> Silent -> Optimized):
[change_performace_with_gui.txt](https://github.com/user-attachments/files/17387224/change_performace_with_gui.txt)
[change_performance_with_button.txt](https://github.com/user-attachments/files/17387228/change_performance_with_button.txt)

Here are the SamsungSystemSupportEngine.exe and its .inf file:
[SamsungSystemSupportEngine.zip](https://github.com/user-attachments/files/17387429/SamsungSystemSupportEngine.zip)

I tested the new updated main and I do not get errors, but I don't know if the high performance mode is actually active as i got the same number of completed operations using `sudo stress-ng --cpu 0 --cpu-load 100 --metrics-brief --perf -t 20`.

Here are the logs for the lid open change ([on] -> off -> on -> off -> on):
[change_turn_on_when_lid_open.txt](https://github.com/user-attachments/files/17387594/change_turn_on_when_lid_open.txt)

mhagnumdw, 2024-10-16:
I'll leave here that it may have some relevance: after updating the kernel (dnf update), the driver was not reloaded, so I rebooted, installed the driver again, the driver was installed successfully, but when changing the profile through Gnome, it did not change `/sys/firmware/acpi/platform_profile` and no error appeared in dmesg (or in journalctl) even with the driver in debug mode. After rebooting, it was ok, that is, changing the profile through Gnome also changed `/sys/firmware/acpi/platform_profile`.

rapgenic, 2024-10-16:
Hi @joshuagrisham I tested your driver and it works as you thought. In theory everything is good, I have only two doubts:
* We don't know whether setting values that make sense or not... sure it seems we can set them but we cannot say for sure
* A question that popped into my mind is does it make sense to use the "quiet" galaxy mode for low-power platform mode? It seems that the "silent" galaxy mode is the only one actually doing something (I managed to squeeze a LOT more battery time using that mode...)

>  installed the driver again, the driver was installed successfully, but when changing the profile through Gnome, it did not change

@mhagnumdw I confirm what you saw. It seems that if (I think) the `power-profiles-daemon` boots without the driver loaded it simply ignores it even if you load it later.

jcarrano, 2024-10-17:
@joshuagrisham I also have a Galaxy Book2 Pro 15 (the XEE version though), but I have 3 choices. You say you have

low-power quiet balanced performance

but I'm missing the quiet.

joshuagrisham, 2024-10-18:
Hi @jcarrano as your device has Intel Arc my guess is that they did not want to allow Samsung's "silent" mode for risk of overheating, so my guess is that your device does not support performance mode 0xb at all, and the logic in the driver is now mapping 0xa ("quiet") to low-power instead? (If so, then "working as intended" and you should be getting the best experience possible IMO :sunglasses:  :+1: ). 

If you check Windows, do you only have 3 modes?

If you check your kernel log during module load with the parameter `debug=on` then my guess is that it might look something like this???

```
[13210.045538] samsung_galaxybook: initializing performance mode and platform profile
[13210.045541] samsung_galaxybook: [DEBUG] get supported performance modes
[13210.045544] samsung_galaxybook: [DEBUG]     43 58 91 00 00 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[13210.045547] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 01 00 00 00 00 00 00 00 00 00
...
[13210.049005] samsung_galaxybook: [DEBUG] response was:
[13210.049009] samsung_galaxybook: [DEBUG]     43 58 91 00 aa 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[13210.049011] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 01 06 00 01 02 0a 14 15 00 00
...
[13210.049044] samsung_galaxybook: will support profile 'performance' with performance mode value 0x15
[13210.049048] samsung_galaxybook: unmapped performance mode value 0x14 will be ignored
[13210.049051] samsung_galaxybook: will support profile 'low-power' with performance mode value 0xa
[13210.049054] samsung_galaxybook: will support profile 'balanced' with performance mode value 0x2
[13210.049057] samsung_galaxybook: unmapped performance mode value 0x1 will be ignored
[13210.049059] samsung_galaxybook: unmapped performance mode value 0x0 will be ignored
```