joshuagrisham/samsung-galaxybook-extras #32: Error when turning start on lid open ON

seaasses, 2024-10-10:
Galaxy Book 3
- Product Name: NP750XFH-XS1BR
- Kernel: 6.8.0-31-generic (x64)
- OS: Ubuntu 24.04.1 LTS
- Driver version commit hash: https://github.com/joshuagrisham/samsung-galaxybook-extras/commit/f1aba1a7a48ec32b0a32d9eba150be998cae1b07

The `sudo dmidecode --type 3` output is

```
Getting SMBIOS data from sysfs.
SMBIOS 3.5.0 present.

Handle 0x0003, DMI type 3, 22 bytes
Chassis Information
	Manufacturer: SAMSUNG ELECTRONICS CO., LTD.
	Type: Notebook
	Lock: Not Present
	Version: N/A
	Serial Number: ------
	Asset Tag: No Asset Tag
	Boot-up State: Safe
	Power Supply State: Safe
	Thermal State: Safe
	Security Status: None
	OEM Information: 0x00000000
	Height: Unspecified
	Number Of Power Cords: 1
	Contained Elements: 0
	SKU Number: Samsung Chassis
```

## The issue

When running `echo true | sudo tee /sys/devices/platform/samsung-galaxybook/start_on_lid_open` I get the error bellow error message

> tee: /sys/devices/platform/samsung-galaxybook/start_on_lid_open: Invalid argument

The error does not occur disabling the lid open option with `echo false | sudo tee /sys/devices/platform/samsung-galaxybook/start_on_lid_open`

## Kernel Logs
```
[ 3293.551620] samsung_galaxybook: [DEBUG] input notification event: 0x61
[ 3378.240994] samsung_galaxybook: [DEBUG] setting start_on_lid_open
[ 3378.241001] samsung_galaxybook: [DEBUG]     43 58 7a 00 00 82 a3 80 01 00 00 00 00 00 00 00
[ 3378.241004] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[ 3378.242131] samsung_galaxybook: [DEBUG] response was:
[ 3378.242134] samsung_galaxybook: [DEBUG]     43 58 7a 00 aa ff 00 00 00 00 00 00 00 00 00 00
[ 3378.242135] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[ 3378.242137] samsung_galaxybook: invalid response when setting start_on_lid_open; returned value was: 0x00 0x00
```
## Final comment
In fact I think the change takes place because opening the lid does turn the laptop on (I will test more), but the error message tells me that there is something to change in the code

seaasses, 2024-10-10:
Here is the full log when the driver is loaded
```
[ 3866.477959] samsung_galaxybook: loading driver
[ 3866.478174] samsung_galaxybook: [DEBUG] received following device quirks:
[ 3866.478177] samsung_galaxybook: [DEBUG]   disable_kbd_backlight       = false
[ 3866.478179] samsung_galaxybook: [DEBUG]   disable_battery_threshold   = false
[ 3866.478181] samsung_galaxybook: [DEBUG]   disable_performance_mode    = false
[ 3866.478182] samsung_galaxybook: [DEBUG]   disable_fan_speed           = false
[ 3866.478184] samsung_galaxybook: [DEBUG]   disable_i8042_filter        = false
[ 3866.478185] samsung_galaxybook: [DEBUG]   disable_acpi_hotkeys        = false
[ 3866.478187] samsung_galaxybook: [DEBUG]   disable_wmi_hotkeys         = false
[ 3866.478189] samsung_galaxybook: [DEBUG] performance_modes override exist in quirks
[ 3866.478362] samsung_galaxybook: initializing ACPI device
[ 3866.478427] samsung_galaxybook: [DEBUG] enabling ACPI feature
[ 3866.478430] samsung_galaxybook: [DEBUG]     43 58 7a 00 00 bb aa 00 00 00 00 00 00 00 00 00
[ 3866.478433] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[ 3866.479540] samsung_galaxybook: [DEBUG] response was:
[ 3866.479543] samsung_galaxybook: [DEBUG]     43 58 7a 00 aa dd cc 00 00 00 00 00 00 00 00 00
[ 3866.479546] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[ 3866.479548] samsung_galaxybook: [DEBUG] enabling ACPI feature
[ 3866.479550] samsung_galaxybook: [DEBUG]     43 58 8a 00 00 bb aa 00 00 00 00 00 00 00 00 00
[ 3866.479552] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[ 3866.480479] samsung_galaxybook: [DEBUG] response was:
[ 3866.480482] samsung_galaxybook: [DEBUG]     43 58 8a 00 aa dd cc 00 00 00 00 00 00 00 00 00
[ 3866.480484] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[ 3866.480486] samsung_galaxybook: initializing platform device
[ 3866.480574] samsung_galaxybook: initializing performance_mode
[ 3866.480576] samsung_galaxybook: [DEBUG] get supported performance modes
[ 3866.480578] samsung_galaxybook: [DEBUG]     43 58 91 00 00 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[ 3866.480580] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 00 00 00 00 00 00 00 00 00 00
[ 3866.480582] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480583] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480585] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480587] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480588] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480590] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480591] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480593] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480595] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480596] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480598] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480599] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480601] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.480603] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482816] samsung_galaxybook: [DEBUG] response was:
[ 3866.482827] samsung_galaxybook: [DEBUG]     43 58 91 00 aa 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[ 3866.482832] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 00 01 01 01 01 00 00 00 00 00
[ 3866.482834] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482836] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482838] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482840] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482842] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482844] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482846] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482847] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482849] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482851] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482853] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482854] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482856] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482858] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[ 3866.482861] samsung_galaxybook: [DEBUG] get performance mode values
[ 3866.482863] samsung_galaxybook: [DEBUG]     43 58 91 00 00 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[ 3866.482865] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 01 00 00 00 00 00 00 00 00 00
[ 3866.482866] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
[ 3866.493219] samsung_galaxybook: [DEBUG]     43 58 86 00 aa 02 00 00 00 00 00 00 00 00 00 00
[ 3866.493221] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[ 3866.493222] samsung_galaxybook: initializing hotkey input device
[ 3866.493280] input: Samsung Galaxy Book extra buttons as /devices/platform/samsung-galaxybook/input/input40
[ 3866.493348] samsung_galaxybook: enabling WMI notifications
[ 3866.493354] samsung_galaxybook: installing WMI notify handler
[ 3866.493386] samsung_galaxybook: driver successfully loaded

```

joshuagrisham, 2024-10-10:
Thanks for this @seaasses ! Hrmm it seems eerily similar to the "issue" on #31 
Would it be possible for you to try to build and run the previous version of the module with debug enabled and see if you get a similar payload response when setting this feature to on vs off?

Also does your laptop in fact even have this feature like when you look in Windows at the Samsung Settings app?

joshuagrisham, 2024-10-10:
Side note: @rapgenic FYI here is a case where checking for pos 4 != 0xaa does not actually catch a failure (pos 4  == 0xaa && pos 5  == 0xff)

joshuagrisham, 2024-10-10:
@seaasses I realized here also it would help with a copy of your DSDT .. either in this issue or #31 if you can please upload it then I think we can actually tell exactly what needs to be adjusted.

Then I will need to tweak the driver a bit to have a DMI match-based quirk maybe or at least to figure out some kind of logic to be able to steer detection of this, since it seems to be different behavior for devices that both have `SAM0429` that we need to account for.

rapgenic, 2024-10-10:
>Side note: @rapgenic FYI here is a case where checking for pos 4 != 0xaa does not actually catch a failure (pos 4 == 0xaa && pos 5 == 0xff)

:fearful: I really never saw that! Do you think we should make the double check then? I still think that pos 4 is significant (it's also called RFLG which kinda sounds like return flag)

I suppose that also `pos 5 == 0xff => error` is correct, since no one reported to you it in the past...

seaasses, 2024-10-13:
> Thanks for this @seaasses ! Hrmm it seems eerily similar to the "issue" on #31 Would it be possible for you to try to build and run the previous version of the module with debug enabled and see if [...]

Hey, @joshuagrisham, sorry for the delay in getting back to you. Yes, actually I have similar issue with the battery saver mode too.

I'm pretty sure that the laptop have this option when using the Samsung Settings. I got the below payload using the https://github.com/joshuagrisham/samsung-galaxybook-extras/commit/5c4456980da0180fbc11ab4169ad39b2c6ceee7b version

```
[98396.301117] samsung_galaxybook: [DEBUG] setting start_on_lid_open
[98396.301123] samsung_galaxybook: [DEBUG]     43 58 7a 00 00 82 a3 80 01 00 00 00 00 00 00 00
[98396.301124] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[98396.303113] samsung_galaxybook: [DEBUG] response was:
[98396.303115] samsung_galaxybook: [DEBUG]     43 58 7a 00 aa 00 00 01 00 00 00 00 00 00 00 00
[98396.303116] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[98396.303117] samsung_galaxybook: [DEBUG] returned start_on_lid_open value was: 0x00 (calculated start_on_lid_open on/off as 0)
[98396.303118] samsung_galaxybook: turned start_on_lid_open on (1)
```
I will get a copy of my DSDT and will post here soon

seaasses, 2024-10-13:
Here is a copy of my DSDT
[dsdt.dsl.txt](https://github.com/user-attachments/files/17355833/dsdt.dsl.txt)

joshuagrisham, 2024-10-13:
Thank you @seaasses it is actually interesting that the older version is sending exactly the same payload but the command is succeeding (even though it is actually recognizing the wrong return value :face_in_clouds:  ).. is it possible to give the entire full log like you did in https://github.com/joshuagrisham/samsung-galaxybook-extras/issues/32#issuecomment-2405240056, but with this older version of the driver? Wondering if maybe there is a regression in some kind of init payload or something...

Also "worst case scenario" is that I might eventually ask you for WinDbg trace plus copy of your background app in Windows, but maybe we don't need that yet for this case (like I wrote in https://github.com/joshuagrisham/samsung-galaxybook-extras/issues/31#issuecomment-2408983829) :innocent:

seaasses, 2024-10-13:
About the other issues:

# Battery saver
## Version https://github.com/joshuagrisham/samsung-galaxybook-extras/commit/5c4456980da0180fbc11ab4169ad39b2c6ceee7b
```
[99634.445284] samsung_galaxybook: [DEBUG] setting battery_saver
[99634.445290] samsung_galaxybook: [DEBUG]     43 58 7a 00 00 82 e9 90 00 00 00 00 00 00 00 00
[99634.445291] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[99634.447733] samsung_galaxybook: [DEBUG] response was:
[99634.447735] samsung_galaxybook: [DEBUG]     43 58 7a 00 aa 00 00 00 00 00 00 00 00 00 00 00
[99634.447736] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[99634.447737] samsung_galaxybook: [DEBUG] returned battery_saver value was: 0x00 (calculated battery_saver on/off as 0)
[99634.447739] samsung_galaxybook: turned battery_saver off (0)
[99637.475061] samsung_galaxybook: [DEBUG] setting battery_saver
[99637.475065] samsung_galaxybook: [DEBUG]     43 58 7a 00 00 82 e9 90 50 00 00 00 00 00 00 00
[99637.475067] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[99637.476781] samsung_galaxybook: [DEBUG] response was:
[99637.476784] samsung_galaxybook: [DEBUG]     43 58 7a 00 aa 00 00 00 00 00 00 00 00 00 00 00
[99637.476784] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[99637.476785] samsung_galaxybook: [DEBUG] returned battery_saver value was: 0x00 (calculated battery_saver on/off as 0)
[99637.476787] samsung_galaxybook: turned battery_saver on (1)
[99654.946122] samsung_galaxybook: [DEBUG] getting battery_saver
[99654.946127] samsung_galaxybook: [DEBUG]     43 58 7a 00 00 82 e9 91 00 00 00 00 00 00 00 00
[99654.946128] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[99654.947784] samsung_galaxybook: [DEBUG] response was:
[99654.947786] samsung_galaxybook: [DEBUG]     43 58 7a 00 aa 00 00 50 00 00 00 00 00 00 00 00
[99654.947787] samsung_galaxybook: [DEBUG]     00 00 00 00 00
```
## Newer version
Trying to set to 80 (failed) and 0.
```
[100509.903720] samsung_galaxybook: [DEBUG] getting battery charge_control_end_threshold
[100509.903725] samsung_galaxybook: [DEBUG]     43 58 7a 00 00 82 e9 91 00 00 00 00 00 00 00 00
[100509.903726] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[100509.905310] samsung_galaxybook: [DEBUG] response was:
[100509.905312] samsung_galaxybook: [DEBUG]     43 58 7a 00 aa 00 00 50 00 00 00 00 00 00 00 00
[100509.905313] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[100509.905314] samsung_galaxybook: [DEBUG] battery charge control is currently on; battery charge_control_end_threshold is 80
[100517.503423] samsung_galaxybook: [DEBUG] setting battery charge_control_end_threshold
[100517.503429] samsung_galaxybook: [DEBUG]     43 58 7a 00 00 82 e9 90 50 00 00 00 00 00 00 00
[100517.503430] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[100517.506024] samsung_galaxybook: [DEBUG] response was:
[100517.506027] samsung_galaxybook: [DEBUG]     43 58 7a 00 aa 00 00 00 00 00 00 00 00 00 00 00
[100517.506028] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[100517.506029] samsung_galaxybook: invalid response when setting charge_control_end_threshold; returned value was: 0x00 0x00
[100526.503368] samsung_galaxybook: [DEBUG] setting battery charge_control_end_threshold
[100526.503373] samsung_galaxybook: [DEBUG]     43 58 7a 00 00 82 e9 90 00 00 00 00 00 00 00 00
[100526.503374] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[100526.505267] samsung_galaxybook: [DEBUG] response was:
[100526.505269] samsung_galaxybook: [DEBUG]     43 58 7a 00 aa 00 00 00 00 00 00 00 00 00 00 00
[100526.505270] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[100526.505271] samsung_galaxybook: set battery charge_control_end_threshold to 0
```
# USB Charging mode
Actually I don't know if my laptop support this, but here are the logs. 

## Version https://github.com/joshuagrisham/samsung-galaxybook-extras/commit/5c4456980da0180fbc11ab4169ad39b2c6ceee7b
```
[99862.141481] samsung_galaxybook: [DEBUG] getting usb_charging
[99862.141487] samsung_galaxybook: [DEBUG]     43 58 67 00 00 80 00 00 00 00 00 00 00 00 00 00
[99862.141488] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[99862.142233] samsung_galaxybook: failed to get response from getting usb_charging; failure code 0xff was reported from the device
[99862.142237] samsung_galaxybook: [DEBUG] response was:
[99862.142238] samsung_galaxybook: [DEBUG]     43 58 67 00 aa ff 00 00 00 00 00 00 00 00 00 00
[99862.142239] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[99867.528036] samsung_galaxybook: [DEBUG] setting usb_charging
[99867.528041] samsung_galaxybook: [DEBUG]     43 58 68 00 00 81 00 00 00 00 00 00 00 00 00 00
[99867.528043] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[99867.528760] samsung_galaxybook: failed to get response from setting usb_charging; failure code 0xff was reported from the device
[99867.528764] samsung_galaxybook: [DEBUG] response was:
[... 9 lines cut ...]
```
## Newer version
```
[100685.255143] samsung_galaxybook: [DEBUG] getting usb_charge
[100685.255148] samsung_galaxybook: [DEBUG]     43 58 67 00 00 80 00 00 00 00 00 00 00 00 00 00
[100685.255149] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[100685.255870] samsung_galaxybook: [DEBUG] response was:
[100685.255872] samsung_galaxybook: [DEBUG]     43 58 67 00 aa ff 00 00 00 00 00 00 00 00 00 00
[100685.255872] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[100685.255873] samsung_galaxybook: [DEBUG] usb_charge is currently on (1)
[100690.936651] samsung_galaxybook: [DEBUG] setting usb_charge
[... cut ...]
[100697.408200] samsung_galaxybook: [DEBUG] setting usb_charge
[100697.408205] samsung_galaxybook: [DEBUG]     43 58 68 00 00 80 00 00 00 00 00 00 00 00 00 00
[100697.408207] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[100697.409076] samsung_galaxybook: [DEBUG] response was:
[100697.409079] samsung_galaxybook: [DEBUG]     43 58 68 00 aa ff 00 00 00 00 00 00 00 00 00 00
[100697.409080] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[100697.409082] samsung_galaxybook: invalid response when setting usb_charge; returned value was: 0xff
``

joshuagrisham, 2024-10-13:
Note to myself: need to also look into this compared to what is available for the fan device in the DSDT .. :) 

```
[ 3866.490498] samsung_galaxybook: Failed to read fan speed list
[ 3866.490502] samsung_galaxybook: Unable to get list of fan speeds
```

seaasses, 2024-10-13:
> Thank you @seaasses it is actually interesting that the older version is sending exactly the same payload but the command is succeeding (even though it is actually recognizing the wrong return value 😶‍🌫️ ).. is it possible to give the entire full log like you did in [#32 (comment)](https://github.com/joshuagrisham/samsung-galaxybook-extras/issues/32#issuecomment-2405240056), but with this older version of the driver? Wondering if maybe there is a regression in some kind of init payload or something...
> 
> Also "worst case scenario" is that I might eventually ask you for WinDbg trace plus copy of your background app in Windows, but maybe we don't need that yet for this case (like I wrote in [#31 (comment)](https://github.com/joshuagrisham/samsung-galaxybook-extras/issues/31#issuecomment-2408983829)) 😇

Yes, of course, here is the full log

```
[101288.072356] samsung_galaxybook: loading driver
[101288.072557] samsung_galaxybook: initializing ACPI device
[101288.072559] samsung_galaxybook: [DEBUG] initializing 0x7a controls
[101288.072560] samsung_galaxybook: [DEBUG]     43 58 7a 00 00 bb aa 00 00 00 00 00 00 00 00 00
[101288.072561] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[101288.073416] samsung_galaxybook: [DEBUG] response was:
[101288.073418] samsung_galaxybook: [DEBUG]     43 58 7a 00 aa dd cc 00 00 00 00 00 00 00 00 00
[101288.073419] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[101288.073420] samsung_galaxybook: device controls successfully initialized via ACPI method CSFI
[101288.073421] samsung_galaxybook: initializing platform device
[101288.073471] samsung_galaxybook: initializing performance_mode
[101288.073472] samsung_galaxybook: [DEBUG] initializing performance_mode control step 1
[101288.073473] samsung_galaxybook: [DEBUG]     43 58 91 00 00 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[101288.073474] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 01 00 00 00 00 00 00 00 00 00
[101288.073475] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.073476] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.073477] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.073478] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.073478] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.073479] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.073480] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.073481] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.073482] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.073482] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.073483] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.073484] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.073485] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.073485] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.074490] samsung_galaxybook: [DEBUG] response was:
[101288.074492] samsung_galaxybook: [DEBUG]     43 58 91 00 aa 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[101288.074493] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 01 06 00 01 02 0a 14 15 00 00
[101288.074493] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.074494] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.074495] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.074495] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.074496] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.074497] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.074497] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.074504] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.074504] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.074505] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.074506] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.074507] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.074508] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.074508] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.074509] samsung_galaxybook: [DEBUG] initializing performance_mode control step 2
[101288.074510] samsung_galaxybook: [DEBUG]     43 58 91 00 00 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[101288.074511] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 00 00 00 00 00 00 00 00 00 00
[101288.074512] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.074513] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[101288.074514] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
[101288.283789] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[101288.283796] samsung_galaxybook: ACPI notifications successfully enabled via ACPI methods SDLS and CSFI
[101288.283797] samsung_galaxybook: initializing hotkey input device
[101288.283837] input: Samsung Galaxybook extra buttons as /devices/platform/samsung-galaxybook/input/input51
[101288.283945] samsung_galaxybook: enabling WMI notifications
[101288.283951] samsung_galaxybook: installing WMI notify handler
[101288.283983] samsung_galaxybook: driver successfully loaded
```

seaasses, 2024-10-13:
> Also "worst case scenario" is that I might eventually ask you for WinDbg trace plus copy of your background app in Windows, but maybe we don't need that yet for this case (like I wrote in [#31 [...]

Based on what you said in [#31 (comment)](https://github.com/joshuagrisham/samsung-galaxybook-extras/issues/31#issuecomment-2408983829) it seems that the WinDbg trace will be required. I completely removed Windows, but will reinstall in dual boot to get these.

joshuagrisham, 2024-10-15:
Hi @seaasses wow sorry to hear about your USB port :( but hopefully it will be ok!! Please let me know if it seems like it could have been caused by the driver or something else strange (thought I would hope not as this should be protected by Samsung's ACPI I think....)

Anyway in the meantime I have tried to read through some differences and making a "best guess" here at something that might could be contributing to the problem.. can you please try the branch here and see if it fixes this problem for you?

https://github.com/joshuagrisham/samsung-galaxybook-extras/tree/issue-32-error-when-turning-start-on-lid-open-on

joshuagrisham, 2024-10-16:
@seaasses also took a quick look at your DSDT regarding the fan speed... your device seems to have 2 ACPI fan devices instead of 1 ("CPU Fan" and "GPU Fan") and the _FST method looks a bit different than mine.. maybe it actually works? :wink: 

If you set the module parameter `fan_speed=off` and then after the module is loaded, check either `sudo sensors` or see if there are any devices under `/sys/bus/acpi/devices/PNP0C0B*` with an attribute `fan_speed_rpm`, do they have any value?

If so then I can add as a quirk for `SAM0429` to disable fan_speed workaround/implementation from this driver :)

joshuagrisham, 2024-10-16:
Regarding battery saver, I noticed a small tweak that was needed and pushed a change to the same branch.. wondering if you can try the latest in https://github.com/joshuagrisham/samsung-galaxybook-extras/tree/issue-32-error-when-turning-start-on-lid-open-on and see if you still have any issues? :innocent: 

Regarding "usb charge" mode, yes it would be interesting to see what you have in Windows for this.. the old version of this driver also seemed to fail on your device?

seaasses, 2024-10-16:
> Anyway in the meantime I have tried to read through some differences and making a "best guess" here at something that might could be contributing to the problem.. can you please try the branch here and see if it fixes this problem for you?

Hey, Joshua, I tested the branch [issue-32-error-when-turning-start-on-lid-open-on](https://github.com/joshuagrisham/samsung-galaxybook-extras/tree/issue-32-error-when-turning-start-on-lid-open-on) and got errors. Here are the logs when the driver is loaded: 

```
[  825.303099] samsung_galaxybook: loading driver
[  825.303529] samsung_galaxybook: initializing ACPI device
[  825.303620] samsung_galaxybook: initializing ACPI power management features
[  825.303623] samsung_galaxybook: [DEBUG] enabling ACPI feature
[  825.303628] samsung_galaxybook: [DEBUG]     43 58 7a 00 00 bb aa 00 00 00 00 00 00 00 00 00
[  825.303632] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  825.304776] samsung_galaxybook: [DEBUG] response was:
[  825.304782] samsung_galaxybook: [DEBUG]     43 58 7a 00 aa dd cc 00 00 00 00 00 00 00 00 00
[  825.304785] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  825.304788] samsung_galaxybook: initializing platform device
[  825.304943] samsung_galaxybook: initializing performance mode and platform profile
[  825.304947] samsung_galaxybook: [DEBUG] get supported performance modes
[  825.304951] samsung_galaxybook: [DEBUG]     43 58 91 00 00 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[  825.304953] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 01 00 00 00 00 00 00 00 00 00
[  825.304956] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.304958] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.304960] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.304962] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.304965] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.304967] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.304969] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.304972] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.304974] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.304976] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.304978] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.304980] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.304983] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.304985] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.307785] samsung_galaxybook: [DEBUG] response was:
[  825.307790] samsung_galaxybook: [DEBUG]     43 58 91 00 aa 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[  825.307792] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 01 06 00 01 02 0a 14 15 00 00
[  825.307794] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.307795] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.307797] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.307799] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.307800] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.307802] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.307804] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.307805] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.307807] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.307809] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.307810] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.307812] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.307813] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.307815] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.307818] samsung_galaxybook: will support profile 'performance' with performance mode value 0x15
[  825.307823] samsung_galaxybook: unmapped performance mode value 0x14 will be ignored
[  825.307825] samsung_galaxybook: will support profile 'low-power' with performance mode value 0xa
[  825.307827] samsung_galaxybook: will support profile 'balanced' with performance mode value 0x2
[  825.307829] samsung_galaxybook: unmapped performance mode value 0x1 will be ignored
[  825.307831] samsung_galaxybook: unmapped performance mode value 0x0 will be ignored
[  825.307838] samsung_galaxybook: [DEBUG] getting performance_mode
[  825.307840] samsung_galaxybook: [DEBUG]     43 58 91 00 00 8d 02 46 82 ca 8b 55 4a ba 0f 6f
[  825.307842] samsung_galaxybook: [DEBUG]     1e 6b 92 1b 8f 51 02 00 00 00 00 00 00 00 00 00
[  825.307844] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[  825.307846] samsung_galaxybook: [DEBUG]     00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
Adapter: ISA adapter
in0:           5.00 V  (min =  +5.00 V, max =  +5.00 V)
curr1:         0.00 A  (max =  +3.00 A)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +38.0°C  

BAT1-acpi-0
Adapter: ACPI interface
in0:          14.89 V  
curr1:       917.00 mA 
```
It seems that there isn't a `fan_speed_rpm`

And I have a question, why do the performance mode hotkeys requires kernel 6.10 or later?
`[ 3596.479728] samsung_galaxybook: performance mode hotkey requires kernel version 6.10 or higher`

joshuagrisham, 2024-10-16:
Hi @seaasses , now hopefully we are getting somewhere :)
Can you please try again, I have updated the branch with additional tweaks.

More info if you are curious:
For other devices we were checking the response from the "set" command but it looks like your device does not follow the exact same API for these methods. In theory as long as we are checking for the success (0xaa) and lack of failure code (0xff) in the responses (which we are doing for all method calls) then it should work and we do not need to do these extra steps (they are overkill.. as you might see yourself in the log the setting was actually holding it was just spitting this error anyway..)

### hotkey support in 6.10+ only

Regarding question on hotkey, well.. it was because in 6.10 they added this function that takes care of the cycle for you (`platform_profile_cycle`)... otherwise the logic is kind of complicated, and instead of building my own custom logic, I decided to use that new function to take care of this toggling. Anyway if/when we try to get this submitted for mainline, I will even take away the print from the log like you pasted and it will ONLY work with the version of the kernel from that point and moving forward 6.12+ depending on when we can get all of these issues ironed out and hopefully submit a patch to mainline :)

I tested with 6.11.3 and it works quite well. Also, a nice bonus with 6.11.3 is that your speakers will start working (as long as your device is in the quirk table of `patch_realtek.c`)  :wink: :sunglasses: 

### fan speed

Regarding the fan speed, hrmm that is a bummer... If you want, can you check the `_FST` method and see if it works on your device or not?  Try to run these commands and post back the results if you like:

```
sudo modprobe acpi_call

# get result of FAN1 _FST
echo '\_SB.PC00.LPCB.H_EC.FAN1._FST' | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call

# get result of FAN2 _FST
echo '\_SB.PC00.LPCB.H_EC.FAN2._FST' | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call
```

On my device (GB2 Pro) the `_FST` method has a bug and I get an ACPI exception when I try to run them, so I sort of reverse-engineered what I saw in the DSDT to try and figure out what it was "supposed" to do, and then implemented that in this Linux driver, instead.

It would be way easier if your device could just directly use the standard `_FST` methods without having to hack together a solution... though I do see similar "fan speed tables" in your DSDT, but it looks a bit more complicated (there are 3 different tables instead of just 1, and it looks like it looks at each different table based on different conditions... so a bit more work would need to be done to figure out exactly what is happening there....)

joshuagrisham, 2024-10-17:
Hi thank you!
Regarding the hotkey support, well, it was going to be a little more complicated than that, and I knew that the maintainers would want me to do what I did anyway (just use the method and remove anything else), so I did not think more on it.
But for now I added some code to hopefully solve this for you ... it will be removed later when the driver is mainlined, just FYI :)

Please try it and let me know if you have any issues!

Regarding your logs, it looks promising, but I would like to see if it works now to set and get all of the different flags, or if you are still getting failures?

Regarding the fan speeds, good news, it looks like your methods are working... I can try to look into this more later :)

Regarding fingerprint driver, if your device using a MoC scanner from EGIS then I made a driver for that also :) https://gitlab.freedesktop.org/libfprint/libfprint/-/blob/master/libfprint/drivers/egismoc/egismoc.c

Please create an issue at libfprint for support for your device if it is the same kind of model but not working for you :sunglasses: 
It can be that there needs to be more development on the driver...for example I have been in correspondence with someone who wanted to implement the 1c7a:0588 but there is a Windows Hello-based "nonce" generated which sort of "signs" the IDs that are stored in the reader, and this particular device rejects the IDs if you do not follow this requirement, and we have not figured out how to make it work yet.... that we can take in libfprint, though :innocent:

joshuagrisham, 2024-10-18:
Regarding your fan speed...

> Of course, I got:
> 
> ```
> marcus@laptop:~$ echo '\_SB.PC00.LPCB.H_EC.FAN1._FST' | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call \_SB.PC00.LPCB.H_EC.FAN1._FST
> [0x0, 0x49, 0xce4]
> 
> marcus@laptop:~$ echo '\_SB.PC00.LPCB.H_EC.FAN2._FST' | sudo tee /proc/acpi/call && sudo cat /proc/acpi/call \_SB.PC00.LPCB.H_EC.FAN2._FST
> [0x0, 0x50, 0xce4]
> ```

On your device, it seems that Samsung got the _FST method right but they did not implement any of the other methods in the ACPI 4.0 specification (https://uefi.org/specs/ACPI/6.5/11_Thermal_Management.html#fan-device)

In your examples, according to your device's _FST method, both of your fans were operating at 3300 RPM at the time (0xce4 = 3300)

For everything to show up "automatically" from the ACPI Fan Device, all 4 methods have to be present (see [drivers/acpi/fan_core.c#L206-L212](https://github.com/torvalds/linux/blob/4d939780b70592e0f4bc6c397e52e518f8fb7916/drivers/acpi/fan_core.c#L206-L212))

I am not sure how it is "typically" done in the kernel when the _FST method is working correctly but not supported out-of-the-box by ACPI Fan support... my assumption is that this driver can just add them to its own hwmon support but I will need to redo the logic a bit to maybe do something like this:

1. get all fans for the device
2. for each one, see if they are somehow supported out-of-the-box already (e.g. all 4 methods are present or similar)
3. if not, test their _FST method
4. if _FST method is good, then add fan_speed_rpm attribute and add hwmon fan channel for it, plus ensure that the "get_speed" method will execute _FST and fetch the speed from the method
5. if _FST method is not good, then check for existence of FANS and FANT and then set those up instead (similar to point 4 above)

I will create a separate issue just to track this specifically

joshuagrisham, 2024-10-18:
Regarding the rest @seaasses I made some tweaks today where I realized there were some very small risk for some null pointer references depending on parameters and potential failures during init etc so hopefully that should be in a better place now as well.
Would you mind to test how it works with the latest on this branch and see if writing / reading all of the different attributes is working or not for your device?  If it looks good then I will merge this branch in and then can keep looking into the fan speed issue a bit later since it will be a bit more work...

joshuagrisham, 2024-10-18:
@seaasses I created a PR and merged in the latest of that old branch into main now. When I did this, the issue also got closed... but if you still have issues specifically with the attributes (e.g. start_on_lid_open etc) then please write here and I will take a look again.

closed 2024-10-18 by joshuagrisham