joshuagrisham/samsung-galaxybook-extras #33: The keyboard is disabled when unloading the module.

jcarrano, 2024-10-15:
Unloading the module with `modprobe -r samsung-galaxybook` disabled the keyboard. Using a on-screen keyboard I was able to reload the module and then the physical keyboard is restored.

I attach the dmesg capture with the driver in debug mode:
[gxnokbd2.txt](https://github.com/user-attachments/files/17382300/gxnokbd2.txt)

joshuagrisham, 2024-10-15:
Oof, sorry about this @jcarrano ! Hrmm it is not super clear I think to me anyway but some things I wonder:
- Do the hotkeys for Keyboard Brightness and "Block Recording" work as supplied by the i8042_filter feature of samsung-galaxybook? (e.g. when you repeatedly press Fn+F9 does it toggle through all of the keyboard backlight levels?)
- If you disable i8042_filter when you load the module (e.g. set parameter `i8042_filter=off`) do you still have this problem or does it work to load and unload the module and your keyboard device is working ok?
- From the debug log it looks like this was an older version of the driver, but from your comment in #31 I am guessing it is the same problem on the latest version of the driver?

joshuagrisham, 2024-10-15:
Side note: I saw in the log you are running 6.11.3 .. do your speakers work? :innocent: :sunglasses:

jcarrano, 2024-10-15:
I tried the version with and without the changes from #31 and it has the same problem.

Then I tried loading (and unloading) with the following parameters and they all display the same bug:

```
sudo modprobe samsung_galaxybook debug=1 i8042_filter=0
sudo modprobe samsung_galaxybook debug=1 i8042_filter=0 acpi_hotkeys=0
sudo modprobe samsung_galaxybook debug=1 i8042_filter=0 acpi_hotkeys=0 wmi_hotkeys=0
```

The keyboard backlight works when the filter is enabled.

Re audio: no, it is still broken in 6.11.3.

joshuagrisham, 2024-10-15:
Yes, please try to check an older version and see if there is a difference, and let me know if it works or not there! It seems very strange to me especially if you get the same behavior without using the hotkey features... ?