joshuagrisham/samsung-galaxybook-extras #37: Failure to compile on Kernel 6.12.0-0.rc3

achapleau, 2024-10-20:
Running Fedora Rawhide with kernel 6.12.0-0.rc3.20241015giteca631b8fe80.32.fc42.x86_64 when I run command`make -C /lib/modules/`uname -r`/build M=$PWD` I get the following output:
`make: Entering directory '/usr/src/kernels/6.12.0-0.rc3.20241015giteca631b8fe80.32.fc42.x86_64'
  CC [M]  /home/andrew/samsung-galaxybook-extras/samsung-galaxybook.o
/home/andrew/samsung-galaxybook-extras/samsung-galaxybook.c: In function ‘galaxybook_wmi_notify’:
/home/andrew/samsung-galaxybook-extras/samsung-galaxybook.c:1578:18: error: implicit declaration of function ‘wmi_get_event_data’; did you mean ‘acpi_get_event_status’? [-Wimplicit-function-declaration]
 1578 |         status = wmi_get_event_data(value, &response);
      |                  ^~~~~~~~~~~~~~~~~~
      |                  acpi_get_event_status
/home/andrew/samsung-galaxybook-extras/samsung-galaxybook.c: In function ‘galaxybook_wmi_init’:
/home/andrew/samsung-galaxybook-extras/samsung-galaxybook.c:1599:70: error: passing argument 2 of ‘wmi_install_notify_handler’ from incompatible pointer type [-Wincompatible-pointer-types]
 1599 |         return wmi_install_notify_handler(GALAXYBOOK_WMI_EVENT_GUID, galaxybook_wmi_notify, NULL);
      |                                                                      ^~~~~~~~~~~~~~~~~~~~~
      |                                                                      |
      |                                                                      void (*)(u32,  void *) {aka void (*)(unsigned int,  void *)}
In file included from /home/andrew/samsung-galaxybook-extras/samsung-galaxybook.c:18:
./include/linux/acpi.h:404:60: note: expected ‘wmi_notify_handler’ {aka ‘void (*)(union acpi_object *, void *)’} but argument is of type ‘void (*)(u32,  void *)’ {aka ‘void (*)(unsigned int,  void *)’}
  404 |                                         wmi_notify_handler handler, void *data);
      |                                         ~~~~~~~~~~~~~~~~~~~^~~~~~~
make[2]: *** [scripts/Makefile.build:229: /home/andrew/samsung-galaxybook-extras/samsung-galaxybook.o] Error 1
make[1]: *** [/usr/src/kernels/6.12.0-0.rc3.20241015giteca631b8fe80.32.fc42.x86_64/Makefile:1982: /home/andrew/samsung-galaxybook-extras] Error 2
make: *** [Makefile:236: __sub-make] Error 2
make: Leaving directory '/usr/src/kernels/6.12.0-0.rc3.20241015giteca631b8fe80.32.fc42.x86_64'
`
Downgrading to stable kernel 6.11.4 solved the issue.

achapleau, 2024-10-20:
Just for fun, tried it on the debug disable kernel 6.12.rc2 out of the NODEBUG kernel repo and same error:
[image]

joshuagrisham, 2024-11-08:
Hi @achapleau just curious if you have tried this now with any later version of 6.12 ? I have found that it is a bit tough to know which versions of latest mainline/"next" will actually build or have a lot of regressions so it has been always a trial-and-error process for me personally to try and build latest mainline.

Also specifically for this WMI stuff I have removed all of the WMI stuff in the latest proposed version of this driver; could you maybe give it a try per the info in https://github.com/joshuagrisham/samsung-galaxybook-extras/pull/40 and see if this helps?

bortan1, 2024-11-16:
Fedora 41 here, also not compiling:
`$ make -C /lib/modules/`uname -r`/build M=$PWD
make: Entering directory '/usr/src/kernels/6.11.7-300.fc41.x86_64'
  CC [M]  /home/dan/Downloads/samsung-galaxybook-extras-main/samsung-galaxybook.o
  MODPOST /home/dan/Downloads/samsung-galaxybook-extras-main/Module.symvers
  CC [M]  /home/dan/Downloads/samsung-galaxybook-extras-main/samsung-galaxybook.mod.o
  LD [M]  /home/dan/Downloads/samsung-galaxybook-extras-main/samsung-galaxybook.ko
  BTF [M] /home/dan/Downloads/samsung-galaxybook-extras-main/samsung-galaxybook.ko
make: Leaving directory '/usr/src/kernels/6.11.7-300.fc41.x86_64'
`

It doesn't even tell me what's wrong. I have no idea how to troubleshoot this.

joshuagrisham, 2024-11-17:
@bortan1 is this with the branch https://github.com/joshuagrisham/samsung-galaxybook-extras/tree/no-quirks-test-before-enable or from main? If from main then I would ask if you can try using the branch instead and see if it works better?

bortan1, 2024-11-17:
> @bortan1 is this with the branch https://github.com/joshuagrisham/samsung-galaxybook-extras/tree/no-quirks-test-before-enable or from main? If from main then I would ask if you can try using the branch instead and see if it works better?

That was inside main. I am getting the same output using the no-quirks branch. If it helps, I also have this:

$ sudo make -C /lib/modules/`uname -r`/build M=$PWD modules_install
sudo depmod
make: Entering directory '/usr/src/kernels/6.11.7-300.fc41.x86_64'
  INSTALL /lib/modules/6.11.7-300.fc41.x86_64/updates/samsung-galaxybook.ko
  SIGN    /lib/modules/6.11.7-300.fc41.x86_64/updates/samsung-galaxybook.ko
At main.c:142:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: crypto/bio/bss_file.c:75
sign-file: ./certs/signing_key.pem
  DEPMOD  /lib/modules/6.11.7-300.fc41.x86_64
make: Leaving directory '/usr/src/kernels/6.11.7-300.fc41.x86_64'

joshuagrisham, 2024-11-17:
Hi again,
I think this issue is related to Fedora + Secure Boot and that you are required to sign the module
Here is some info I found regarding this but I am not sure if it is still the same for the latest versions of Fedora: https://docs.fedoraproject.org/en-US/fedora/f40/system-administrators-guide/kernel-module-driver-configuration/Working_with_Kernel_Modules/#sect-signing-kernel-modules-for-secure-boot

bortan1, 2024-11-17:
> Hi again, I think this issue is related to Fedora + Secure Boot and that you are required to sign the module Here is some info I found regarding this but I am not sure if it is still the same for [...]

This is secure boot disabled:

$ make -C /lib/modules/`uname -r`/build M=$PWD
make: Entering directory '/usr/src/kernels/6.11.7-300.fc41.x86_64'
  CC [M]  /home/dan/Downloads/samsung-galaxybook-extras-main/samsung-galaxybook.o
  MODPOST /home/dan/Downloads/samsung-galaxybook-extras-main/Module.symvers
  CC [M]  /home/dan/Downloads/samsung-galaxybook-extras-main/samsung-galaxybook.mod.o
  LD [M]  /home/dan/Downloads/samsung-galaxybook-extras-main/samsung-galaxybook.ko
  BTF [M] /home/dan/Downloads/samsung-galaxybook-extras-main/samsung-galaxybook.ko
make: Leaving directory '/usr/src/kernels/6.11.7-300.fc41.x86_64'
dan@fedora:~/Downloads/samsung-galaxybook-extras-main$ sudo make -C /lib/modules/`uname -r`/build M=$PWD modules_install
sudo depmod
Place your right index finger on the fingerprint reader
make: Entering directory '/usr/src/kernels/6.11.7-300.fc41.x86_64'
  INSTALL /lib/modules/6.11.7-300.fc41.x86_64/updates/samsung-galaxybook.ko
  SIGN    /lib/modules/6.11.7-300.fc41.x86_64/updates/samsung-galaxybook.ko
At main.c:142:
- SSL error:FFFFFFFF80000002:system library::No such file or directory: crypto/bio/bss_file.c:67
- SSL error:10000080:BIO routines::no such file: crypto/bio/bss_file.c:75
sign-file: ./certs/signing_key.pem
  DEPMOD  /lib/modules/6.11.7-300.fc41.x86_64
make: Leaving directory '/usr/src/kernels/6.11.7-300.fc41.x86_64'

achapleau, 2024-11-30:
Using branch refactor-use-existing-platform-device on kernel 6.12.1-200.fc41.x86_64 compiles and works as expected.

closed 2024-11-30 by achapleau