joshuagrisham/samsung-galaxybook-extras #41: SAM0427, losing KB backlight with  performance_mode activated

PremiumUsername, 2024-11-08:
Using the latest iteration of the driver I have some new problems. Firstly, if I don't actively use `performance_mode=off` I can't get KB backlight to work. 

```
[  528.147305] samsung_galaxybook: failed get supported performance modes with ACPI method CSXI; device did not respond with success code 0xaa
[  528.147308] samsung_galaxybook: failure initializing performance mode and platform profile
[  528.147344] Samsung Galaxy Book Extras SAM0427:00: probe with driver Samsung Galaxy Book Extras failed with error -5
```

The second problem is that I can't build the driver for an older kernel (6.6). The only reason why I use the older kernel is because for some reason ( this problme is there even with the module unloaded) I lose the touchpad after resume from suspend with anything newer. I don't know if this driver could be used to fix that. It seems to be an ACPI address problem according to the logs.  

I always feel a bit at unease and like I'm just complaining when making bug reports, so I'll just add that I'm very grateful for your all your efforts.

PremiumUsername, 2024-11-08:
@joshuagrisham Sorry, I missed that development. I tried the branch now and It doesn't seem like the KB backlight  work now. I haven't tried much else. I don't know if I managed to build it correctly because I don't see much information in dmesg even with `debug=on`. I got this in dmesg:
```
[  846.789771] samsung_galaxybook: unknown parameter 'acpi_hotkeys' ignored
[  846.789774] samsung_galaxybook: unknown parameter 'wmi_hotkeys' ignored
[  846.789835] samsung_galaxybook: loading driver
[  846.789944] samsung_galaxybook: driver successfully loaded 
```

joshuagrisham, 2024-11-09:
This is quite strange as it seems like there is actually nothing happening. Would you mind to share the exact command you are using to load the module and/or if you are installing it then what your kernel boot command line is ?

PremiumUsername, 2024-11-09:
I'm sorry @joshuagrisham , it seems I botched the last try. I'm trying to figure out the problem with the non functioning touchpad so I had some obscure acpi kernel parameters that I fear messed things up. With those removed, it does work! here is the dmesg:

```
[  599.415003] samsung_galaxybook: unknown parameter 'acpi_hotkeys' ignored
[  599.415006] samsung_galaxybook: unknown parameter 'wmi_hotkeys' ignored
[  599.415059] samsung_galaxybook: loading driver
[  599.415205] samsung_galaxybook: initializing ACPI device
[  599.415228] samsung_galaxybook: initializing ACPI power management features
[  599.415229] samsung_galaxybook: [DEBUG] enabling ACPI feature
[  599.415230] samsung_galaxybook: [DEBUG]     43 58 7a 00 00 bb aa 00 00 00 00 00 00 00 00 00
[  599.415231] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.415677] samsung_galaxybook: [DEBUG] response was:
[  599.415679] samsung_galaxybook: [DEBUG]     43 58 7a 00 aa dd cc 00 00 00 00 00 00 00 00 00
[  599.415679] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.415680] samsung_galaxybook: performance_mode is disabled
[  599.415681] samsung_galaxybook: initializing battery charge threshold control
[  599.415681] samsung_galaxybook: [DEBUG] getting battery charge_control_end_threshold
[  599.415682] samsung_galaxybook: [DEBUG]     43 58 7a 00 00 82 e9 91 00 00 00 00 00 00 00 00
[  599.415683] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.417113] samsung_galaxybook: [DEBUG] response was:
[  599.417114] samsung_galaxybook: [DEBUG]     43 58 7a 00 aa 00 00 55 00 00 00 00 00 00 00 00
[  599.417115] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.417115] samsung_galaxybook: [DEBUG] battery charge control is currently on; battery charge_control_end_threshold is 85
[  599.417121] ACPI: battery: new extension: Samsung Galaxy Book Battery Extension
[  599.417122] samsung_galaxybook: initializing platform device attributes
[  599.417123] samsung_galaxybook: [DEBUG] getting start_on_lid_open
[  599.417124] samsung_galaxybook: [DEBUG]     43 58 7a 00 00 82 a3 81 00 00 00 00 00 00 00 00
[  599.417124] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.418646] samsung_galaxybook: [DEBUG] response was:
[  599.418648] samsung_galaxybook: [DEBUG]     43 58 7a 00 aa 00 00 01 00 00 00 00 00 00 00 00
[  599.418649] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.418650] samsung_galaxybook: [DEBUG] start_on_lid_open is currently on (1)
[  599.418651] samsung_galaxybook: [DEBUG] getting usb_charge
[  599.418652] samsung_galaxybook: [DEBUG]     43 58 67 00 00 80 00 00 00 00 00 00 00 00 00 00
[  599.418653] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.419081] samsung_galaxybook: [DEBUG] response was:
[  599.419082] samsung_galaxybook: [DEBUG]     43 58 67 00 aa 00 00 00 00 00 00 00 00 00 00 00
[  599.419083] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.419084] samsung_galaxybook: [DEBUG] usb_charge is currently off (0)
[  599.419085] samsung_galaxybook: initializing ACPI allow_recording feature
[  599.419086] samsung_galaxybook: [DEBUG] enabling ACPI feature
[  599.419087] samsung_galaxybook: [DEBUG]     43 58 8a 00 00 bb aa 00 00 00 00 00 00 00 00 00
[  599.419088] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.419514] samsung_galaxybook: [DEBUG] response was:
[  599.419515] samsung_galaxybook: [DEBUG]     43 58 8a 00 aa dd cc 00 00 00 00 00 00 00 00 00
[  599.419516] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.419518] samsung_galaxybook: [DEBUG] getting allow_recording
[  599.419519] samsung_galaxybook: [DEBUG]     43 58 8a 00 00 81 00 00 00 00 00 00 00 00 00 00
[  599.419520] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.419973] samsung_galaxybook: [DEBUG] response was:
[  599.419974] samsung_galaxybook: [DEBUG]     43 58 8a 00 aa 01 00 00 00 00 00 00 00 00 00 00
[  599.419976] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.419977] samsung_galaxybook: [DEBUG] allow_recording is currently on (1)
[  599.419978] samsung_galaxybook: initializing platform device
[  599.420031] samsung_galaxybook: initializing kbd_backlight
[  599.420032] samsung_galaxybook: [DEBUG] enabling ACPI feature
[  599.420033] samsung_galaxybook: [DEBUG]     43 58 78 00 00 bb aa 00 00 00 00 00 00 00 00 00
[  599.420034] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.420458] samsung_galaxybook: [DEBUG] response was:
[  599.420460] samsung_galaxybook: [DEBUG]     43 58 78 00 aa dd cc 00 00 00 00 00 00 00 00 00
[  599.420461] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.420462] samsung_galaxybook: [DEBUG] getting kbd_backlight brightness
[  599.420463] samsung_galaxybook: [DEBUG]     43 58 78 00 00 81 00 00 00 00 00 00 00 00 00 00
[  599.420465] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.420896] samsung_galaxybook: [DEBUG] response was:
[  599.420898] samsung_galaxybook: [DEBUG]     43 58 78 00 aa 00 04 00 00 00 00 00 00 00 00 00
[  599.420899] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.420900] samsung_galaxybook: [DEBUG] current kbd_backlight brightness is 0
[  599.420901] samsung_galaxybook: registering LED class samsung-galaxybook::kbd_backlight
[  599.420929] samsung_galaxybook: [DEBUG] getting kbd_backlight brightness
[  599.420931] samsung_galaxybook: [DEBUG]     43 58 78 00 00 81 00 00 00 00 00 00 00 00 00 00
[  599.420932] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.421357] samsung_galaxybook: [DEBUG] response was:
[... cut ...]
[  599.421795] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.423262] samsung_galaxybook: [DEBUG] response was:
[  599.423263] samsung_galaxybook: [DEBUG]     43 58 86 00 aa 02 00 00 00 00 00 00 00 00 00 00
[  599.423265] samsung_galaxybook: [DEBUG]     00 00 00 00 00
[  599.423266] samsung_galaxybook: initializing input device
[  599.423310] input: Samsung Galaxy Book extra buttons as /devices/platform/samsung-galaxybook/input/input18
[  599.423386] samsung_galaxybook: driver successfully loaded
```

joshuagrisham, 2024-11-09:
Thank you @PremiumUsername ! I wonder if you remove all of the other parameters except for debug=on does it still work for you? My intention was that I would want it to work "out of the box" for even you and that you did not need to enable or disable anything specifically (the driver will try to detect what works and what does not work and only enable those things that appear to be supported)

PremiumUsername, 2024-11-09:
Not sure what you mean @joshuagrisham.  I loaded the module with just  'sudo modprobe samsung-galaxybook debug=true'. I don't know where the lines about unknown parameters come from.  I didn't explicitly deactivate anything, and everything that worked before seems to be working still so I'd say it's working as I think you intended it to. 
I even have two new keys working, fn+F11 and fn+F12. They don't do anything but they can be seen with evtest.

joshuagrisham, 2024-11-17:
Hi sorry it was a bit unclear @PremiumUsername ; what I meant was that it seems from your log that some parameters are being applied so my guess is that you have them somewhere in your kernel command line or something (e.g. if you are using GRUB then maybe they are set in `/etc/default/grub` or similar)

But no worries, you can just try to remove the module and then load it from the compiled module while turning everything on and it should be the same result.. maybe something like this?

```sh
# remove module if already loaded
sudo rmmod samsung-galaxybook
# insert module from locally built file and enable all params
sudo insmod samsung-galaxybook.ko kbd_backlight=on battery_threshold=on performance_mode=on allow_recording=on fan_speed=on i8042_filter=on debug=on
```

The "issue" or open question I still have is that your existing parameters are disabling some of the things on purpose but I wanted to just see if the logic can detect and disable them automatically instead :)