joshuagrisham/samsung-galaxybook-extras #42: Minor Issue - Power Mode Change Not Reflected in GNOME Control Center When Changed via Keyboard

Santigf12, 2024-11-08:
Description: When changing the power mode directly using the keyboard on my Samsung Galaxy Book, the GNOME Control Center does not reflect the change. However, if I adjust the power mode directly within the GNOME Control Center, the change is applied correctly and functions as expected.

This appears to be a minor state synchronization issue between the hardware (keyboard) settings change and the software (GNOME Control Center).

Steps to Reproduce:

    Change the power mode using the keyboard.
    Open GNOME Control Center and observe the power mode setting.
    Change the power mode within GNOME Control Center and note that it functions as expected.

Expected Behavior: The GNOME Control Center should update its power mode state when changes are made via the keyboard.

Minor issue and does not impact functionality significantly, but it may be confusing if you want to the status in the control centre

Samsung Galaxy Extras working and compiled in Fedora 41 using secure boot

System:
  OS: Fedora Linux 41 (Workstation Edition)
  Kernel: 6.11.6-300.fc41.x86_64
  Device: Samsung Galaxy Book 4 Ultra 2024 (Model 960XGL)

joshuagrisham, 2024-11-08:
Hi @Santigf12  thanks for reaching out.

### Regarding the power profile issue:

I am curious is this where you have booted and the module was present or do you load the module using modprobe or insmod after you boot?  There are some "quirks" with UPower's power-profile-daemon which make so this kind of stuff only works if everything is in place before the daemon starts and even that you do not remove and re-add the module after startup.

I would also ask if maybe you can test with the branch of this driver mentioned in the PR here: https://github.com/joshuagrisham/samsung-galaxybook-extras/pull/40 as maybe it is slightly improved for this compared to what is on the main branch as of right now?

### Regarding speakers:

Can you try something like this:

```
sudo tee /etc/modprobe.d/audio-fix.conf <<< 'options snd-hda-intel model=alc298-samsung-amp-v2-4-amps'
```
And then reboot or even power off/power on?
(keeping in mind this will only work with 6.11.3 or higher)

If this works then we will need to add your device to the quirk table in `patch_realtek.c` in the kernel and then if/when your device is added you would be able to remove this new file `/etc/modprobe.d/audio-fix.conf` that was created from the above...

### Regarding fingerprint sensor:

If your device has a MoC sensor from Egis / "LightTuning" then there is a libfprint driver I have developed here:

https://gitlab.freedesktop.org/libfprint/libfprint/-/blob/master/libfprint/drivers/egismoc/egismoc.c

But my guess is that your device vendor id + product id needs to be added to the list (`egismoc_id_table`) however I think the best course of action is to see if there is an existing issue for your vid+pid and if not then please create a new one there :sunglasses: 
https://gitlab.freedesktop.org/libfprint/libfprint/-/issues

One thing to note, however, if your device has a newer sensor like the 1c7a:0588 or 1c7a:05a5 then we have some issues with lack of support for SDCP for these devices.. I am trying to figure it out but running into problems and hoping for help from the vendor (whom I have sent some mails to!)...

### Regarding battery not detected after reboot:

No clue about this unfortunately, and I have not really heard much about this .. depending on which subsystem the problem is happening in, it might make sense to file a Kernel bug report? Do you have any failure or warning messages when you see this happen?

Santigf12, 2024-11-08:
Hello, 

Thanks for responding

##Power profile issue

jcarrano, 2024-11-08:
TLDR: disable the keyboard handling on the driver and handle it in userspace

Detailed explanation:

Detecting the power mode was changed is not straightforward. That's an issue with the gnome control center, and it is not easy to solve as it would require the gnome software to implement a mechanism for monitoring the change, and that can be different for each platform.

Monitoring the sysfs file (e.g. with inotify)  does not work. What does work (and you can try it) is to monitior udev events (`sudo udevadm monitor`) but you would need the software to interpret these events as a power mode change.

What should work is to disable the automated handing of the power mode key in the driver and to map the Fn-F11 shortcut to gnome (IDK how to do this, I use XFCE) and then let GNOME invoke the power-profile-daemon or similar.