joshuagrisham/samsung-galaxybook-extras #46: Issue compiling on Linux 6.12.1

observantdragon, 2024-11-29:
Hi,

I have been using your driver for the past few months on a book 3 360 13.3 inch. I have had no issues with it on prior versions of the kernel before 6.12. Since the upgrade to 6.12 the compile fails on archlinux.

Additionally since I have not seen much feedback for this model, I just wanted to mention that it works for me. However on this model I have the opposite issue that everyone seems to be reporting regarding audio on samsung notebooks. My internal speakers work fine with no workarounds, however the headphone jack does not work properly. The headphones are detected and auto mute the internal speakers but there is "no" audio from the headphones. Increasing the volume to the maximum though does have some audio play extremely faint and distorted through some headphones. I had tried the script that was mentioned elsewhere as a workaround but running that script just hangs and does not solve the issue.

Other than that the fingerprint reader does not work (Canvas Bio CB200), and without the driver, the keyboard backlight turns to max brightness if it was set at any level in windows (off otherwise). HDR works fine as well on gamescope and kwin.

Here is the dkms log for the compile issue:
`DKMS make.log for samsung-galaxybook-extras for kernel 6.12.1-arch1-1 (x86_64)
Mon 25 Nov 22:06:32 EST 2024
make: Entering directory '/usr/lib/modules/6.12.1-arch1-1/build'
  CC [M]  /var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.o
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c: In function ‘get_acpi_device_description’:
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:258:26: error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’
  258 |         if (acpi_dev->pnp.str_obj != NULL && acpi_dev->pnp.str_obj->buffer.length > 0) {
      |                          ^
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:258:59: error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’
  258 |         if (acpi_dev->pnp.str_obj != NULL && acpi_dev->pnp.str_obj->buffer.length > 0) {
      |                                                           ^
In file included from ./include/linux/percpu.h:5,
                 from ./arch/x86/include/asm/msr.h:15,
                 from ./arch/x86/include/asm/tsc.h:10,
                 from ./arch/x86/include/asm/timex.h:6,
                 from ./include/linux/timex.h:67,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:13,
                 from /var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:15:
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:259:65: error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’
  259 |                 char *buf = kzalloc(sizeof(*buf) * acpi_dev->pnp.str_obj->buffer.length, GFP_KERNEL);
      |                                                                 ^
./include/linux/alloc_tag.h:206:16: note: in definition of macro ‘alloc_hooks_tag’
  206 |         typeof(_do_alloc) _res = _do_alloc;                             \
      |                ^~~~~~~~~
./include/linux/slab.h:1016:49: note: in expansion of macro ‘alloc_hooks’
 1016 | #define kzalloc(...)                            alloc_hooks(kzalloc_noprof(__VA_ARGS__))
      |                                                 ^~~~~~~~~~~
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:259:29: note: in expansion of macro ‘kzalloc’
  259 |                 char *buf = kzalloc(sizeof(*buf) * acpi_dev->pnp.str_obj->buffer.length, GFP_KERNEL);
      |                             ^~~~~~~
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:259:65: error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’
  259 |                 char *buf = kzalloc(sizeof(*buf) * acpi_dev->pnp.str_obj->buffer.length, GFP_KERNEL);
      |                                                                 ^
./include/linux/alloc_tag.h:206:34: note: in definition of macro ‘alloc_hooks_tag’
  206 |         typeof(_do_alloc) _res = _do_alloc;                             \
      |                                  ^~~~~~~~~
./include/linux/slab.h:1016:49: note: in expansion of macro ‘alloc_hooks’
 1016 | #define kzalloc(...)                            alloc_hooks(kzalloc_noprof(__VA_ARGS__))
      |                                                 ^~~~~~~~~~~
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:259:29: note: in expansion of macro ‘kzalloc’
  259 |                 char *buf = kzalloc(sizeof(*buf) * acpi_dev->pnp.str_obj->buffer.length, GFP_KERNEL);
      |                             ^~~~~~~
./include/linux/alloc_tag.h:212:1: error: initialization of ‘char *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
  212 | ({                                                                      \
      | ^
./include/linux/slab.h:1016:49: note: in expansion of macro ‘alloc_hooks’
 1016 | #define kzalloc(...)                            alloc_hooks(kzalloc_noprof(__VA_ARGS__))
      |                                                 ^~~~~~~~~~~
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:259:29: note: in expansion of macro ‘kzalloc’
  259 |                 char *buf = kzalloc(sizeof(*buf) * acpi_dev->pnp.str_obj->buffer.length, GFP_KERNEL);
      |                             ^~~~~~~
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:261:49: error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’
  261 |                         (wchar_t *)acpi_dev->pnp.str_obj->buffer.pointer,
      |                                                 ^
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:262:38: error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’
  262 |                         acpi_dev->pnp.str_obj->buffer.length,
      |                                      ^
make[2]: *** [scripts/Makefile.build:229: /var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.o] Error 1
make[1]: *** [/usr/lib/modules/6.12.1-arch1-1/build/Makefile:1936: /var/lib/dkms/samsung-galaxybook/extras/build] Error 2
make: *** [Makefile:224: __sub-make] Error 2
make: Leaving directory '/usr/lib/modules/6.12.1-arch1-1/build'
`

joshuagrisham, 2024-11-29:
Hi @observantdragon  thank you for reporting this! 

This method I created to fetch the fan "description" is not super clean IMO and I would rather that there would be a standard ACPI method like `acpi_get_description` or similar (like there already exists `acpi_get_name` for example) but that I will have to take with the ACPI tree maintainers :laughing: 

What I did in the meantime was basically copied their `description_show` method that they use with sysfs here, but there was a change in how this stuff works in 6.12 so it no longer worked.

In the end it is not super critical but I just thought it would be nice to take the Fan device description from the ACPI as in some newer models Samsung has actually used this to give nicer names (e.g. "CPU Fan" and "GPU Fan" etc) so to me it was best case to use these descriptions as the fan labels for when you monitor the speeds via hwmon.

Anyway I make a correction to follow how the maintainers changed this in 6.12 but I only put it in the branch which I am prepping for a mainline patch. Would you mind to take a look at the information here and test with the branch that is mentioned there? https://github.com/joshuagrisham/samsung-galaxybook-extras/pull/44

Do note that some of the sysfs interface will change and you can also take a look at using udev vs not in case that is important to you :) All of that info is written in the PR..

Incidentally I have still not found any branch or RC of 6.12 I was able to actually install as of yet, every one I have tried I hit this same bug: https://bugs.launchpad.net/ubuntu/+source/backport-iwlwifi-dkms/+bug/2083994  /  https://bugs.launchpad.net/ubuntu/+source/backport-iwlwifi-dkms/+bug/2085453

But I did test this change now against 6.5.x and 6.11.3 and it seems to be working for me in both, plus it matches what the ACPI maintainers did now in 6.12 so I feel pretty good that this part should be working better now?

joshuagrisham, 2024-11-29:
Regarding your headphone jack issue, this sounds like the same issue which exists on other models and there is potentially already a fix for it (just would need to add your device ID to the quirk table).

Can you try something like this:

```
sudo tee /etc/modprobe.d/audio-fix.conf <<< 'options snd-hda-intel model=alc298-samsung-amp'
```

And then reboot or even power off/power on?

If this works well let me know if you want/need help that I can submit a kernel patch for you (otherwise you are of course welcome to submit one yourself!!).

And then if/when your device is added to the quirk table there, you would be able to remove the file `/etc/modprobe.d/audio-fix.conf` once you upgraded to that version of the kernel.

observantdragon, 2024-11-29:
> But I did test this change now against 6.5.x and 6.11.3 and it seems to be working for me in both, plus it matches what the ACPI maintainers did now in 6.12 so I feel pretty good that this part should be working better now?

Yup, I can confirm that it now works again on 6.12.1 on arch for me. Indeed, agreed regarding the fan names. I have tested that the battery charge limit and keyboard functions work.

observantdragon, 2024-11-30:
I've now found out that I can fix the headphones by running.
`echo 0x1a 0x707 0xc5 > /sys/class/sound/hwC0D0/init_verbs`

So presumably adding my device to `ALC298_FIXUP_SAMSUNG_HEADPHONE_VERY_QUIET` should work!

joshuagrisham, 2024-11-30:
Thanks @observantdragon and good that you could confirm this :) I actually was sort of thinking that is probably what it would be based on what you described, but there is not an existing device / "model name" in patch_realtek that gives only that (only the ones that give the headphone fix AND some speaker amp fixes).. as a guess/hope I asked you to try the one that gave both (which seems like it did not work?) so yes the right course of action would be to add your device / subdevice ID to the quirk table with only ALC298_FIXUP_SAMSUNG_HEADPHONE_VERY_QUIET I think (and/or a new name under ALC256 that is an alias / points to the same function).

If you want then I can help prepare a patch for this otherwise you can feel free to take it as a challenge for yourself , just let me know if you want me to do anything :)

observantdragon, 2024-11-30:
> (which seems like it did not work?) so yes the right course of action would be to add your device / subdevice ID to the quirk table with only ALC298_FIXUP_SAMSUNG_HEADPHONE_VERY_QUIET I think (and/or a new name under ALC256 that is an alias / points to the same function).

It did not work via setting it via `modprobe.d/` and nor did any of the other samsung related model's. I could try again via a patched kernel, but I feel like that shouldn't matter, alsa-info.sh did show that they were being set. And since the internal speakers in this laptop work out of the box unlike the others I guess they are not necessary.

>If you want then I can help prepare a patch for this otherwise you can feel free to take it as a challenge for yourself , just let me know if you want me to do anything :)

Already got the draft patch ready! Just wanted to investigate the headphone mic quality first and compare some things with Windows. Although I realise now that this might just be my headset.

On another note, I don't use these but the lock trackpad and disable mic and camera keys do not seem to work for me even with the layout file. Not sure if they are supposed to. My device is set to need the FN + modifier for those.

joshuagrisham, 2024-12-01:
@observantdragon I think in this case it might be a good opportunity to use `RtHDDump.exe` in Windows and then you can see exactly what this is set to in different conditions (e.g. take a dump with headphones not plugged in and another one with headphones plugged in)

Disable any kind of script/patch in Linux and then power off totally, then power back on, then you can check the value of all of the same pins and coeffs in Linux and compare what is in the value dumps from Windows vs the values you see in Linux.

Sort of like this (assuming right path for your device/codec):

```sh
echo 1 | sudo tee /sys/module/snd_hda_codec/parameters/dump_coef
cat /proc/asound/card0/codec#0
```

Hopefully it will be very clear what the value is for this coeff in the Windows dumps and very clear that it is NOT that value in Linux; then just try making the value match (e.g. using `hda-verb`) and hopefully it works as you want :sunglasses: 

From there it is to figure out how to weave that into patch_realtek.c in the right way so that it matches your device and applies the right fix :)

observantdragon, 2024-12-01:
@joshuagrisham That is a good idea, I'll look into capturing it from Windows and comparing.

On another note, I was wrong about the block recording key not working, it does work, I just assumed it would disconnect the devices but it just makes the output black (for the webcam).

Additionally I also noticed the following in dmesg
`[  144.847992] samsung_galaxybook: loading out-of-tree module taints kernel.
[  144.848009] samsung_galaxybook: module verification failed: signature and/or required key missing - tainting kernel
[  144.850023] samsung_galaxybook: found matched device SAM0429:00; loading driver
[  144.853490] samsung_galaxybook: will support platform profile 'performance' with performance mode value 0x15
[  144.853494] samsung_galaxybook: will support platform profile 'low-power' with performance mode value 0xb
[  144.853496] samsung_galaxybook: will support platform profile 'quiet' with performance mode value 0xa
[  144.853498] samsung_galaxybook: will support platform profile 'balanced' with performance mode value 0x2
[  144.860146] ACPI: battery: new extension: Samsung Galaxy Book Battery Extension
[  144.873355] samsung_galaxybook: registering LED class using default name of samsung-galaxybook::kbd_backlight
[  144.874725] samsung_galaxybook: found fan device PNP0C0B:00
[  144.876046] ACPI Error: Needed [Integer/String/Buffer], found [Reference] 00000000e1aaeb2e (20240827/exresop-388)
[  144.876064] ACPI Error: AE_AML_OPERAND_TYPE, While resolving operands for [Add] (20240827/dswexec-433)
[  144.876076] ACPI Error: Aborting method \_SB.PC00.LPCB.FAN0._FST due to previous error (AE_AML_OPERAND_TYPE) (20240827/psparse-529)
[  144.876090] samsung_galaxybook: Get fan state failed
[  144.877232] samsung_galaxybook: initialized fan speed reporting for device PNP0C0B:00 (FAN0) with the following levels:
[  144.877240] samsung_galaxybook:   PNP0C0B:00 (FAN0) fan speed level 0 = 0
[  144.877243] samsung_galaxybook:   PNP0C0B:00 (FAN0) fan speed level 1 = 3510
[  144.877246] samsung_galaxybook:   PNP0C0B:00 (FAN0) fan speed level 2 = 3820
[  144.877249] samsung_galaxybook:   PNP0C0B:00 (FAN0) fan speed level 3 = 4430
[  144.877250] samsung_galaxybook:   PNP0C0B:00 (FAN0) fan speed level 4 = 4740
[  144.877253] samsung_galaxybook:   PNP0C0B:00 (FAN0) fan speed level 5 = 5740
[  144.881476] input: Samsung Galaxy Book Extra Buttons as /devices/platform/SAM0429:00/input/input29
[  144.881574] samsung_galaxybook: driver successfully loaded
`
Are the ACPI errors to be expected? My laptop has only one fan. `sensors` (lm_sensors) does report the fan speeds, I'm a little surprised it starts at 3510rpm as when it reports that the fan is fairly quiet (but spinning) and that seems a little high.

Another quirk I noticed is if I moprobe -r `sudo modprobe -r samsung_galaxybook` , my keyboard stops working, but maybe that is to be expected.

observantdragon, 2024-12-01:
@joshuagrisham Regarding the coeffs on Windows, I looked into it and there are only a few differences between windows and linux, however changing them in Linux does not fix any issue. 

There are no differences on either windows and linux, whether the headset is plugged in or not or whether sound is playing or not.

I went through what I found [here](https://asus-linux.org/blog/sound-2021-01-11/) and the `0x1a 0x707 0xc5` fix I used earlier seems to be correct. (Equivalent to the `ALC298_FIXUP_SAMSUNG_HEADPHONE_VERY_QUIET` fix in the kernel.

Applying this seems to change to pin-ctls of the Amp from
`Pin-ctls: 0x00: VREF_HIZ` to `Pin-ctls: 0x05: VREF_100`

```
Node 0x1a [Pin Complex] wcaps 0x40048b: Stereo Amp-In
  Amp-In caps: ofs=0x00, nsteps=0x03, stepsize=0x27, mute=0
  Amp-In vals:  [0x00 0x00]
  Pincap 0x00003724: IN Detect
    Vref caps: HIZ 50 GRD 80 100
  Pin Default 0x411111f0: [N/A] Speaker at Ext Rear
    Conn = 1/8, Color = Black
    DefAssociation = 0xf, Sequence = 0x0
    Misc = NO_PRESENCE
  Pin-ctls: 0x05: VREF_100
  Unsolicited: tag=00, enabled=0
  Power states:  D0 D1 D2 D3 EPSS
  Power: setting=D0, actual=D0
```

Also the headphone mic in jack always worked even without a fix, this issue was with my headset and also present on Windows, a different headset works! So I'm going to go ahead with sending a patch.

joshuagrisham, 2024-12-02:
Hi again!
Regarding if you should see the ACPI errors .. yes, unfortunately, there is a bug in the ACPI for some of these devices where when you try to execute the `_FST` (fan status; see: https://uefi.org/specs/ACPI/6.5/11_Thermal_Management.html#fst-fan-status) method I think they are returning a reference to the structure instead of returning the structure itself. Incidentally, the "default" value when the fan speed level is 0 (the fan is not spinning at all) then it returns all 0s and works... anyway, there is more info in the readme here: https://github.com/joshuagrisham/samsung-galaxybook-extras?tab=readme-ov-file#fan-speed

The value you are seeing is basically what it "looks like" their code would return, if it did not have this exception. No idea if this is a true-to-life RPM value, just reporting what it looks like it should report in case the bug did not exist :)
 
Regarding your keyboard stopping to work when you remove the module -- I have heard a similar report to this and was never able to look into it more or get more details. If you have any other debug information you might be able to provide about this, it would be appreciated!  It might be that if you disable the i8042_filter then you will not have this problem, but then some of the hotkey actions will not work automatically (e.g. you will need to rig up your own actions based on the hotkeys, including adjusting kbd backlight level etc).

joshuagrisham, 2024-12-02:
> Also the headphone mic in jack always worked even without a fix, this issue was with my headset and also present on Windows, a different headset works! So I'm going to go ahead with sending a patch.

I did not follow 100% of the thread here but it sounds like you have a good grasp on what is going on here and what needs to be fixed. Great, go forth and conquer, I say!

observantdragon, 2024-12-02:
> Regarding if you should see the ACPI errors .. yes, unfortunately, there is a bug in the ACPI

That's what I would have guessed since a lot of laptops have ACPI  and UEFI quirks.

@joshuagrisham 
> If you have any other debug information you might be able to provide about this, it would be appreciated! It might be that if you disable the i8042_filter then you will not have this problem, but [...]

I haven´t looked much into it but I don't have this problem with `rmmod`, only `modprobe -r` . When the keyboard stops working `lsmod` shows that `i8042` is also removed. So I assume it is just because it is a dependency of `samsung_galaxybook`, and when that is removed it assumes `i8042` is no longer used. I have the same behaviour with `i8042_filter=off`.

```
Module                  Size  Used by
samsung_galaxybook     65536  0
platform_profile       12288  1 samsung_galaxybook
sparse_keymap          12288  2 samsung_galaxybook,intel_hid
i8042                  57344  1 samsung_galaxybook
serio                  28672  4 serio_raw,atkbd,i8042
```

joshuagrisham, 2024-12-05:
Hi @observantdragon 
Regarding using `modprobe -r`, which removes all dependencies including ones that might actually be needed: I asked this question to the PD x86 maintainers and they basically said: "don't do that" :laughing: 

The suggestion is to **only** use `rmmod` with this module.

Note also that using `modprobe -r` can also remove some ACPI battery support (since there is a battery extension added by the driver, as well), depending on what is compiled as modules vs not and what other devices exist/modules are loaded on your device.

Another option is if you configure your kernel to include i8042 instead of building it as a module (`y` instead of `m` in the config, basically), and then there is no module to remove.

observantdragon, 2024-12-06:
> Hi @observantdragon Regarding using `modprobe -r`, which removes all dependencies including ones that might actually be needed: I asked this question to the PD x86 maintainers and they basically said: "don't do that" 😆
> 
> The suggestion is to **only** use `rmmod` with this module.
> 
> Note also that using `modprobe -r` can also remove some ACPI battery support (since there is a battery extension added by the driver, as well), depending on what is compiled as modules vs not and what other devices exist/modules are loaded on your device.
> 
> Another option is if you configure your kernel to include i8042 instead of building it as a module (`y` instead of `m` in the config, basically), and then there is no module to remove.

Indeed, and yeah I saw their response on the mailing list. Hopefully there won't be a need to rmod, it is just something I noticed.

joshuagrisham, 2024-12-07:
I am glad you reported it as there have been other reports of exactly this same behavior and I was not able to connect the dots until this additional information from you, so thank you for that! :beers:

closed 2024-12-07 by observantdragon