joshuagrisham/samsung-galaxybook-extras #48: Having Problems with process

TopBlok, 2024-12-06:
Hello. I  am a new linux user who just purchased a galaxy book 2 360.  I was able to follow instructions on a separate page of yours to add the line to grub to get display brightness working.   I am having and issue getting started here.  

This is the error I get when trying to start the process. 

root@MannyGalaxyBook2:~# make -C /lib/modules/`uname -r`/build M=$PWD
make: Entering directory '/usr/src/linux-headers-6.8.0-49-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
scripts/Makefile.build:41: /root/Makefile: No such file or directory
make[2]: *** No rule to make target '/root/Makefile'.  Stop.
make[1]: *** [/usr/src/linux-headers-6.8.0-49-generic/Makefile:1925: /root] Error 2
make: *** [Makefile:240: __sub-make] Error 2
make: Leaving directory '/usr/src/linux-headers-6.8.0-49-generic'

I am a total Linux Noob.  Any help would be greatly appreciated.  My discord is samuellchimpson if you can connect with me.

TopBlok, 2024-12-07:
> Hi @TopBlok <https://github.com/TopBlok> !
>
> Hrmm well I could not readily find any "official" Linux Mint guide for
> this, but if you were running Ubuntu I would direct you to something like
> this: https://wiki.ubuntu.com/Kernel/BuildYourOwnKernel
>
> Essentially you need to have all of the dependencies you would "normally"
> use to build the entire Kernel, but instead of building the entire kernel,
> we are just building a single module from this repository.
>
> I did find a post here that goes a bit more in depth on the process and
> what is required:
> https://forums.linuxmint.com/viewtopic.php?p=2274066#p2274066
>
> The most important step there is step # 2 (installing all of the needed
> dependency packages) and then you should also grab whatever appropriate
> "linux-headers" and possibly "linux-source" package for your currently
> installed Kernel (you can probably install them using something like sudo
> apt install linux-headers linux-source )
>
> If you are using secure boot and/or your kernel requires signed modules
> then I also found this somewhere (again unfortunately I don't run Mint so
> not sure exactly what is available there in which version vs not..):
>
>
> [... 13 lines cut ...]
> sitting at directory ~ for the user root i.e. the root user's home
> directory).
>
> Hopefully this helps you get moving a bit more, and by the way, welcome to
> Linux! 😎

Thank you for the information.  I am not on mint for any other reason than
I originally liked the GUI.  I am going to install Ubunutu instead and see
if i can get this installed.  Will report back with any issues i have.  I
am less concerned with the version of Linux as I am with just getting some
of the OEM features of the hardware working.

TopBlok, 2024-12-08:
This is where I got to.   Not sure what to do next:

***@***.***:~$ dir
Desktop    Downloads  Pictures samsung-galaxybook-extras  Templates
Documents  Music      Public snap   Videos
***@***.***:~$ cd samsung-galaxybook-extras
***@***.***:~/samsung-galaxybook-extras$ sudo dkms build
samsung-galaxybook/extras
[sudo] password for manny-sidhu:
Module samsung-galaxybook/extras already built for kernel 6.8.0-49-generic
(x86_64), skip. You may override by specifying --force.
***@***.***:~/samsung-galaxybook-extras$ sudo dkms build
samsung-galaxybook/extras --force
Sign command: /usr/bin/kmodsign
Signing key: /var/lib/shim-signed/mok/MOK.priv
Public certificate (MOK): /var/lib/shim-signed/mok/MOK.der

Building module:
Cleaning build area...
make -j12 KERNELRELEASE=6.8.0-49-generic -C
/lib/modules/6.8.0-49-generic/build
M=/var/lib/dkms/samsung-galaxybook/extras/build...
Signing module
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.ko
Cleaning build area...
***@***.***:~/samsung-galaxybook-extras$ sudo dkms install
samsung-galaxybook/extras
Module samsung-galaxybook/extras already installed on kernel
6.8.0-49-generic (x86_64), skip. You may override by specifying --force.
***@***.***:~/samsung-galaxybook-extras$ sudo dkms install
samsung-galaxybook/extras --force
Module samsung-galaxybook-extras for kernel 6.8.0-49-generic (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

samsung-galaxybook.ko.zst:
 - Uninstallation
   - Deleting from: /lib/modules/6.8.0-49-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.
depmod....

samsung-galaxybook.ko.zst:
Running module version sanity check.
 - Original module
 - Installation
   - Installing to /lib/modules/6.8.0-49-generic/updates/dkms/
depmod...

TopBlok, 2024-12-08:
YAAAYYYYYY!!!! Went back to Linxu mint.  Followed the instructions.  Ran
into some problems with the signing.  Chat gpt walked me through it.  After
about 6-7 hours of time playing around with this i had to delete all secure
boot variables in my bios, not just disable it.   After doing that, it
loaded just fine.

Can confirm - back light for both screen and keyboard working now.   How do
I find out what performance mode I am in with fnc F11?   I can cycle
through them and notice changes in fan but not sure how to confirm what
setting it is in.  Only function keys that work are screen brightness, all
three sound, keyboard backlight, and performance mode.   Unable to lock
touch pad.

[  334.360042] No Arguments are initialized for method [_FST]

[  334.360044] ACPI Error: Aborting method \_SB.PC00.LPCB.FAN0._FST due to
previous error (AE_AML_OPERAND_TYPE) (20230628/psparse-529)
[  334.360052] samsung_galaxybook: Get fan state failed
[  334.361309] samsung_galaxybook: initialized fan speed reporting for
device PNP0C0B:00 (FAN0) with the following levels:
[  334.361314] samsung_galaxybook:   PNP0C0B:00 (FAN0) fan speed level 0 = 0
[  334.361316] samsung_galaxybook:   PNP0C0B:00 (FAN0) fan speed level 1 =
3510
[  334.361317] samsung_galaxybook:   PNP0C0B:00 (FAN0) fan speed level 2 =
3820
[  334.361318] samsung_galaxybook:   PNP0C0B:00 (FAN0) fan speed level 3 =
4430
[  334.361319] samsung_galaxybook:   PNP0C0B:00 (FAN0) fan speed level 4 =
4740
[  334.361320] samsung_galaxybook:   PNP0C0B:00 (FAN0) fan speed level 5 =
5740
[  334.364887] input: Samsung Galaxy Book extra buttons as
/devices/platform/samsung-galaxybook/input/input47
[  334.365103] samsung_galaxybook: driver successfully loaded
***@***.***:~/samsung-galaxybook-extras$

joshuagrisham, 2025-01-23:
Hi @TopBlok ,
There is information on how to check and set the profile in the README; here is a direct link: https://github.com/joshuagrisham/samsung-galaxybook-extras?tab=readme-ov-file#performance-mode

Great to hear you got it working 😎