joshuagrisham/samsung-galaxybook-extras #50: dkms build does not work

pyxfluff, 2024-12-20:
I installed Linux on my Galaxy Book2 Pro but it does not have working sound nor a keyboard backlight so I am trying this but I keep getting errors when I try building it. Here is the log:

```
DKMS make.log for samsung-galaxybook/extras for kernel 6.12.4-zen1-1-zen (x86_64)
Fri Dec 20 02:41:18 PM EST 2024
make: Entering directory '/usr/lib/modules/6.12.4-zen1-1-zen/build'
  CC [M]  /var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.o
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c: In function ‘get_acpi_device_description’:
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:256:26: error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’
  256 |         if (acpi_dev->pnp.str_obj != NULL && acpi_dev->pnp.str_obj->buffer.length > 0) {
      |                          ^
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:256:59: error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’
  256 |         if (acpi_dev->pnp.str_obj != NULL && acpi_dev->pnp.str_obj->buffer.length > 0) {
      |                                                           ^
In file included from ./include/linux/percpu.h:5,
                 from ./arch/x86/include/asm/msr.h:15,
                 from ./arch/x86/include/asm/tsc.h:10,
                 from ./arch/x86/include/asm/timex.h:6,
                 from ./include/linux/timex.h:67,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:13,
                 from /var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:15:
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:257:65: error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’
  257 |                 char *buf = kzalloc(sizeof(*buf) * acpi_dev->pnp.str_obj->buffer.length, GFP_KERNEL);
      |                                                                 ^
./include/linux/alloc_tag.h:206:16: note: in definition of macro ‘alloc_hooks_tag’
  206 |         typeof(_do_alloc) _res = _do_alloc;                             \
      |                ^~~~~~~~~
./include/linux/slab.h:1016:49: note: in expansion of macro ‘alloc_hooks’
 1016 | #define kzalloc(...)                            alloc_hooks(kzalloc_noprof(__VA_ARGS__))
      |                                                 ^~~~~~~~~~~
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:257:29: note: in expansion of macro ‘kzalloc’
  257 |                 char *buf = kzalloc(sizeof(*buf) * acpi_dev->pnp.str_obj->buffer.length, GFP_KERNEL);
      |                             ^~~~~~~
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:257:65: error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’
  257 |                 char *buf = kzalloc(sizeof(*buf) * acpi_dev->pnp.str_obj->buffer.length, GFP_KERNEL);
      |                                                                 ^
./include/linux/alloc_tag.h:206:34: note: in definition of macro ‘alloc_hooks_tag’
  206 |         typeof(_do_alloc) _res = _do_alloc;                             \
      |                                  ^~~~~~~~~
./include/linux/slab.h:1016:49: note: in expansion of macro ‘alloc_hooks’
 1016 | #define kzalloc(...)                            alloc_hooks(kzalloc_noprof(__VA_ARGS__))
      |                                                 ^~~~~~~~~~~
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:257:29: note: in expansion of macro ‘kzalloc’
  257 |                 char *buf = kzalloc(sizeof(*buf) * acpi_dev->pnp.str_obj->buffer.length, GFP_KERNEL);
      |                             ^~~~~~~
./include/linux/alloc_tag.h:212:1: error: initialization of ‘char *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
  212 | ({                                                                      \
      | ^
./include/linux/slab.h:1016:49: note: in expansion of macro ‘alloc_hooks’
 1016 | #define kzalloc(...)                            alloc_hooks(kzalloc_noprof(__VA_ARGS__))
      |                                                 ^~~~~~~~~~~
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:257:29: note: in expansion of macro ‘kzalloc’
  257 |                 char *buf = kzalloc(sizeof(*buf) * acpi_dev->pnp.str_obj->buffer.length, GFP_KERNEL);
      |                             ^~~~~~~
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:259:49: error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’
  259 |                         (wchar_t *)acpi_dev->pnp.str_obj->buffer.pointer,
      |                                                 ^
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:260:38: error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’
  260 |                         acpi_dev->pnp.str_obj->buffer.length,
      |                                      ^
make[2]: *** [scripts/Makefile.build:229: /var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.o] Error 1
make[1]: *** [/usr/lib/modules/6.12.4-zen1-1-zen/build/Makefile:1934: /var/lib/dkms/samsung-galaxybook/extras/build] Error 2
make: *** [Makefile:224: __sub-make] Error 2
make: Leaving directory '/usr/lib/modules/6.12.4-zen1-1-zen/build'

```

927A927, 2024-12-24:
Yes, the same error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’ with the stable mainline kernel 6.12.4

jusqua, 2024-12-26:
Hi, @pyxfluff and @927A927.
This problem is related to how 6.12 kernel changes and #46 already talked about this.

Simple solving, you need to do is use the branch from #44 PR, and remove the current added DKMS module.

This will completelly remove the added module:
```bash
rm -rf /var/lib/dkms/samsung-galaxybook
```

Go to `samsung-galaxy-extras` project folder and change branch to `refactor-use-existing-platform-device`:
```bash
git switch refactor-use-existing-platform-device
```

And repeat the steps to add dkms module (i.e. add, build, install).
Tell me if you have any problems.

927A927, 2024-12-26:
Thank you, I was looking for something like this for a lot.

pyxfluff, 2024-12-26:
Hi @joshuagrisham, I am seeing the same error after a git pull.
```
Sign command: /usr/lib/modules/6.12.6-zen1-1-zen/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j16 KERNELRELEASE=6.12.6-zen1-1-zen -C /usr/lib/modules/6.12.6-zen1-1-zen/build M=/var/lib/dkms/samsung-galaxybook/extras/build

Error! Bad return status for module build on kernel: 6.12.6-zen1-1-zen (x86_64)
Consult /var/lib/dkms/samsung-galaxybook/extras/build/make.log for more information.
[pyx@archlinux2 ~]$ cat /var/lib/dkms/samsung-galaxybook/extras/build/make.log
DKMS make.log for samsung-galaxybook/extras for kernel 6.12.6-zen1-1-zen (x86_64)
Thu Dec 26 06:03:13 PM EST 2024
make: Entering directory '/usr/lib/modules/6.12.6-zen1-1-zen/build'
  CC [M]  /var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.o
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c: In function ‘get_acpi_device_description’:
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:256:26: error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’
  256 |         if (acpi_dev->pnp.str_obj != NULL && acpi_dev->pnp.str_obj->buffer.length > 0) {
      |                          ^
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:256:59: error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’
  256 |         if (acpi_dev->pnp.str_obj != NULL && acpi_dev->pnp.str_obj->buffer.length > 0) {
      |                                                           ^
In file included from ./include/linux/percpu.h:5,
                 from ./arch/x86/include/asm/msr.h:15,
                 from ./arch/x86/include/asm/tsc.h:10,
                 from ./arch/x86/include/asm/timex.h:6,
                 from ./include/linux/timex.h:67,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:13,
                 from /var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:15:
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:257:65: error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’
  257 |                 char *buf = kzalloc(sizeof(*buf) * acpi_dev->pnp.str_obj->buffer.length, GFP_KERNEL);
      |                                                                 ^
./include/linux/alloc_tag.h:206:16: note: in definition of macro ‘alloc_hooks_tag’
  206 |         typeof(_do_alloc) _res = _do_alloc;                             \
      |                ^~~~~~~~~
./include/linux/slab.h:1016:49: note: in expansion of macro ‘alloc_hooks’
 1016 | #define kzalloc(...)                            alloc_hooks(kzalloc_noprof(__VA_ARGS__))
      |                                                 ^~~~~~~~~~~
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:257:29: note: in expansion of macro ‘kzalloc’
  257 |                 char *buf = kzalloc(sizeof(*buf) * acpi_dev->pnp.str_obj->buffer.length, GFP_KERNEL);
      |                             ^~~~~~~
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:257:65: error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’
  257 |                 char *buf = kzalloc(sizeof(*buf) * acpi_dev->pnp.str_obj->buffer.length, GFP_KERNEL);
      |                                                                 ^
./include/linux/alloc_tag.h:206:34: note: in definition of macro ‘alloc_hooks_tag’
  206 |         typeof(_do_alloc) _res = _do_alloc;                             \
      |                                  ^~~~~~~~~
./include/linux/slab.h:1016:49: note: in expansion of macro ‘alloc_hooks’
 1016 | #define kzalloc(...)                            alloc_hooks(kzalloc_noprof(__VA_ARGS__))
      |                                                 ^~~~~~~~~~~
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:257:29: note: in expansion of macro ‘kzalloc’
  257 |                 char *buf = kzalloc(sizeof(*buf) * acpi_dev->pnp.str_obj->buffer.length, GFP_KERNEL);
      |                             ^~~~~~~
./include/linux/alloc_tag.h:212:1: error: initialization of ‘char *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
  212 | ({                                                                      \
      | ^
./include/linux/slab.h:1016:49: note: in expansion of macro ‘alloc_hooks’
 1016 | #define kzalloc(...)                            alloc_hooks(kzalloc_noprof(__VA_ARGS__))
      |                                                 ^~~~~~~~~~~
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:257:29: note: in expansion of macro ‘kzalloc’
  257 |                 char *buf = kzalloc(sizeof(*buf) * acpi_dev->pnp.str_obj->buffer.length, GFP_KERNEL);
      |                             ^~~~~~~
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:259:49: error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’
  259 |                         (wchar_t *)acpi_dev->pnp.str_obj->buffer.pointer,
      |                                                 ^
/var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.c:260:38: error: ‘struct acpi_device_pnp’ has no member named ‘str_obj’
  260 |                         acpi_dev->pnp.str_obj->buffer.length,
      |                                      ^
make[2]: *** [scripts/Makefile.build:229: /var/lib/dkms/samsung-galaxybook/extras/build/samsung-galaxybook.o] Error 1
make[1]: *** [/usr/lib/modules/6.12.6-zen1-1-zen/build/Makefile:1935: /var/lib/dkms/samsung-galaxybook/extras/build] Error 2
make: *** [Makefile:224: __sub-make] Error 2
make: Leaving directory '/usr/lib/modules/6.12.6-zen1-1-zen/build'

```

pyxfluff, 2024-12-26:
Thanks, seems to have fixed, but my sound is still not working properly. Any fixes for that?

closed 2024-12-26 by pyxfluff