joshuagrisham/samsung-galaxybook-extras #51: High temperatures on NVMe

alexpacio, 2025-01-20:
Hi,

I'm using this kernel module in Fedora 41, kernel 6.12.10, in a Galaxy Book 3 Pro (SAM0429). 

I feel like the fans are not running properly because even when the system is idle, the first sensor on the NVMe disk stays at 65-75 celsius degrees and reaches very easily 90 cel degrees under heavy load. Maybe the fan speed policy is too conservative or maybe only some of the fans are working.

How can I ensure that all the fans available are controlled by the kernel module and how can I make the fan speed thresholds more aggressive?

joshuagrisham, 2025-01-23:
Hi @alexpacio ,

The fans on these devices are completely controlled by the embedded controller, and the only option we have is to set different "performance mode profiles" that Samsung have made available which hopefully work correctly on your device (High performance, Optimized, etc).

Basically just toggle the "performance mode" hotkey and/or set the platform profile using various tools (see the README for more info on this) and see if there is any specific profile that seems to work well for you.

Otherwise, unfortunately, there is not really anything else we can do. If you feel like no profile works to get exactly what you are looking for, then I think the only other option is to file a report with Samsung? (in theory these performance modes will work exactly the same in Linux vs Windows, so in theory you can reproduce the issue in Windows and send it to them from that perspective?)

alexpacio, 2025-01-23:
Thank you! Closing the ticket then.

closed 2025-01-23 by alexpacio