joshuagrisham/samsung-galaxybook-extras #53: Compiling errors on samsung-galaxybook-extras module

clahrion, 2025-01-22:
Hello Joshua,

I have a Samsung Galaxybook2 Pro 950XEE, and installed a Debian 12 on it. So, as all linux users, I experienced the errors on sound activation, keyboard light control, touchpad battery controls etc. I solved the sound problem using your galaxy-book2-pro-linux project (excellent!!), but I have experienced some issues (error messages) when compiling the samsung-galaxybook-extras module... I have installed all the needed linux-headers and modules, but the error messages I found from GCC are:

```
clahrion@MrSpock:~/Softwares/Debian/extras_samsung$ sudo make -C /lib/modules/`uname -r`/build M=$PWD
make: Entering directory '/usr/src/linux-headers-6.1.0-29-amd64'
CC [M]  /home/clahrion/Softwares/Debian/extras_samsung/samsung-galaxybook.o
/home/clahrion/Softwares/Debian/extras_samsung/samsung-galaxybook.c:809:24: error: initialization of ‘int (*)(struct power_supply *)’ from incompatible pointer type ‘int (*)(struct power_supply *, struct acpi_battery_hook *)’ [-Werror=incompatible-pointer-types]
809 |         .add_battery = galaxybook_battery_add,
|                        ^~~~~~~~~~~~~~~~~~~~~~
/home/clahrion/Softwares/Debian/extras_samsung/samsung-galaxybook.c:809:24: note: (near initialization for ‘galaxybook_battery_hook.add_battery’)
/home/clahrion/Softwares/Debian/extras_samsung/samsung-galaxybook.c:810:27: error: initialization of ‘int (*)(struct power_supply *)’ from incompatible pointer type ‘int (*)(struct power_supply *, struct acpi_battery_hook *)’ [-Werror=incompatible-pointer-types]
810 |         .remove_battery = galaxybook_battery_remove,
|                           ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/clahrion/Softwares/Debian/extras_samsung/samsung-galaxybook.c:810:27: note: (near initialization for ‘galaxybook_battery_hook.remove_battery’)
/home/clahrion/Softwares/Debian/extras_samsung/samsung-galaxybook.c:1184:17: warning: initialization discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
1184 |         .info = galaxybook_hwmon_info,
|                 ^~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[1]: *** [/usr/src/linux-headers-6.1.0-29-common/scripts/Makefile.build:255: /home/clahrion/Softwares/Debian/extras_samsung/samsung-galaxybook.o] Error 1
make: *** [/usr/src/linux-headers-6.1.0-29-common/Makefile:2034: /home/clahrion/Softwares/Debian/extras_samsung] Error 2
make: Leaving directory '/usr/src/linux-headers-6.1.0-29-amd64'
```

I think on "samsung-galaxybook.c" code there is any variable declaration incompatibility. Please, could you help me to solve that? 

Thank you very much!

Best Regards,

Caio Bastos

joshuagrisham, 2025-01-23:
Hi @clahrion ! It looks like your kernel version is 6.1.0 which is quite old at this point.. from the error message it looks like the interface for creating battery extensions had not been updated then.

Do you have possibility of running a later kernel? I have for example tested the driver a fair amount with a 6.5.x kernel and it worked quite well for me. Otherwise I can recommend 6.11.3 or higher in which case you should get the sound driver fix automatically from the kernel and not need to run any kind of script etc 🙂

clahrion, 2025-01-24:
Hi @joshuagrisham !

Thanks for clarifying me! I just upgraded the kernel version to 6.10.11 and got to compile the samsung-galaxybook-extras module! As you said, the kernel 6.11 has the sound problem solved, but I could not use it because the wifi module doesn't work on kernel versions higher than 6.10. So, anyway, your module is working very well in my laptop now.

Thanks!!