joshuagrisham/samsung-galaxybook-extras #57: Support for Samsung Galaxy Book5 Pro 360

WaveDevelops, 2025-01-29:
Hello! :)

Apologies in advance if this is the wrong way to go about this, this isn't a direct issue but there's no discussion forum on this repo so I figured this would be the best way to communicate.

I've just gotten one of these new Galaxy Books and am excited for this driver to reach mainline for ease of use.

But basically, I was curious if there would be anything I could provide from my laptop that could assist in validating the current driver set(It appears mine uses SAM0430, so excellent news there) or improve support for 360-related features for mainline (Such as the convertability/tablet mode)?

joshuagrisham, 2025-02-23:
Hi @WaveDevelops thanks for posting here!

I am a little lost in the details in some of your post but first step I wonder is if you might have tried the very latest and in some ways totally refactored version of the driver that will be coming in mainline soon-ish (probably 6.14 I think) ? Details how you can find it and test here: https://github.com/joshuagrisham/samsung-galaxybook-extras/discussions/58

Regarding your keyboard hotkeys, just so I am clear ..

1. Performance Mode: there is no hotkey toggle for this on your keyboard?  How is it for the device in Windows, is it only via the Samsung Settings app that you can change this value?  And do the modes line up well with what you see in the Windows app vs what you get as platform profile choices from `samsung-galaxybook` in Linux?
2. F9 is "Backlight" do you mean screen backlight (where mine is F2 for + and F3 for -) or Keyboard backlight (mine is F9 as a single multi-level toggle key) ?
3. Then you have F10 for microphone mute/unmute (reports 0x6e) - does the microphone actually get muted and you just see the unrecognized key warning, or does nothing happen at all?
4. and F11 for Camera toggle? (reports 0x6f) again does the camera actually get blocked or not? And if you use the "block_recording" firmware attribute of the driver, is the camera and/or microphone blocked when you enable or disable it?
5. For "settings" what I did in my update to [systemd's default hwdb](https://github.com/systemd/systemd/blob/af69ea714d8e75055561acfd607970742136655f/hwdb.d/60-keyboard.hwdb#L1919) was set it to "prog1" as that is the mapping used almost universally by various "settings" hotkeys; then on my device I just set up a shortcut in GNOME to map "prog1" to open up the GNOME settings app. If you want I can create another PR to systemd to add mapping for this 0x7c on your device to work in the same way

WaveDevelops, 2025-02-23:
I've not tested out the refactor yet! I'll try and do that this week. :)

1. Nope! No power mode toggle at all, pretty interesting! In Windows either through the app, or the integration with the Windows Power Management settings itself. I'm not sure, but when using the driver are the Gnome Power Management options overwritten, or are those different?
2. My apologies, keyboard backlight is what I was referring to for F9. The screen dimmer/brightener are F2 and F3 respectively, and work fine.
3. Microphone does not get muted no, with all the "unrecognized" messages there's no change. In any case, the Microphone itself is on by default and works fine with or without the driver.
4. Same as above, no blocking/unblocking seems to occur. Though it's worth noting that I've had no luck getting the camera detected in Linux at all. I'm not sure if this is a Samsung problem, or a Lunar Lake problem that will just get fixed with future kernel updates.
5. That makes sense I think? Ultimately I don't think I'd ever use this shortcut anyway, but if you want uniformity for future users that grab Samsung's latest laptops it might be handy? Up to you!

melvyn2, 2025-03-10:
> Probably unrelated to to this driver as I see no mention of this functionality anyway, but the laptop also doesn't detect being flipped 360 - and thus doesn't go into tablet mode (Keyboard still works, touchpad still works, etc... Makes it unusable flipped)

Samsung uses a custom Intel Sensor Hub firmware which unfortunately can't be upstreamed to linux-firmware. See https://dnsense.pub/posts/9-book5-sensor-hub/

I'm looking forwards to trying out this module but unfortunately it does not yet work on the 6.15 RC series.

WaveDevelops, 2025-03-26:
@melvyn2 You are my hero! This was my biggest gripe with using Linux on this device because I actually do want to use tablet mode relatively often to make game art.

I'm actually on a Fedora Kinoite variant (Bazzite specifically), so I had to wrap my own rpm to layer the firmware into it's proper place.

This Reddit post helped me out:
https://www.reddit.com/r/Fedora/comments/1c3cc2l/kinoitesilverblue_how_to_install_custom_firmware/

But for a "full guide", the steps are below:

1. Nab the .bin file from your Windows partition at /Windows/System32/DriverStore/FileRepository/ishheciextensiontemplate.inf_amd64_????????????????/FwImage/0003/ as per melvyn2's website post.
2. Rename the file to ish_lnlm_053dca6a.bin
3. Create an "intel" folder with an "ish" folder inside it, and place the .bin inside so that the full directory is /intel/ish/ish_lnlm_053dca6a.bin
4. Install FPM via whatever method works best for you: https://fpm.readthedocs.io/en/latest/installation.html (I just followed the basic Fedora instructions in a Fedora Distrobox. But you could also use Brew or Layer ruby, whatever you find easiest).
5. From the directory containing your /intel/ folder, run: fpm -s dir -t rpm -n **intelsensor_patch** intel=/lib/firmware/ (Replace **intelsensor_patch** with whatever you want to call the .rpm so you can easily identify it with **rpm-ostree status**)
6. Open the .rpm file with ARK or any other archival tool and verify that the full director is: /lib/firmware/intel/ish/ish_lnlm_053dca6a.bin
7. Locate the created .rpm file, and run rpm-ostree install **RPMNAME**.rpm (The Reddit post above mentions using --force-replacefiles flag, but I didn't find that necessary as there was no pre-existing file).
8. Reboot, and voila! Working tablet mode. :)

Now just gotta wait for this module to hit mainline and this laptop will basically be fully functional!

WaveDevelops, 2025-03-26:
Oh and in reviewing my original post, I forgot to mention that another thing that I found to not be working with the driver is the speakers. Though once again this may just be outside the scope of the driver itself. Plugging a headset into the audio jack works fine, and all volume functionality does also, so it's an interesting issue. I probably forgot to mention it because I always either use a headset or bluetooth for audio anyway - so it's not a priority for me.

melvyn2, 2025-03-26:
Not at all related to the topic at hand, but have you noticed any full-system hard freezes (sometimes with a black screen instead of a frozen screen) on the book5? It's not something I've heard about on any other lunar lake systems, and I was worried that it could be a hardware issue but I haven't seen it happen on windows. I'm pretty confused as to what kind of issue I'm facing.

WaveDevelops, 2025-03-26:
I've had a few of them since I got the laptop, ya. Tho they've been exceptionally rare and I haven't found a pattern to them (But I've been lazy about checking logs after them to see if anything looks amiss). That said, it's far less frequent than my older Acer laptop - which also only got freezes on Linux - so I'm more inclined to think it's just instability from something I commonly have installed than the hardware itself... But hard to say lol

And actually I haven't had one in quite a while now, so maybe a firmware update clinched mine... Tho now that I've said that I'll probably have one later today...

Anyway not sure about the off-topic rules here, but feel free to HMU on Discord if ya wanna discuss other Book5 shenanigans that doesn't relate to drivers/firmware. :)
Discord Handle: twilitwave

melvyn2, 2025-03-26:
For audio see: https://github.com/thesofproject/linux/issues/5242

The freezes have mostly happened under load, which is especially annoying for playing games... But even then they sometimes happen when mostly idle, so I also don't really see much of a pattern. Hopefully it's something that can be fixed.