joshuagrisham/samsung-galaxybook-extras #62: Compilation error - Galaxybook 4 360

brunnogama, 2025-03-06:
Hey, when I try to compile, it is giving some errors, see log.
Kernel: k 6.13.5-200.fc41.x86_64 #1 S
Fedora 41 - Gnome 47

```
sudo dkms build samsung-galaxybook/extras

Sign command: /lib/modules/6.13.5-200.fc41.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub
Certificate or key are missing, generating self signed certificate for MOK...

Cleaning build area... done.
Building module(s)...(bad exit status: 2)
Failed command:
make -j12 KERNELRELEASE=6.13.5-200.fc41.x86_64 -C /lib/modules/6.13.5-200.fc41.x86_64/build M=/var/lib/dkms/samsung-galaxybook/extras/build

Error! Bad return status for module build on kernel: 6.13.5-200.fc41.x86_64 (x86_64)
Consult /var/lib/dkms/samsung-galaxybook/extras/build/make.log for more information.

```

LOG:

DKMS (dkms-3.1.5) make.log for samsung-galaxybook/extras for kernel 6.13.5-200.fc41.x86_64 (x86_64)
qui 06 mar 2025 18:31:06 -03
Cleaning build area
# command: make -C /lib/modules/6.13.5-200.fc41.x86_64/build M=/var/lib/dkms/samsung-galaxybook/extras/build clean
make: Entrando no diretório '/usr/src/kernels/6.13.5-200.fc41.x86_64'

make[1]: Entrando no diretório '/var/lib/dkms/samsung-galaxybook/extras/build'

make[1]: Saindo do diretório '/var/lib/dkms/samsung-galaxybook/extras/build'

make: Saindo do diretório '/usr/src/kernels/6.13.5-200.fc41.x86_64'

# exit code: 0
# elapsed time: 00:00:00
Building module(s)
# command: make -j12 KERNELRELEASE=6.13.5-200.fc41.x86_64 -C /lib/modules/6.13.5-200.fc41.x86_64/build M=/var/lib/dkms/samsung-galaxybook/extras/build
make: Entrando no diretório '/usr/src/kernels/6.13.5-200.fc41.x86_64'

make[1]: Entrando no diretório '/var/lib/dkms/samsung-galaxybook/extras/build'

  CC [M]  samsung-galaxybook.o
samsung-galaxybook.c:1936:10: error: ‘struct platform_driver’ has no member named ‘remove_new’; did you mean ‘remove’?
 1936 |         .remove_new = galaxybook_remove,
      |          ^~~~~~~~~~
      |          remove
make[3]: *** [/usr/src/kernels/6.13.5-200.fc41.x86_64/scripts/Makefile.build:196: samsung-galaxybook.o] Error 1
make[2]: *** [/usr/src/kernels/6.13.5-200.fc41.x86_64/Makefile:2024: .] Error 2
make[1]: *** [/usr/src/kernels/6.13.5-200.fc41.x86_64/Makefile:263: __sub-make] Error 2
make[1]: Saindo do diretório '/var/lib/dkms/samsung-galaxybook/extras/build'

make: *** [Makefile:263: __sub-make] Error 2
make: Saindo do diretório '/usr/src/kernels/6.13.5-200.fc41.x86_64'

# exit code: 2
# elapsed time: 00:00:02

melvyn2, 2025-03-10:
More on 6.14-rc5:
```c
make -C /nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/build M=/build/source modules
make[1]: Entering directory '/nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/build'
make[2]: Entering directory '/build/source'
  CC [M]  samsung-galaxybook.o
samsung-galaxybook.c:105:41: error: field 'profile_handler' has incomplete type
  105 |         struct platform_profile_handler profile_handler;
      |                                         ^~~~~~~~~~~~~~~
In file included from /nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/source/include/linux/container_of.h:5,
                 from /nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/source/include/linux/kernel.h:22,
                 from samsung-galaxybook.c:14:
/nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/source/include/linux/build_bug.h:78:41: error: static assertion failed: "ARRAY_SIZE(profile_names) == PLATFORM_PROFILE_LAST"
   78 | #define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
      |                                         ^~~~~~~~~~~~~~
/nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/source/include/linux/build_bug.h:77:34: note: in expansion of macro '__static_assert'
   77 | #define static_assert(expr, ...) __static_assert(expr, ##__VA_ARGS__, #expr)
      |                                  ^~~~~~~~~~~~~~~
samsung-galaxybook.c:1275:1: note: in expansion of macro 'static_assert'
 1275 | static_assert(ARRAY_SIZE(profile_names) == PLATFORM_PROFILE_LAST);
      | ^~~~~~~~~~~~~
samsung-galaxybook.c: In function 'galaxybook_platform_profile_set':
/nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/source/include/linux/compiler_types.h:483:27: error: expression in static assertion is not an integer
  483 | #define __same_type(a, b) __builtin_types_compatible_p(typeof(a), typeof(b))
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/source/include/linux/build_bug.h:78:56: note: in definition of macro '__static_assert'
   78 | #define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
      |                                                        ^~~~
/nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/source/include/linux/container_of.h:20:9: note: in expansion of macro 'static_assert'
   20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
      |         ^~~~~~~~~~~~~
/nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/source/include/linux/container_of.h:20:23: note: in expansion of macro '__same_type'
   20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
      |                       ^~~~~~~~~~~
samsung-galaxybook.c:1280:49: note: in expansion of macro 'container_of'
 1280 |         struct samsung_galaxybook *galaxybook = container_of(pprof,
      |                                                 ^~~~~~~~~~~~
samsung-galaxybook.c: In function 'galaxybook_platform_profile_get':
/nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/source/include/linux/compiler_types.h:483:27: error: expression in static assertion is not an integer
  483 | #define __same_type(a, b) __builtin_types_compatible_p(typeof(a), typeof(b))
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/source/include/linux/build_bug.h:78:56: note: in definition of macro '__static_assert'
   78 | #define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
      |                                                        ^~~~
/nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/source/include/linux/container_of.h:20:9: note: in expansion of macro 'static_assert'
   20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
      |         ^~~~~~~~~~~~~
/nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/source/include/linux/container_of.h:20:23: note: in expansion of macro '__same_type'
   20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
      |                       ^~~~~~~~~~~
samsung-galaxybook.c:1296:49: note: in expansion of macro 'container_of'
 1296 |         struct samsung_galaxybook *galaxybook = container_of(pprof,
      |                                                 ^~~~~~~~~~~~
samsung-galaxybook.c: In function 'galaxybook_profile_init':
samsung-galaxybook.c:1436:15: error: too few arguments to function 'platform_profile_register'
 1436 |         err = platform_profile_register(&galaxybook->profile_handler);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from samsung-galaxybook.c:23:
/nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/source/include/linux/platform_profile.h:47:16: note: declared here
   47 | struct device *platform_profile_register(struct device *dev, const char *name,
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~
samsung-galaxybook.c: In function 'galaxybook_profile_exit':
[... cut ...]
make[3]: *** [/nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/source/Makefile:1989: .] Error 2
make[2]: *** [/nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/source/Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/build/source'
make[1]: *** [/nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/source/Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/nix/store/vnrjiy1ycy29595hh17kksf8p7rr9mf2-linux-6.14-rc5-dev/lib/modules/6.14.0-rc5/build'
make: *** [Makefile:5: all] Error 2
```

joshuagrisham, 2025-04-24:
Hi @brunnogama , please try the version of the driver in the [pre-6.14](https://github.com/joshuagrisham/samsung-galaxybook-extras/tree/pre-6.14) branch if you wish to use it with kernel 6.13.x -- it has a small update that will hopefully avoid this problem.

@melvyn2 there is some new information in the readme file of the main branch; my recommendation is to use the mainline driver if you are running 6.14.x+ and not run the old prototype from this repo 🙂 (and @brunnogama if you have chance to run 6.14.x+ then of course it is good for you to do this as well! 😉 )

melvyn2, 2025-04-24:
I don't see the driver in 6.14: https://web.git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/platform/x86?h=v6.14

joshuagrisham, 2025-04-26:
> I don't see the driver in 6.14: https://web.git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/platform/x86?h=v6.14

Hi again @melvyn2 ,
The driver was not merged into mainline as part of the 6.14 release but was moved into linux next and the PDX86 branches during 6.14 .. for 6.15 it has now been merged to mainline and barring any other issues should release with 6.15.

Regardless as it was merged into the other trees during 6.14 then it should be working with the various APIs that are also in 6.14 (this is actually exactly how I am running it on my own device -- 6.14 with manually adding and building the driver on top).

See https://github.com/joshuagrisham/samsung-galaxybook-extras/issues/67 for more info