joshuagrisham/samsung-galaxybook-extras #65: Build error on 6.14.1-300.fc42.x86_64 / Fedora 42 beta

m0unds, 2025-04-12:
Hi, ran into a build issue with Fedora 42 pre-release on a Galaxybook2 Pro
gcc (GCC) 15.0.1 20250329 (Red Hat 15.0.1-0)

$ make -C /lib/modules/`uname -r`/build M=$PWD
make: Entering directory '/usr/src/kernels/6.14.1-300.fc42.x86_64'
make[1]: Entering directory '/home/m0unds/samsung-galaxybook-extras'
  CC [M]  samsung-galaxybook.o
samsung-galaxybook.c:105:41: error: field ‘profile_handler’ has incomplete type
  105 |         struct platform_profile_handler profile_handler;
      |                                         ^~~~~~~~~~~~~~~
In file included from /usr/src/kernels/6.14.1-300.fc42.x86_64/include/linux/container_of.h:5,
                 from /usr/src/kernels/6.14.1-300.fc42.x86_64/include/linux/kernel.h:22,
                 from samsung-galaxybook.c:14:
/usr/src/kernels/6.14.1-300.fc42.x86_64/include/linux/build_bug.h:78:41: error: static assertion failed: "ARRAY_SIZE(profile_names) == PLATFORM_PROFILE_LAST"
   78 | #define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
      |                                         ^~~~~~~~~~~~~~
/usr/src/kernels/6.14.1-300.fc42.x86_64/include/linux/build_bug.h:77:34: note: in expansion of macro ‘__static_assert’
   77 | #define static_assert(expr, ...) __static_assert(expr, ##__VA_ARGS__, #expr)
      |                                  ^~~~~~~~~~~~~~~
samsung-galaxybook.c:1275:1: note: in expansion of macro ‘static_assert’
 1275 | static_assert(ARRAY_SIZE(profile_names) == PLATFORM_PROFILE_LAST);
      | ^~~~~~~~~~~~~
samsung-galaxybook.c: In function ‘galaxybook_platform_profile_set’:
/usr/src/kernels/6.14.1-300.fc42.x86_64/include/linux/compiler_types.h:483:27: error: expression in static assertion is not an integer
  483 | #define __same_type(a, b) __builtin_types_compatible_p(typeof(a), typeof(b))
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/src/kernels/6.14.1-300.fc42.x86_64/include/linux/build_bug.h:78:56: note: in definition of macro ‘__static_assert’
   78 | #define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
      |                                                        ^~~~
/usr/src/kernels/6.14.1-300.fc42.x86_64/include/linux/container_of.h:20:9: note: in expansion of macro ‘static_assert’
   20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
      |         ^~~~~~~~~~~~~
/usr/src/kernels/6.14.1-300.fc42.x86_64/include/linux/container_of.h:20:23: note: in expansion of macro ‘__same_type’
   20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
      |                       ^~~~~~~~~~~
samsung-galaxybook.c:1280:49: note: in expansion of macro ‘container_of’
 1280 |         struct samsung_galaxybook *galaxybook = container_of(pprof,
      |                                                 ^~~~~~~~~~~~
samsung-galaxybook.c: In function ‘galaxybook_platform_profile_get’:
/usr/src/kernels/6.14.1-300.fc42.x86_64/include/linux/compiler_types.h:483:27: error: expression in static assertion is not an integer
  483 | #define __same_type(a, b) __builtin_types_compatible_p(typeof(a), typeof(b))
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/src/kernels/6.14.1-300.fc42.x86_64/include/linux/build_bug.h:78:56: note: in definition of macro ‘__static_assert’
   78 | #define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
      |                                                        ^~~~
/usr/src/kernels/6.14.1-300.fc42.x86_64/include/linux/container_of.h:20:9: note: in expansion of macro ‘static_assert’
   20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
      |         ^~~~~~~~~~~~~
/usr/src/kernels/6.14.1-300.fc42.x86_64/include/linux/container_of.h:20:23: note: in expansion of macro ‘__same_type’
   20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
      |                       ^~~~~~~~~~~
samsung-galaxybook.c:1296:49: note: in expansion of macro ‘container_of’
 1296 |         struct samsung_galaxybook *galaxybook = container_of(pprof,
      |                                                 ^~~~~~~~~~~~
samsung-galaxybook.c: In function ‘galaxybook_profile_init’:
samsung-galaxybook.c:1436:15: error: too few arguments to function ‘platform_profile_register’; expected 4, have 1
 1436 |         err = platform_profile_register(&galaxybook->profile_handler);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from samsung-galaxybook.c:23:
/usr/src/kernels/6.14.1-300.fc42.x86_64/include/linux/platform_profile.h:50:16: note: declared here
   50 | struct device *platform_profile_register(struct device *dev, const char *name,
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~
samsung-galaxybook.c: In function ‘galaxybook_profile_exit’:
samsung-galaxybook.c:1459:9: error: too few arguments to function ‘platform_profile_remove’; expected 1, have 0
 1459 |         platform_profile_remove();
      |         ^~~~~~~~~~~~~~~~~~~~~~~
/usr/src/kernels/6.14.1-300.fc42.x86_64/include/linux/platform_profile.h:53:5: note: declared here
   53 | int platform_profile_remove(struct device *dev);
      |     ^~~~~~~~~~~~~~~~~~~~~~~
samsung-galaxybook.c: In function ‘galaxybook_probe’:
[... 22 lines cut ...]
samsung-galaxybook.c:1896:37: error: passing argument 1 of ‘i8042_remove_filter’ from incompatible pointer type [-Wincompatible-pointer-types]
 1896 |                 i8042_remove_filter(galaxybook_i8042_filter);
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~
      |                                     |
      |                                     bool (*)(unsigned char,  unsigned char,  struct serio *) {aka _Bool (*)(unsigned char,  unsigned char,  struct serio *)}
/usr/src/kernels/6.14.1-300.fc42.x86_64/include/linux/i8042.h:79:40: note: expected ‘i8042_filter_t’ {aka ‘_Bool (*)(unsigned char,  unsigned char,  struct serio *, void *)’} but argument is of type ‘bool (*)(unsigned char,  unsigned char,  struct serio *)’ {aka ‘_Bool (*)(unsigned char,  unsigned char,  struct serio *)’}
   79 | int i8042_remove_filter(i8042_filter_t filter);
      |                         ~~~~~~~~~~~~~~~^~~~~~
samsung-galaxybook.c:1529:13: note: ‘galaxybook_i8042_filter’ declared here
 1529 | static bool galaxybook_i8042_filter(unsigned char data, unsigned char str,
      |             ^~~~~~~~~~~~~~~~~~~~~~~
/usr/src/kernels/6.14.1-300.fc42.x86_64/include/linux/i8042.h:70:16: note: ‘i8042_filter_t’ declared here
   70 | typedef bool (*i8042_filter_t)(unsigned char data, unsigned char str, struct serio *serio,
      |                ^~~~~~~~~~~~~~
samsung-galaxybook.c: At top level:
samsung-galaxybook.c:1936:10: error: ‘struct platform_driver’ has no member named ‘remove_new’; did you mean ‘remove’?
 1936 |         .remove_new = galaxybook_remove,
      |          ^~~~~~~~~~
      |          remove
make[3]: *** [/usr/src/kernels/6.14.1-300.fc42.x86_64/scripts/Makefile.build:207: samsung-galaxybook.o] Error 1
make[2]: *** [/usr/src/kernels/6.14.1-300.fc42.x86_64/Makefile:2034: .] Error 2
make[1]: *** [/usr/src/kernels/6.14.1-300.fc42.x86_64/Makefile:263: __sub-make] Error 2
make[1]: Leaving directory '/home/m0unds/samsung-galaxybook-extras'
make: *** [Makefile:263: __sub-make] Error 2
make: Leaving directory '/usr/src/kernels/6.14.1-300.fc42.x86_64'

joshuagrisham, 2025-04-13:
Hi @m0unds 
The driver has been added to mainline now (it will release first with 6.15) and there are several breaking changes that came with various APIs used in 6.14.
My recommendation is that you actually build the mainline driver against 6.14 instead (or use 6.15-rc1 entirely if you like) and do not use the driver code provided in this repository; I have updated the README file now in this repository to try and make this very clear (https://github.com/joshuagrisham/samsung-galaxybook-extras/blob/main/README.md) but please feel free to have a look and say if this makes sense or not 🙂

m0unds, 2025-04-13:
Perfect. I’ll take a look. Thanks!

m0unds, 2025-04-13:
Built out of tree, worked perfect. Thanks again for your work on this!

closed 2025-04-13 by m0unds