joshuagrisham/samsung-galaxybook-extras #66: Add support Galaxy Book 3 Pro 14' (940XFG)

kity-linuxero, 2025-04-20:
Hello everyone, first of all, thank you to Joshua for his great work on this driver.

I have the 14-inch **Galaxy Book 3 Pro** (940XFG). I've tested it with 6.15-rc2 and it works very well, but the things that don't work yet are:
- Speakers (this model has 4 speakers).
- Fingerprint sensor
- Webcam.

Is there anything I can do to help support this model? For example, *providing some info about my hardware*? Unfortunately, I don't have the necessary knowledge to develop the driver myself, but I'd like to contribute in any way I can.

Please reply to this issue if you have any ideas for me.

Thanks, greetings from Argentina.

joshuagrisham, 2025-04-24:
Hi @kity-linuxero !

Regarding the speakers, have you looked into what it says in the Readme file regarding this? (https://github.com/joshuagrisham/samsung-galaxybook-extras/tree/main?tab=readme-ov-file#sound-from-the-speakers-enabling-speaker-amps) 
Specifically I wonder if you have tried either of the two existing models, and if they work then we should be able to send a patch to get it added to mainline if you like. Otherwise there is some additional info there about doing traces to figure it out, as well!
Also I have heard if you are using something like Arch then you might have to manually install the SOF Project firmware (e.g. from this package I guess?? https://archlinux.org/packages/extra/x86_64/sof-firmware/ ) in case it is missing

Regarding the fingerprint sensor.. hmm maybe it depends on which exact model of fingerprint sensor you have? Because I created a driver for this as well and tested it on a GB3 Pro (but don't remember if it was 14" or even 940XFG..): https://gitlab.freedesktop.org/libfprint/libfprint/-/blob/master/libfprint/drivers/egismoc/egismoc.c
So in theory if you have a later version of libfprint and your reader's vendor/product ID matches one of these, then hopefully it works?  There are some reports of devices with upgraded firmware on the reader that forces SDCP which we have not quite gotten working though (could be good to know) -- there are some issues about this in libfprint's gitlab so feel free to read through them and/or create a new issue there if your device is not working!

Webcam unfortunately I have no idea, sorry! I guess a start would be if you had a vendor/product ID you could post from `lsusb` and any kernel logs ... but I have heard also that there is some kind of Intel driver you can install that might get it working?

rodrigost23, 2025-07-29:
I have the same issue as @DaFray31 with fingerprint in the Galaxy Book 3 Ultra: it just deletes all enrolled fingers with the same message? `Deleted stored finger 7 for user myuser as it is unknown to device.`

Camera does not work as well.

DaFray31, 2025-07-29:
> That laptop has four speakers, like the Galaxy Book 3 Pro 14. Did you make any changes to get the audio output working? Or is the fix for the audio already included in the Linux 6.15.7-200.fc42.x86_64 kernel?

I haven't touched or modified anything for the speakers. All four are working. I use JamesDSP to have better EQ, but nothing more. 
So yep, maybe fixed in the 6.15.7

rodrigost23, 2025-09-17:
OK, fingerprint needs https://gitlab.freedesktop.org/libfprint/libfprint/-/merge_requests/544 to work.
But how about camera?