joshuagrisham/samsung-galaxybook-extras #68: Kernel 6.15rc

Michedev, 2025-05-15:
Hey,

first of all thanks for the effort for this kernel module. I wanted to ask if it is possible to use this module on the release candidate 5 of the kenel 6.15, as I installed it but it did not seem activated and I didn't notice hardware difference. I tried to check on lsmod the activated modules but I did not see any samsung or galaxy module. Should I check in other ways? 

Thanks in advance for the reply

IltonPfleger, 2025-05-19:
Yes, me too, after installing the mainline version on 6.14 it seems to have no effect on the system, the platform_driver pointers and others do not appear, it seems that the module simply does not load and there is nothing in dmesg other than the module's missing signature messages.

antipodite, 2025-05-19:
Having the same issue, I can load the module with `modprobe` but there seems to be no effect, keyboard backlight fn key still does nothing on kernel 6.14.6

manatails, 2025-05-20:
I am on 6.15-rc6 and the module loads up as intended. The backlight fn key doesn't work for me but it's just the key that isn't working and manually controlling the backlight from KDE settings does work.

IltonPfleger, 2025-05-21:
I solve my problem removing acpi_osi=! acpi_osi="Windows 2022" from the kernel command line. Everything works fine now.

Michedev, 2025-05-27:
I tried to use the last mainline version 6.15.0-1 but I have a compilation error similar to https://github.com/joshuagrisham/samsung-galaxybook-extras/issues/62#issuecomment-2711811184  - long story short  I have the error _field ‘profile_handler’ has incomplete type_

I put here the log file 

```
DKMS (dkms-3.2.1) make.log for samsung-galaxybook/extras for kernel 6.15.0-1-mainline (x86_64)
Tue May 27 09:46:12 AM CEST 2025

Building module(s)
# command: make -j8 KERNELRELEASE=6.15.0-1-mainline -C /usr/lib/modules/6.15.0-1-mainline/build M=/var/lib/dkms/samsung-galaxybook/extras/build
make: Entering directory '/usr/lib/modules/6.15.0-1-mainline/build'
make[1]: Entering directory '/var/lib/dkms/samsung-galaxybook/extras/build'
  CC [M]  samsung-galaxybook.o
samsung-galaxybook.c:105:41: error: field ‘profile_handler’ has incomplete type
  105 |         struct platform_profile_handler profile_handler;
      |                                         ^~~~~~~~~~~~~~~
In file included from /usr/lib/modules/6.15.0-1-mainline/build/include/linux/container_of.h:5,
                 from /usr/lib/modules/6.15.0-1-mainline/build/include/linux/kernel.h:22,
                 from samsung-galaxybook.c:14:
/usr/lib/modules/6.15.0-1-mainline/build/include/linux/build_bug.h:78:41: error: static assertion failed: "ARRAY_SIZE(profile_names) == PLATFORM_PROFILE_LAST"
   78 | #define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
      |                                         ^~~~~~~~~~~~~~
/usr/lib/modules/6.15.0-1-mainline/build/include/linux/build_bug.h:77:34: note: in expansion of macro ‘__static_assert’
   77 | #define static_assert(expr, ...) __static_assert(expr, ##__VA_ARGS__, #expr)
      |                                  ^~~~~~~~~~~~~~~
samsung-galaxybook.c:1275:1: note: in expansion of macro ‘static_assert’
 1275 | static_assert(ARRAY_SIZE(profile_names) == PLATFORM_PROFILE_LAST);
      | ^~~~~~~~~~~~~
samsung-galaxybook.c: In function ‘galaxybook_platform_profile_set’:
/usr/lib/modules/6.15.0-1-mainline/build/include/linux/compiler_types.h:498:27: error: expression in static assertion is not an integer
  498 | #define __same_type(a, b) __builtin_types_compatible_p(typeof(a), typeof(b))
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/modules/6.15.0-1-mainline/build/include/linux/build_bug.h:78:56: note: in definition of macro ‘__static_assert’
   78 | #define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
      |                                                        ^~~~
/usr/lib/modules/6.15.0-1-mainline/build/include/linux/container_of.h:20:9: note: in expansion of macro ‘static_assert’
   20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
      |         ^~~~~~~~~~~~~
/usr/lib/modules/6.15.0-1-mainline/build/include/linux/container_of.h:20:23: note: in expansion of macro ‘__same_type’
   20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
      |                       ^~~~~~~~~~~
samsung-galaxybook.c:1280:49: note: in expansion of macro ‘container_of’
 1280 |         struct samsung_galaxybook *galaxybook = container_of(pprof,
      |                                                 ^~~~~~~~~~~~
samsung-galaxybook.c: In function ‘galaxybook_platform_profile_get’:
/usr/lib/modules/6.15.0-1-mainline/build/include/linux/compiler_types.h:498:27: error: expression in static assertion is not an integer
  498 | #define __same_type(a, b) __builtin_types_compatible_p(typeof(a), typeof(b))
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/modules/6.15.0-1-mainline/build/include/linux/build_bug.h:78:56: note: in definition of macro ‘__static_assert’
   78 | #define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
      |                                                        ^~~~
/usr/lib/modules/6.15.0-1-mainline/build/include/linux/container_of.h:20:9: note: in expansion of macro ‘static_assert’
   20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
      |         ^~~~~~~~~~~~~
/usr/lib/modules/6.15.0-1-mainline/build/include/linux/container_of.h:20:23: note: in expansion of macro ‘__same_type’
   20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
      |                       ^~~~~~~~~~~
samsung-galaxybook.c:1296:49: note: in expansion of macro ‘container_of’
 1296 |         struct samsung_galaxybook *galaxybook = container_of(pprof,
      |                                                 ^~~~~~~~~~~~
samsung-galaxybook.c: In function ‘galaxybook_profile_init’:
samsung-galaxybook.c:1436:15: error: too few arguments to function ‘platform_profile_register’; expected 4, have 1
 1436 |         err = platform_profile_register(&galaxybook->profile_handler);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from samsung-galaxybook.c:23:
/usr/lib/modules/6.15.0-1-mainline/build/include/linux/platform_profile.h:50:16: note: declared here
   50 | struct device *platform_profile_register(struct device *dev, const char *name,
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~
samsung-galaxybook.c: In function ‘galaxybook_profile_exit’:
samsung-galaxybook.c:1459:9: error: too few arguments to function ‘platform_profile_remove’; expected 1, have 0
[... 29 lines cut ...]
 1896 |                 i8042_remove_filter(galaxybook_i8042_filter);
      |                                     ^~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
make[3]: *** [/usr/lib/modules/6.15.0-1-mainline/build/scripts/Makefile.build:203: samsung-galaxybook.o] Error 1
make[2]: *** [/usr/lib/modules/6.15.0-1-mainline/build/Makefile:2003: .] Error 2
make[1]: *** [/usr/lib/modules/6.15.0-1-mainline/build/Makefile:248: __sub-make] Error 2
make[1]: Leaving directory '/var/lib/dkms/samsung-galaxybook/extras/build'
make: *** [Makefile:248: __sub-make] Error 2
make: Leaving directory '/usr/lib/modules/6.15.0-1-mainline/build'

# exit code: 2
# elapsed time: 00:00:02
----------------------------------------------------------------

```

IltonPfleger, 2025-05-27:
> I tried to use the last mainline version 6.15.0-1 but I have a compilation error similar to [#62 (comment)](https://github.com/joshuagrisham/samsung-galaxybook-extras/issues/62#issuecomment-2711811184) - long story short I have the error _field ‘profile_handler’ has incomplete type_
> 
> My model is NP944XHA-KG2DE
> 
> I put here the log file
> 
> ```
> DKMS (dkms-3.2.1) make.log for samsung-galaxybook/extras for kernel 6.15.0-1-mainline (x86_64)
> Tue May 27 09:46:12 AM CEST 2025
> 
> Building module(s)
> # command: make -j8 KERNELRELEASE=6.15.0-1-mainline -C /usr/lib/modules/6.15.0-1-mainline/build M=/var/lib/dkms/samsung-galaxybook/extras/build
> make: Entering directory '/usr/lib/modules/6.15.0-1-mainline/build'
> make[1]: Entering directory '/var/lib/dkms/samsung-galaxybook/extras/build'
>   CC [M]  samsung-galaxybook.o
> samsung-galaxybook.c:105:41: error: field ‘profile_handler’ has incomplete type
>   105 |         struct platform_profile_handler profile_handler;
>       |                                         ^~~~~~~~~~~~~~~
> In file included from /usr/lib/modules/6.15.0-1-mainline/build/include/linux/container_of.h:5,
>                  from /usr/lib/modules/6.15.0-1-mainline/build/include/linux/kernel.h:22,
>                  from samsung-galaxybook.c:14:
> /usr/lib/modules/6.15.0-1-mainline/build/include/linux/build_bug.h:78:41: error: static assertion failed: "ARRAY_SIZE(profile_names) == PLATFORM_PROFILE_LAST"
>    78 | #define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
>       |                                         ^~~~~~~~~~~~~~
> /usr/lib/modules/6.15.0-1-mainline/build/include/linux/build_bug.h:77:34: note: in expansion of macro ‘__static_assert’
>    77 | #define static_assert(expr, ...) __static_assert(expr, ##__VA_ARGS__, #expr)
>       |                                  ^~~~~~~~~~~~~~~
> samsung-galaxybook.c:1275:1: note: in expansion of macro ‘static_assert’
>  1275 | static_assert(ARRAY_SIZE(profile_names) == PLATFORM_PROFILE_LAST);
>       | ^~~~~~~~~~~~~
> samsung-galaxybook.c: In function ‘galaxybook_platform_profile_set’:
> /usr/lib/modules/6.15.0-1-mainline/build/include/linux/compiler_types.h:498:27: error: expression in static assertion is not an integer
>   498 | #define __same_type(a, b) __builtin_types_compatible_p(typeof(a), typeof(b))
>       |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
> /usr/lib/modules/6.15.0-1-mainline/build/include/linux/build_bug.h:78:56: note: in definition of macro ‘__static_assert’
>    78 | #define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
>       |                                                        ^~~~
> /usr/lib/modules/6.15.0-1-mainline/build/include/linux/container_of.h:20:9: note: in expansion of macro ‘static_assert’
>    20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
>       |         ^~~~~~~~~~~~~
> /usr/lib/modules/6.15.0-1-mainline/build/include/linux/container_of.h:20:23: note: in expansion of macro ‘__same_type’
>    20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
>       |                       ^~~~~~~~~~~
> samsung-galaxybook.c:1280:49: note: in expansion of macro ‘container_of’
>  1280 |         struct samsung_galaxybook *galaxybook = container_of(pprof,
>       |                                                 ^~~~~~~~~~~~
> samsung-galaxybook.c: In function ‘galaxybook_platform_profile_get’:
> /usr/lib/modules/6.15.0-1-mainline/build/include/linux/compiler_types.h:498:27: error: expression in static assertion is not an integer
>   498 | #define __same_type(a, b) __builtin_types_compatible_p(typeof(a), typeof(b))
>       |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
> /usr/lib/modules/6.15.0-1-mainline/build/include/linux/build_bug.h:78:56: note: in definition of macro ‘__static_assert’
>    78 | #define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
>       |                                                        ^~~~
> /usr/lib/modules/6.15.0-1-mainline/build/include/linux/container_of.h:20:9: note: in expansion of macro ‘static_assert’
>    20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
>       |         ^~~~~~~~~~~~~
> /usr/lib/modules/6.15.0-1-mainline/build/include/linux/container_of.h:20:23: note: in expansion of macro ‘__same_type’
>    20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
>       |                       ^~~~~~~~~~~
> samsung-galaxybook.c:1296:49: note: in expansion of macro ‘container_of’
>  1296 |         struct samsung_galaxybook *galaxybook = container_of(pprof,
>       |                                                 ^~~~~~~~~~~~
> samsung-galaxybook.c: In function ‘galaxybook_profile_init’:
> samsung-galaxybook.c:1436:15: error: too few arguments to function ‘platform_profile_register’; expected 4, have 1
>  1436 |         err = platform_profile_register(&galaxybook->profile_handler);
>       |               ^~~~~~~~~~~~~~~~~~~~~~~~~
> In file included from samsung-galaxybook.c:23:
> /usr/lib/modules/6.15.0-1-mainline/build/include/linux/platform_profile.h:50:16: note: declared here
>    50 | struct device *platform_profile_register(struct device *dev, const char *name,
>       |                ^~~~~~~~~~~~~~~~~~~~~~~~~
[... 32 lines cut ...]
>       |                                     ^~~~~~~~~~~~~~~~~~~~~~~
>       |                                     |
[... cut ...]
> make[1]: *** [/usr/lib/modules/6.15.0-1-mainline/build/Makefile:248: __sub-make] Error 2
> make[1]: Leaving directory '/var/lib/dkms/samsung-galaxybook/extras/build'
> make: *** [Makefile:248: __sub-make] Error 2
> make: Leaving directory '/usr/lib/modules/6.15.0-1-mainline/build'
> 
> # exit code: 2
> # elapsed time: 00:00:02
> ----------------------------------------------------------------
> ```

What kernel version are you currently on, the installed headers do not match mainline 6.15 try using the older version in the 6.14 branch.

Michedev, 2025-05-27:
Thanks for the reply but I tried to compile this repo with my kernel (6.14) 1 commit before of HEAD and also the pre-6.14 branch and I get the same error (at least by reading the first lines)

```(pre-6.14)>> make -C /lib/modules/6.14.7-arch2-1/build M=$PWD
make: Entering directory '/usr/lib/modules/6.14.7-arch2-1/build'
make[1]: Entering directory '/home/mikedev/samsung-galaxybook-extras'
  CC [M]  samsung-galaxybook.o
samsung-galaxybook.c:105:41: error: field ‘profile_handler’ has incomplete type
  105 |         struct platform_profile_handler profile_handler;
      |                                         ^~~~~~~~~~~~~~~
In file included from /usr/lib/modules/6.14.7-arch2-1/build/include/linux/container_of.h:5,
                 from /usr/lib/modules/6.14.7-arch2-1/build/include/linux/kernel.h:22,
                 from samsung-galaxybook.c:14:
/usr/lib/modules/6.14.7-arch2-1/build/include/linux/build_bug.h:78:41: error: static assertion failed: "ARRAY_SIZE(profile_names) == PLATFORM_PROFILE_LAST"
   78 | #define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
      |                                         ^~~~~~~~~~~~~~
/usr/lib/modules/6.14.7-arch2-1/build/include/linux/build_bug.h:77:34: note: in expansion of macro ‘__static_assert’
   77 | #define static_assert(expr, ...) __static_assert(expr, ##__VA_ARGS__, #expr)
      |                                  ^~~~~~~~~~~~~~~
samsung-galaxybook.c:1275:1: note: in expansion of macro ‘static_assert’
 1275 | static_assert(ARRAY_SIZE(profile_names) == PLATFORM_PROFILE_LAST);
      | ^~~~~~~~~~~~~
samsung-galaxybook.c: In function ‘galaxybook_platform_profile_set’:
/usr/lib/modules/6.14.7-arch2-1/build/include/linux/compiler_types.h:483:27: error: expression in static assertion is not an integer
  483 | #define __same_type(a, b) __builtin_types_compatible_p(typeof(a), typeof(b))
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/modules/6.14.7-arch2-1/build/include/linux/build_bug.h:78:56: note: in definition of macro ‘__static_assert’
   78 | #define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
      |                                                        ^~~~
/usr/lib/modules/6.14.7-arch2-1/build/include/linux/container_of.h:20:9: note: in expansion of macro ‘static_assert’
   20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
      |         ^~~~~~~~~~~~~
/usr/lib/modules/6.14.7-arch2-1/build/include/linux/container_of.h:20:23: note: in expansion of macro ‘__same_type’
   20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
      |                       ^~~~~~~~~~~
samsung-galaxybook.c:1280:49: note: in expansion of macro ‘container_of’
 1280 |         struct samsung_galaxybook *galaxybook = container_of(pprof,
      |                                                 ^~~~~~~~~~~~
samsung-galaxybook.c: In function ‘galaxybook_platform_profile_get’:
/usr/lib/modules/6.14.7-arch2-1/build/include/linux/compiler_types.h:483:27: error: expression in static assertion is not an integer
  483 | #define __same_type(a, b) __builtin_types_compatible_p(typeof(a), typeof(b))
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/modules/6.14.7-arch2-1/build/include/linux/build_bug.h:78:56: note: in definition of macro ‘__static_assert’
   78 | #define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
      |                                                        ^~~~
/usr/lib/modules/6.14.7-arch2-1/build/include/linux/container_of.h:20:9: note: in expansion of macro ‘static_assert’
   20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
      |         ^~~~~~~~~~~~~
/usr/lib/modules/6.14.7-arch2-1/build/include/linux/container_of.h:20:23: note: in expansion of macro ‘__same_type’
   20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
      |                       ^~~~~~~~~~~
samsung-galaxybook.c:1296:49: note: in expansion of macro ‘container_of’
 1296 |         struct samsung_galaxybook *galaxybook = container_of(pprof,
      |                                                 ^~~~~~~~~~~~
samsung-galaxybook.c: In function ‘galaxybook_profile_init’:
samsung-galaxybook.c:1436:15: error: too few arguments to function ‘platform_profile_register’; expected 4, have 1
 1436 |         err = platform_profile_register(&galaxybook->profile_handler);
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from samsung-galaxybook.c:23:
/usr/lib/modules/6.14.7-arch2-1/build/include/linux/platform_profile.h:50:16: note: declared here
   50 | struct device *platform_profile_register(struct device *dev, const char *name,
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~
samsung-galaxybook.c: In function ‘galaxybook_profile_exit’:
samsung-galaxybook.c:1459:9: error: too few arguments to function ‘platform_profile_remove’; expected 1, have 0
 1459 |         platform_profile_remove();
      |         ^~~~~~~~~~~~~~~~~~~~~~~
/usr/lib/modules/6.14.7-arch2-1/build/include/linux/platform_profile.h:53:5: note: declared here
   53 | int platform_profile_remove(struct device *dev);
      |     ^~~~~~~~~~~~~~~~~~~~~~~
samsung-galaxybook.c: In function ‘galaxybook_probe’:
samsung-galaxybook.c:1839:44: error: passing argument 1 of ‘i8042_install_filter’ from incompatible pointer type [-Wincompatible-pointer-types]
[... 17 lines cut ...]
[... cut ...]
   70 | typedef bool (*i8042_filter_t)(unsigned char data, unsigned char str, struct serio *serio,
      |                ^~~~~~~~~~~~~~
make[3]: *** [/usr/lib/modules/6.14.7-arch2-1/build/scripts/Makefile.build:207: samsung-galaxybook.o] Error 1
make[2]: *** [/usr/lib/modules/6.14.7-arch2-1/build/Makefile:1997: .] Error 2
make[1]: *** [/usr/lib/modules/6.14.7-arch2-1/build/Makefile:251: __sub-make] Error 2
make[1]: Leaving directory '/home/mikedev/samsung-galaxybook-extras'
make: *** [Makefile:251: __sub-make] Error 2
make: Leaving directory '/usr/lib/modules/6.14.7-arch2-1/build'
```

IltonPfleger, 2025-05-27:
> Thanks for the reply but I tried to compile this repo with my kernel (6.14) 1 commit before of HEAD and also the pre-6.14 branch and I get the same error (at least by reading the first lines)
> 
> ```
> make: Entering directory '/usr/lib/modules/6.14.7-arch2-1/build'
> make[1]: Entering directory '/home/mikedev/samsung-galaxybook-extras'
>   CC [M]  samsung-galaxybook.o
> samsung-galaxybook.c:105:41: error: field ‘profile_handler’ has incomplete type
>   105 |         struct platform_profile_handler profile_handler;
>       |                                         ^~~~~~~~~~~~~~~
> In file included from /usr/lib/modules/6.14.7-arch2-1/build/include/linux/container_of.h:5,
>                  from /usr/lib/modules/6.14.7-arch2-1/build/include/linux/kernel.h:22,
>                  from samsung-galaxybook.c:14:
> /usr/lib/modules/6.14.7-arch2-1/build/include/linux/build_bug.h:78:41: error: static assertion failed: "ARRAY_SIZE(profile_names) == PLATFORM_PROFILE_LAST"
>    78 | #define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
>       |                                         ^~~~~~~~~~~~~~
> /usr/lib/modules/6.14.7-arch2-1/build/include/linux/build_bug.h:77:34: note: in expansion of macro ‘__static_assert’
>    77 | #define static_assert(expr, ...) __static_assert(expr, ##__VA_ARGS__, #expr)
>       |                                  ^~~~~~~~~~~~~~~
> samsung-galaxybook.c:1275:1: note: in expansion of macro ‘static_assert’
>  1275 | static_assert(ARRAY_SIZE(profile_names) == PLATFORM_PROFILE_LAST);
>       | ^~~~~~~~~~~~~
> samsung-galaxybook.c: In function ‘galaxybook_platform_profile_set’:
> /usr/lib/modules/6.14.7-arch2-1/build/include/linux/compiler_types.h:483:27: error: expression in static assertion is not an integer
>   483 | #define __same_type(a, b) __builtin_types_compatible_p(typeof(a), typeof(b))
>       |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
> /usr/lib/modules/6.14.7-arch2-1/build/include/linux/build_bug.h:78:56: note: in definition of macro ‘__static_assert’
>    78 | #define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
>       |                                                        ^~~~
> /usr/lib/modules/6.14.7-arch2-1/build/include/linux/container_of.h:20:9: note: in expansion of macro ‘static_assert’
>    20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
>       |         ^~~~~~~~~~~~~
> /usr/lib/modules/6.14.7-arch2-1/build/include/linux/container_of.h:20:23: note: in expansion of macro ‘__same_type’
>    20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
>       |                       ^~~~~~~~~~~
> samsung-galaxybook.c:1280:49: note: in expansion of macro ‘container_of’
>  1280 |         struct samsung_galaxybook *galaxybook = container_of(pprof,
>       |                                                 ^~~~~~~~~~~~
> samsung-galaxybook.c: In function ‘galaxybook_platform_profile_get’:
> /usr/lib/modules/6.14.7-arch2-1/build/include/linux/compiler_types.h:483:27: error: expression in static assertion is not an integer
>   483 | #define __same_type(a, b) __builtin_types_compatible_p(typeof(a), typeof(b))
>       |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
> /usr/lib/modules/6.14.7-arch2-1/build/include/linux/build_bug.h:78:56: note: in definition of macro ‘__static_assert’
>    78 | #define __static_assert(expr, msg, ...) _Static_assert(expr, msg)
>       |                                                        ^~~~
> /usr/lib/modules/6.14.7-arch2-1/build/include/linux/container_of.h:20:9: note: in expansion of macro ‘static_assert’
>    20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
>       |         ^~~~~~~~~~~~~
> /usr/lib/modules/6.14.7-arch2-1/build/include/linux/container_of.h:20:23: note: in expansion of macro ‘__same_type’
>    20 |         static_assert(__same_type(*(ptr), ((type *)0)->member) ||       \
>       |                       ^~~~~~~~~~~
> samsung-galaxybook.c:1296:49: note: in expansion of macro ‘container_of’
>  1296 |         struct samsung_galaxybook *galaxybook = container_of(pprof,
>       |                                                 ^~~~~~~~~~~~
> samsung-galaxybook.c: In function ‘galaxybook_profile_init’:
> samsung-galaxybook.c:1436:15: error: too few arguments to function ‘platform_profile_register’; expected 4, have 1
>  1436 |         err = platform_profile_register(&galaxybook->profile_handler);
>       |               ^~~~~~~~~~~~~~~~~~~~~~~~~
> In file included from samsung-galaxybook.c:23:
> /usr/lib/modules/6.14.7-arch2-1/build/include/linux/platform_profile.h:50:16: note: declared here
>    50 | struct device *platform_profile_register(struct device *dev, const char *name,
>       |                ^~~~~~~~~~~~~~~~~~~~~~~~~
> samsung-galaxybook.c: In function ‘galaxybook_profile_exit’:
> samsung-galaxybook.c:1459:9: error: too few arguments to function ‘platform_profile_remove’; expected 1, have 0
>  1459 |         platform_profile_remove();
>       |         ^~~~~~~~~~~~~~~~~~~~~~~
> /usr/lib/modules/6.14.7-arch2-1/build/include/linux/platform_profile.h:53:5: note: declared here
>    53 | int platform_profile_remove(struct device *dev);
>       |     ^~~~~~~~~~~~~~~~~~~~~~~
> samsung-galaxybook.c: In function ‘galaxybook_probe’:
> samsung-galaxybook.c:1839:44: error: passing argument 1 of ‘i8042_install_filter’ from incompatible pointer type [-Wincompatible-pointer-types]
[... cut ...]
> make[1]: *** [/usr/lib/modules/6.14.7-arch2-1/build/Makefile:251: __sub-make] Error 2
> make[1]: Leaving directory '/home/mikedev/samsung-galaxybook-extras'
> make: *** [Makefile:251: __sub-make] Error 2
> make: Leaving directory '/usr/lib/modules/6.14.7-arch2-1/build'
> ```

Here’s what I’ve done so far:

1. Cloned the mainline Linux kernel repository (link available on GitHub).

2. Copied samsung_galaxybook.c and firmware_attributes_class.h into a local folder.

3. Retrieved the Makefile from the 6.14 branch.

4. Attempted to compile the code using the 6.14 build steps.

joshuagrisham, 2025-05-28:
Hi all,
I posted in another issue about this just today, maybe what I wrote there can be useful? https://github.com/joshuagrisham/samsung-galaxybook-extras/issues/70#issuecomment-2916722910

Basically if you get the code for 6.15 and build and install the entire 6.15 kernel then you should not take any files from this repo (the driver is already in mainline)

Michedev, 2025-05-28:
Thanks @IltonPfleger it worked! @joshuagrisham  It also worked for 6.15 but I will try with the mainline kernel as camera and speakers they don't work at the moment on my model

Michedev, 2025-05-29:
@joshuagrisham I download `linux-mainline` package on arch, enabled manually with `make menuconfig` the module samsung-galaxybook but I have the same error I had posted in a previous message

> My model is NP944XHA-KG2DE
>
> I put here the log file
>
> ```
> DKMS (dkms-3.2.1) make.log for samsung-galaxybook/extras for kernel 6.15.0-1-mainline (x86_64)
> Tue May 27 09:46:12 AM CEST 2025
>
> Building [...]