joshuagrisham/samsung-galaxybook-extras #69: Test on Notebook 9 Pro 15” (NP940X5M)

diego-karsa, 2025-05-26:
I installed the module on my NP940X5M (i7-7500U) running Kubuntu 5.27.12. I installed the module on the branch for <6.14 (using 6.8) and also the mainline 6.15rc7 kernel. Unfortunately I don't see any changes. Backlit keyboard doesn't responded to the backlit key and the computer doesn't turn on when opening the lid. I haven't seen changes in the power profiles or battery threshold, though I'm not sure if I'm expected to see something different in Kubuntu's system menu or not. No fan control device exposed via hmon.

If you have any suggestions or debug you wished I tried, let me know. I'm ok with it as long as it doesn't risk bricking my computer :D

joshuagrisham, 2025-05-31:
Hi @diego-karsa , thanks for testing this and it is fantastic news! Based on the ini files for the Windows device drivers, I always suspected that it would probably "mostly" work but I am curious was there any issue or error messages or any other weird behavior that seems like it should be "fixed"?

The intention with the code in the driver is that it should "only" enable the features that seem to be returning successful/recognized response codes, and anything else will not be enabled and ignored (e.g. the sysfs files will not be created to use the feature or similar). Does all of this seem to be working correctly and it seems "safe" to add this device ID to the driver even in mainline?

diego-karsa, 2025-06-03:
Hi @joshuagrisham, thanks for your great work on the driver!

Everything seems to be working well on my end.

The ACPI method CSXI and fan-related methods (Get fan state, read fan speed list) fail gracefully, but the module still loads without issues:

```
$ sudo dmesg | grep samsung
[    4.298998] samsung_galaxybook: found matched device SAM0426:00; loading driver
[    4.306491] samsung_galaxybook: failed get supported performance modes with ACPI method CSXI; device did not respond with success code 0xaa
[    4.362930] samsung_galaxybook: registering LED class using default name of samsung-galaxybook::kbd_backlight
[    4.372216] samsung_galaxybook: found fan device PNP0C0B:00
[    4.372222] samsung_galaxybook: Get fan state failed
[    4.372225] samsung_galaxybook: failed to read fan speed list
[    4.372226] samsung_galaxybook: unable to initialize fan speeds for fan device PNP0C0B:00 (FAN0)
[    4.372232] samsung_galaxybook: found fan device PNP0C0B:01
[    4.372235] samsung_galaxybook: Get fan state failed
[    4.372237] samsung_galaxybook: failed to read fan speed list
[    4.372238] samsung_galaxybook: unable to initialize fan speeds for fan device PNP0C0B:01 (FAN1)
[    4.396818] samsung_galaxybook: driver successfully loaded
```
The device exposes the following entries under /sys/devices/platform/SAM0426:00/:
```
$ find /sys/devices/platform/SAM0426\:00/
/sys/devices/platform/SAM0426:00/
/sys/devices/platform/SAM0426:00/uevent
/sys/devices/platform/SAM0426:00/start_on_lid_open
/sys/devices/platform/SAM0426:00/allow_recording
/sys/devices/platform/SAM0426:00/firmware_node
/sys/devices/platform/SAM0426:00/power
/sys/devices/platform/SAM0426:00/power/runtime_active_time
/sys/devices/platform/SAM0426:00/power/runtime_active_kids
/sys/devices/platform/SAM0426:00/power/runtime_usage
/sys/devices/platform/SAM0426:00/power/runtime_status
/sys/devices/platform/SAM0426:00/power/autosuspend_delay_ms
/sys/devices/platform/SAM0426:00/power/async
/sys/devices/platform/SAM0426:00/power/runtime_suspended_time
/sys/devices/platform/SAM0426:00/power/runtime_enabled
/sys/devices/platform/SAM0426:00/power/control
/sys/devices/platform/SAM0426:00/leds
/sys/devices/platform/SAM0426:00/leds/samsung-galaxybook::kbd_backlight
/sys/devices/platform/SAM0426:00/leds/samsung-galaxybook::kbd_backlight/uevent
/sys/devices/platform/SAM0426:00/leds/samsung-galaxybook::kbd_backlight/brightness
/sys/devices/platform/SAM0426:00/leds/samsung-galaxybook::kbd_backlight/power
/sys/devices/platform/SAM0426:00/leds/samsung-galaxybook::kbd_backlight/power/runtime_active_time
/sys/devices/platform/SAM0426:00/leds/samsung-galaxybook::kbd_backlight/power/runtime_active_kids
/sys/devices/platform/SAM0426:00/leds/samsung-galaxybook::kbd_backlight/power/runtime_usage
/sys/devices/platform/SAM0426:00/leds/samsung-galaxybook::kbd_backlight/power/runtime_status
/sys/devices/platform/SAM0426:00/leds/samsung-galaxybook::kbd_backlight/power/autosuspend_delay_ms
/sys/devices/platform/SAM0426:00/leds/samsung-galaxybook::kbd_backlight/power/async
/sys/devices/platform/SAM0426:00/leds/samsung-galaxybook::kbd_backlight/power/runtime_suspended_time
/sys/devices/platform/SAM0426:00/leds/samsung-galaxybook::kbd_backlight/power/runtime_enabled
/sys/devices/platform/SAM0426:00/leds/samsung-galaxybook::kbd_backlight/power/control
/sys/devices/platform/SAM0426:00/leds/samsung-galaxybook::kbd_backlight/device
/sys/devices/platform/SAM0426:00/leds/samsung-galaxybook::kbd_backlight/trigger
/sys/devices/platform/SAM0426:00/leds/samsung-galaxybook::kbd_backlight/brightness_hw_changed
/sys/devices/platform/SAM0426:00/leds/samsung-galaxybook::kbd_backlight/subsystem
/sys/devices/platform/SAM0426:00/leds/samsung-galaxybook::kbd_backlight/max_brightness
/sys/devices/platform/SAM0426:00/usb_charge
/sys/devices/platform/SAM0426:00/driver
/sys/devices/platform/SAM0426:00/driver_override
/sys/devices/platform/SAM0426:00/subsystem
/sys/devices/platform/SAM0426:00/input
/sys/devices/platform/SAM0426:00/input/input20
/sys/devices/platform/SAM0426:00/input/input20/uevent
/sys/devices/platform/SAM0426:00/input/input20/capabilities
/sys/devices/platform/SAM0426:00/input/input20/capabilities/rel
/sys/devices/platform/SAM0426:00/input/input20/capabilities/abs
/sys/devices/platform/SAM0426:00/input/input20/capabilities/ff
/sys/devices/platform/SAM0426:00/input/input20/capabilities/led
[... 36 lines cut ...]
/sys/devices/platform/SAM0426:00/input/input20/event14/power/control
/sys/devices/platform/SAM0426:00/input/input20/event14/dev
/sys/devices/platform/SAM0426:00/input/input20/event14/device
/sys/devices/platform/SAM0426:00/input/input20/event14/subsystem
/sys/devices/platform/SAM0426:00/input/input20/phys
/sys/devices/platform/SAM0426:00/input/input20/name
/sys/devices/platform/SAM0426:00/input/input20/modalias
/sys/devices/platform/SAM0426:00/modalias
```

Features I've tested:

- Keyboard backlight: Works well. Fn + F9 keys controls brightness correctly. Sysfs entries are created as expected:
```
$ ls /sys/class/leds/samsung-galaxybook::kbd_backlight/
brightness  brightness_hw_changed  device  max_brightness  power  subsystem  trigger  uevent
```
- Start on lid open: Functional. I just had to set /sys/devices/platform/SAM0426:00/start_on_lid_open to 1.
- USB charging when powered off (charging a device using the USB-C): Works. I tested charging a phone via USB-C by setting /sys/devices/platform/SAM0426:00/usb_charge to 1.

I don't believe the battery threshold control is exposed by the module. I initially thought it was, since Kubuntu offers an option to set it through System Settings — but that feature works even without the module loaded, so it must be using a different mechanism.

I've been using the laptop intensively for the past week without encountering any issues. From my end, everything looks stable and safe to include in mainline.

Let me know if you'd like me to test anything else!

joshuagrisham, 2025-06-06:
Thanks again @diego-karsa that sounds like great news! It looks like from the info you pasted that you are using the driver from my code here instead of the one in mainline but I think it should work just as well in mainline (it is more that the platform attributes will be located in a different path, and maybe have slightly different names/values, but the functionality should be the same)

I have submitted a patch to add your device ID to mainline as well, you can find it here: https://lore.kernel.org/platform-driver-x86/20250606130909.207047-1-josh@joshuagrisham.com/T/#u

diego-karsa, 2025-06-19:
Hi Joshua. I couldn't find any mention of `_FST` in the DSDT. This is all I see for "PNP0C0B":

```
        Device (FAN0)
        {
            Name (_HID, EisaId ("PNP0C0B") /* Fan (Thermal Solution) */)  // _HID: Hardware ID
            Name (_UID, Zero)  // _UID: Unique ID
            Name (_PR0, Package (0x01)  // _PR0: Power Resources for D0
            {
                FN00
            })
        }

        PowerResource (FN01, 0x00, 0x0000)
        {
            Method (_STA, 0, Serialized)  // _STA: Status
            {
                Return (VFN1) /* \VFN1 */
            }

            Method (_ON, 0, Serialized)  // _ON_: Power On
            {
                VFN1 = One
            }

            Method (_OFF, 0, Serialized)  // _OFF: Power Off
            {
                VFN1 = Zero
            }
        }

        Device (FAN1)
        {
            Name (_HID, EisaId ("PNP0C0B") /* Fan (Thermal Solution) */)  // _HID: Hardware ID
            Name (_UID, One)  // _UID: Unique ID
            Name (_PR0, Package (0x01)  // _PR0: Power Resources for D0
            {
                FN01
            })
        } 
```