joshuagrisham/samsung-galaxybook-extras #71: No sound on Galaxy Book5 Pro 360

manatails, 2025-06-05:
Recently bought this laptop and I am pleasantly surprised to find out that the Linux experience is much better than I expected to be. Thank you for your hard work on the drivers and the guides.

The only thing I am concerned about is the lack of onboard sound, and unfortunately the verb scripts didn't work on this model.

After looking into it further, I discovered that this model has a speaker amp MAX98390 from Maxim on Intel I2C bus. On Windows Device Manager it is registered as 'ADI Amp Device'.

The chip is already seemingly supported by the Linux kernel, but it does not work out of the box as on these laptops it appears as an ACPI device with ID 'MAX98390', but the existing driver only recognizes the ID 'MX98390'.

```
#ifdef CONFIG_ACPI
static const struct acpi_device_id max98390_acpi_match[] = {
	{ "MX98390", 0 }, // Change this to 'MAX98390'
	{},
};
MODULE_DEVICE_TABLE(acpi, max98390_acpi_match);
#endif
```

I updated the ACPI ID from `sound/soc/codecs/max98390.c` and recompiled the module, now it comes up on dmesg.

> [    4.525926] max98390 i2c-MAX98390:00: no optional property 'temperature_calib' found, default:
> [    4.525927] max98390 i2c-MAX98390:00: no optional property 'r0_calib' found, default:
> [    4.525928] max98390 i2c-MAX98390:00: max98390_i2c_probe: r0_calib: 0x0,temperature_calib: 0x0
> [    4.526182] max98390 i2c-MAX98390:00: MAX98390 revisionID: 0x42

Still, it seems that I need a topology file to activate the amp. But I lack expertise on those areas.
The audio codec on this model is ALC298 I believe, I would appreciate any guidance on enabling speaker output on this laptop.

melvyn2, 2025-06-14:
With this patch:
```patch
diff --git a/sound/soc/codecs/max98390.c b/sound/soc/codecs/max98390.c
index 76296176f948..beba1d66dc0a 100644
--- a/sound/soc/codecs/max98390.c
+++ b/sound/soc/codecs/max98390.c
@@ -1118,6 +1118,7 @@ MODULE_DEVICE_TABLE(of, max98390_of_match);
 #ifdef CONFIG_ACPI
 static const struct acpi_device_id max98390_acpi_match[] = {
 	{ "MX98390", 0 },
+	{ "MAX98390", 0 },
 	{},
 };
 MODULE_DEVICE_TABLE(acpi, max98390_acpi_match);
```
I get that the amp is found by its dedicated driver:
```
[    4.467579] max98390 i2c-MAX98390:00: no optional property 'temperature_calib' found, default:
[    4.467582] max98390 i2c-MAX98390:00: no optional property 'r0_calib' found, default:
[    4.467583] max98390 i2c-MAX98390:00: max98390_i2c_probe: r0_calib: 0x0,temperature_calib: 0x0
[    4.467829] max98390 i2c-MAX98390:00: MAX98390 revisionID: 0x42
```
But then the `snd_soc_acpi_intel_match` module doesn't pick up the amp:
```
> dmesg | grep '\bamp\b'
>
```
Even though dynamic debugging is clearly enabled for it with `snd_soc_acpi_intel_match.dyndbg=+p`:
```
> grep snd_soc_acpi_intel_match /proc/dynamic_debug/control
sound/soc/intel/common/soc-acpi-intel-ssp-common.c:87 [snd_soc_acpi_intel_match]snd_soc_acpi_intel_detect_codec_type =p "codec %s found\n"
sound/soc/intel/common/soc-acpi-intel-ssp-common.c:104 [snd_soc_acpi_intel_match]snd_soc_acpi_intel_detect_amp_type =p "amp %s found\n"
```

manatails, 2025-06-17:
I am also getting the same results as @melvyn2, and unfortunately the verb scripts didn't work with the driver enabled either.

topzyray, 2025-08-03:
Thanks to @joshuagrisham. I was able to get my Galaxy Book Ion 950XDA sound working perfectly today. After 4 days of frantic research, trials, failures, I was able to fix it. I decided to create a script to automate the fix using Bash and made it open source. A friend of my used this today on another device and it also worked. @manatails you can checkout the script [https://github.com/topzyray/galaxybook-linux-setup](https://github.com/topzyray/galaxybook-linux-setup). I hope it works for you. Thank you.

tope-taiwo, 2025-08-06:
Hi @CReimer . Sorry to hear that. Sorry, I forgot to include that there are addtional checks to ensure things are right.

Can you confirm the following:
1. What is your Ubuntu and kernel version?
2. What specifically is not working? Internal speakers, microphone, or both?
3. Is sound available via headphones (I meant wired-headphone)?
4. Have you tried connected your device to a bluetooth device to see if you get sound?
5. Can you check if your device default sound card and verify if the sound channels are not muted using `alsamixer`?

Also, can you share the output of the following in a .md file?
1.` sudo dmesg | grep -i 'sof\|snd\|audio'` - this helps to show kernel messages related to sound open firmware, audio or the likes
2. `lspci | grep -i audio` - this will list your audio devices
3. ps -e | grep -E 'pipewire|pulseaudio' - to  check if your sound system is active and confirm if your Ubuntu version uses pipewire or pulseaudio

These can help debug further. Thank you.

kimusan, 2025-08-06:
I am seeing similar results with the galaxy book5 pro 14" running Ubuntu 24.04 and kernel 6.16.0:

[dmesg.txt](https://github.com/user-attachments/files/21634115/dmesg.txt)

[lspci.txt](https://github.com/user-attachments/files/21634128/lspci.txt)

[ps.txt](https://github.com/user-attachments/files/21634138/ps.txt)

CReimer, 2025-09-18:
@joshuagrisham Do these dumps show you anything that I might have missed?